syssi/esphome-pipsolar #120: Apiconnectionerror

test2a, 2024-04-02:
My ESP8266 just randomly decides to loose connection. Sometimes after a day, sometimes multiple times during the day.

I did write earlier issue where unknown was showing, this one is different as no output is given and home assistant cant even connect to the device.

My question is, is this related to WiFi signal ( I doubt it, the ESP and router position doesn't change ) or device gets overloaded or something else ?

I cant have a wired connection for some wiring problem but is there some way I can ensure the device is reliable ?

Today I went to the esp, disconected its power and tried to get a connection. It didnt.

syssi, 2024-04-02:
Please attach a serial connection to the ESP and monitor/provide a log.

syssi, 2024-04-02:
Long story short: Home Assistant doesn't provide any details about the ESP. HA just reports "the device is gone and communication using the API isn't possible anymore". If you take a look at the log book of the ESPHome device you should find more details about the root cause (WiFi connection lost, crashs or boot loops etc.).

closed 2024-08-12 by syssi