syssi/esphome-pipsolar #124: Hybrid Solar inverter 8.2kw/10.2KW

Nando1497, 2024-05-04:
Hello,
Amazing job you did .... Is-il possible to extend your work to support for this inverter: https://www.nextpowersolar.com/48V-8-2KW-10-2KW-220V-HYBRID-SOLAR-INVERTER-DUAL-PV-and-OUTPUT-pd747238888.html 
Modul:  m5stack-atom and  RS-232 

I'm willing to offer you more than a caffee.
thanks in advance

Nando1497, 2024-05-04:
[image]

syssi, 2024-05-08:
Did you try the protocol test already?

syssi, 2024-05-08:
https://diysolarforum.com/threads/victor-max-10-2-kw-rs-232-protocol.82006/

Nando1497, 2024-05-08:
No, forgive my ignorance.. how do I do this test?

syssi, 2024-05-08:
Please try this one and provide the output: https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

Could you ask the seller/manufacturer about some protocol details? It looks like your inverter doesn't support the PIP protocol.

Nando1497, 2024-05-15:
[logs_inverter_run.txt](https://github.com/syssi/esphome-pipsolar/files/15324731/logs_inverter_run.txt)

Patogoma, 2024-05-15:
I have an Anern 10.2k inverter. Is the same model and the pip8048 version works well except the QPIGS2 command. SmartESS shows correctly PV2 information.

syssi, 2024-05-15:
Please try this YAML as next step. It should work out of the box:

https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp8266-example.yaml

patogomaII, 2024-05-16:
[image]
No way...

syssi, 2024-05-16:
I don't understand your comment. What's not working as expected?

patogomaII, 2024-05-16:
current_max_charging_current doesn't work either

patogomaII, 2024-05-16:
PV2 is unknow. QPGIS2 returns NAK

patogomaII, 2024-05-16:
Perhaps this inverters uses a custom protocol with length and CRCs definition like this:
2.10 QPIGS2<cr>: Device general status parameters inquiry
Computer: QPIGS 2<CRC><cr>
Device: (N1N2N3N4 N6N7N8.N10 N12N13.N15N16 N18N19N20N21N22 b24b25b26b27b28b29b30b31
N33N34N35N36 N38N39N40N41 N42N43N44N45 N47N48N49.N51 N53N54.N56N57 N58N59N60N61
N63N64N65N66N67<CRC><cr>
I extract this from https://github.com/opengd/solar-sis/files/3041962/Communication.Protocol-20150924.-.Customer.pdf
and they have a same issue.

Nando1497, 2024-05-17:
I can't compile it gives me these errors:
[image]
[image]

Nando1497, 2024-05-17:
Hi PatogomaII, do you also have this Hybrid Solar inverter 8.2kw/10.2KW model? If so, could you kindly send me the .yaml

Thank you

patogomaII, 2024-05-17:
I have the same inverter but with other name. The yaml thar say syssi compiles and work correctly, but fails reading pv2 data. If you have compilation problems, look at the tipe of esp, pins used or something like it.

Nando1497, 2024-05-17:
Thanks for the advice, I managed.. However, for PV2 I don't read it either but comparing the data from the inverter it seems that it only exposes PV1 by adding both values..

patogomaII, 2024-05-17:
I compare values with the inverters screen and only shows PV1 data.

Nando1497, 2024-05-17:
Sorry.. I confirm it does not correspond with the "KW" value "pv1_charging_power", but the "V" pv1_input_voltage value corresponds.

patogomaII, 2024-05-17:
In my case, all the values of PV1 correspond to same values in the inverter screen.

Nando1497, 2024-05-19:
I found this
https://esphome.io/components/pipsolar.html#output-pipsolar-set-level-action

patogomaII, 2024-05-20:
And what do you want to do with this?

Nando1497, 2024-05-20:
Do you receive this log?
[image]

syssi, 2024-05-20:
This warning can be ignored!

patogomaII, 2024-05-23:
Ok. Actually, people from the PVControl+ proyect are testing a POWMR 10.2kw inverter (like us but of other father), and they are retrieving data from PV2 using 4563 and 4564 registers of serial communication.

patogomaII, 2024-05-29:
[image]
There's the data. I can take it with mqtt.

patogomaII, 2024-05-30:
[image]
Here they are!!!!

patogomaII, 2024-06-20:
Ok. I have mine running with its own dongle programmed with this:
https://github.com/odya/esphome-powmr-hybrid-inverter/tree/main
Only need to modify the lines where reads pvdata and add similar code to read pv2data with registers 4563 and 4564, and voilá.

Nando1497, 2024-06-23:
Ciao, grazie per l'aiuto. Non riesco a trovare il parametro pvdata, sto cercando nel file inverter.yaml.. Corretto ?

Nando1497, 2024-06-23:
Hi, thanks for the help. I can't find the "pvdata" parameter, I'm looking in the inverter.yaml file.. Correct?

patogomaII, 2024-06-23:
Look here:
https://github.com/odya/esphome-powmr-hybrid-inverter/issues/31

Nando1497, 2024-06-23:
OK, thanks a lot

closed 2024-06-23 by Nando1497