syssi/esphome-pipsolar #127: New sensors and custom settings on mi invertor Effekta AX- Series

tiimsvk, 2024-05-13:
Hello. I made a new configuration file for my solar inverter Effekta AX-Series AX-P1 3000-24

It is now possible to add to entity_category entities
It is now possible to toggle the state of QFLAG (lcd backlight, buzzer alarm, etc.)
Fixed MUCHGC and MNCHGC values
Added values such as power solar, battery
Added values such as total energy

Connecting the inverter via a digital isolator outside the USB interface. (see yaml)

```
#-------------------------------------------
# BUG AND REPAIR AND PLANNED
#-------------------------------------------
# Added:
# -1 relay input 230AC, output 2 relay
# -CT clamp input
# -BL0939 power meter output
# -Battery temperature
# -Segment display? 
# -Mosfet output 24V

#-------------------------------------------
# REVISION
#-------------------------------------------
# 2024.05.09 - Fisrt version

#-------------------------------------------
# SUBTITUTIONS - basic definitions and description of the device
#-------------------------------------------
substitutions:
  friendly_name: energy-solar
  device_name: energy-solar
  created_by: "StudioTiiM 2024"
  device_description: "Monitoring solar inverter"

#-------------------------------------------
# POWER CONSUMPTION 
# whit multimeter
#-------------------------------------------
# Power consumption 
# Normal-  wifi:            mA

#-------------------------------------------
# DEFINE PINS AND VARIABILES
# CIRCUIT AND MODULES - basic definitions and description of the device
#-------------------------------------------

 # ----- MCU -----
  # Module: ESP32-S + adapter
   # Circuit: Solder on Prototype PCB expansion board ESP32 7x9cm
   # LDO:  Voltage in: max 6.5V, out: 3.3 +/- 0.3V, Current: 500mA / Drop voltage 0.260V / Quiescent current: 40µA
   # BATTERY PROTECTION: LP4054 Voltage in: max 6.5V, out: 4.2V, Current charging: 500mA Ires: 2kΩ

   # Power Consumption: Peak 450mA  / Normal wifi:  150mA  / Normal: 50mA  / Wifi disable: / DeepSleep: 0.01mA 
   # Specification: Chip Family: ESP32 / Chip Model: ESP32-D0WDQ6 (revision 1) / Number of Cores: 2 /Max CPU Frequency: 240MHz / Has Bluetooth: YES / Has Embedded Flash: NO / Has Factory-Calibrated ADC: YES /MAC Address: C8:C9:A3:C7:C4:F4
   
   # Pinout:
    #                    +-----------------------+
    #                    |  +--+  +--+  +--+--+  |
    #                    |  |  |  |  |  |  |  |  |
    #                    |  |  +--+  +--+  |  |  |
    #                    |                       |
    #             GND --+|                       |+-- GND
    #             3V3 --+|                       |+-- GPIO23 VSPI
    #              EN --+|                       |+-- GPIO22 VSPI SCL
    # ADC1 VP  GPIO36 --+|   +----------+        |+-- GPIO01 TXD0 CLKOU3
    # ADC1 VN  GPIO39 --+|   | MCU      |        |+-- GPIO03 RXD0 CLKOUT2
    # ADC1     GPIO34 --+|   |          |        |+-- GPIO21 VSPI SDA
    # ADC1     GPIO35 --+|   |          |        |+-- NC
[... 1749 lines cut ...]
        #"110": "110A"
        #"120": "120A"
        
  #- platform: pipsolar
  #  pipsolar_id: effekta_ax
  #  battery_bulk_voltage:
  #    name: bulk voltage battery
  #    optionsmap:
  #      "28.8": "28.8"
  #      "28.9": "28.9"
  #      "29.0": "29.0"
  #    statusmap:
  #      "28.8": "28.8V"
  #      "28.9": "28.9V"
  #      "29.0": "29.0V"

#-------------------------------------------
# TIME COMPONENT
#-------------------------------------------
time:
  - platform: sntp
```

Code:
[pipsolar.zip](https://github.com/syssi/esphome-pipsolar/files/15289552/pipsolar.zip)

tiimsvk, 2024-06-04:
Final: https://github.com/tiimsvk/Effekta-AX-Series-Axpert-MKS-esphome

genus-giuseppe, 2024-07-12:
HI,
I have a 10.2 kW inverter with 2 pv inputs. can you help me with the configuration?

Not all commands work with pipsolar

tiimsvk, 2024-07-21:
> HI, I have a 10.2 kW inverter with 2 pv inputs. can you help me with the configuration?
> 
> Not all commands work with pipsolar

it is necessary to have the uart communication function turned on.  and in the log you will see what you receive, based on that you can already divide the string into the necessary values.