syssi/esphome-pipsolar #131: Axpert_KSMKSV_RS232_Protocol

SeByDocKy, 2024-06-01:
Hi @syssi,

Not sure if it was implemented but seems there is a new variant of the protocol : 
[Axpert_KSMKSV_RS232_Protocol-C20170821.pdf](https://github.com/user-attachments/files/15521132/Axpert_KSMKSV_RS232_Protocol-C20170821.pdf)

Can you confirm it's a new protocol ?

jopl, 2024-06-29:
Dear SeByDocKy,
I can confirm it fits more to my AXPERT MAX 11KW TWIN, at least in QPIRI and QPIGS inquiry commands. And it is very similar to protocol described in attached Communication_Protocol_20150924_Customer.pdf. But "yours" is more corresponding ...
[Communication_Protocol_20150924_Customer.pdf](https://github.com/user-attachments/files/16038625/Communication_Protocol_20150924_Customer.pdf)

SeByDocKy, 2024-06-29:
I don't own myself this inverter but when I will get a bit of time, I will make a PR for this inverter. Modifications from existing are small in a first glance.

jopl, 2024-06-29:
SeByDocKy - thanks so much.
In the meanwhile, I found out addition (uncommenting) of

external_components:
  - source: github://syssi/esphome-pipsolar@2424mse1
    refresh: 0s

plus addition of "select" entity from example of @2424mse1 to my yaml, i.e.:
 
select:
  - platform: pipsolar
    pipsolar_id: ${inv_id}
    output_source_priority:
      id: ${inv_id}_output_source_priority_select
      name: ${inv_id}_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"

  - platform: pipsolar
    pipsolar_id: ${inv_id}
    charging_discharging_control:
      # See MAX_Communication_Protocol_20200526.pdf page 18
      name: "${devicename} charging discharging control"
      optionsmap:
        "111": "PBATCD111"
        "011": "PBATCD011"
        "101": "PBATCD101"
        "110": "PBATCD110"
        "010": "PBATCD010"
        "100": "PBATCD100"
        "001": "PBATCD001"
        "000": "PBATCD000"
      statusmap:
        "111": "111"
        "011": "011"
        "101": "101"
        "110": "110"
        "010": "010"
        "100": "100"
        "001": "001"
        "000": "000"

helped me and now I am able to read data without CRC NOK :-).
Regards, Josef.

SeByDocKy, 2024-07-02:
Do you mean the branch 2424mse1 is the right one for your inverter ?

jopl, 2024-07-02:
Finally, "- source: github://syssi/esphome-pipsolar@pip8048" showed the best results. It supports 2 FV sections.
Regards, Josef.

duymelody9x, 2024-07-26:
> Finally, "- source: github://syssi/esphome-pipsolar@pip8048" showed the best results. It supports 2 FV sections. Regards, Josef.

hi can you show me your yaml file i am also having problem with add select

syssi, 2024-07-26:
@duymelody9x Do you want to use an ESP32 or ESP8266?

syssi, 2024-07-26:
These are the YAMLs you are looking for:

https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp8266-example.yaml
https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

If the build fails please try a clean build: `esphome clean esp8266-example.yaml`. If you use the ESPHome Dashboard there is a CLEAN build button:

[image]