syssi/esphome-pipsolar #133: PIP 8048 zero values

sergeyshadow, 2024-06-11:
Hello
Earlier @syssi helped me set up monitoring of my inverter and I was happy )
But after some time my inverter's external display module failed. I replaced it. But now ESP returns 0 in all values, judging by the debug log. At the same time, the data is read correctly on the display itself and when connected via USB.
What am I doing wrong?

> [11:56:47][D][pipsolar:863]: CRC OK
[11:56:47][D][pipsolar:544]: Decode QFLAG
[11:56:48][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[11:56:48][D][pipsolar:926]: Sending polling command : QBATCD with length 6
[11:56:48][D][uart_debug:158]: <<< "QBATCD\xAA&\r"
[11:56:48][D][pipsolar:860]: checking crc on incoming message
[11:56:48][D][pipsolar:863]: CRC OK
[11:56:48][D][pipsolar:758]: Decode QBATCD
[11:56:48][D][pipsolar.select:031]: got value: QBATCD
[11:56:48][D][pipsolar.select:036]: could not find option QBATCD in mapping
[11:56:49][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:56:49][D][pipsolar:926]: Sending polling command : QPIRI with length 5
[11:56:49][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[11:56:49][D][pipsolar:860]: checking crc on incoming message
[11:56:49][D][pipsolar:863]: CRC OK
[11:56:49][D][pipsolar:479]: Decode QPIRI
[11:56:49][V][sensor:043]: 'pipsolar battery_bulk_voltage': Received new state 0.000000
[11:56:49][D][sensor:094]: 'pipsolar battery_bulk_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:49][V][sensor:043]: 'pipsolar battery_float_voltage': Received new state 0.000000
[11:56:49][D][sensor:094]: 'pipsolar battery_float_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:49][V][sensor:043]: 'pipsolar current_max_ac_charging_current': Received new state 0.000000
[11:56:49][D][sensor:094]: 'pipsolar current_max_ac_charging_current': Sending state 0.00000 A with 0 decimals of accuracy
[11:56:49][D][pipsolar.select:031]: got value: 0
[11:56:49][D][pipsolar.select:036]: could not find option 0 in mapping
[11:56:49][V][sensor:043]: 'pipsolar current_max_charging_current': Received new state 0.000000
[11:56:49][D][sensor:094]: 'pipsolar current_max_charging_current': Sending state 0.00000 A with 0 decimals of accuracy
[11:56:49][D][pipsolar.select:031]: got value: 0
[11:56:49][D][pipsolar.select:036]: could not find option 0 in mapping
[11:56:49][D][pipsolar.select:031]: got value: 0
[11:56:49][D][pipsolar.select:033]: found mapped option Utility Solar Battery for option 0
[11:56:50][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[11:56:50][D][pipsolar:926]: Sending polling command : QPIGS with length 5
[11:56:50][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[11:56:50][D][pipsolar:860]: checking crc on incoming message
[11:56:50][D][pipsolar:863]: CRC OK
[11:56:50][D][pipsolar:502]: Decode QPIGS
[11:56:50][V][sensor:043]: 'pipsolar grid_voltage': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar grid_frequency': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar ac_output_voltage': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar ac_output_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar ac_output_frequency': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar ac_output_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar ac_output_apparent_power': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar ac_output_apparent_power': Sending state 0.00000 VA with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar ac_output_active_power': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar ac_output_active_power': Sending state 0.00000 W with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar output_load_percent': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar output_load_percent': Sending state 0.00000 % with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar bus_voltage': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar bus_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar battery_voltage': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar battery_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar battery_charging_current': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar battery_capacity_percent': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar battery_capacity_percent': Sending state 0.00000 % with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar inverter_heat_sink_temperature': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar inverter_heat_sink_temperature': Sending state 0.00000 °C with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar pv1_input_current': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar pv1_input_current': Sending state 0.00000 A with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar pv1_input_voltage': Received new state 0.000000
[11:56:50][D][sensor:094]: 'pipsolar pv1_input_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[11:56:50][V][sensor:043]: 'pipsolar battery_voltage_scc': Received new state 0.000000
[... 31 lines cut ...]
[11:56:51][W][component:237]: Component esphome.coroutine took a long time for an operation (60 ms).
[11:56:51][W][component:238]: Components should block for at most 30 ms.
[11:56:52][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:56:52][D][pipsolar:926]: Sending polling command : QMOD with length 4
[11:56:52][D][uart_debug:158]: <<< "QMODI\xC1\r"
[11:56:52][D][pipsolar:860]: checking crc on incoming message
[11:56:52][D][pipsolar:863]: CRC OK
[11:56:52][D][pipsolar:536]: Decode QMOD
[11:56:52][V][text_sensor:013]: 'pipsolar device_mode': Received new state M
[11:56:52][D][text_sensor:064]: 'pipsolar device_mode': Sending state 'M'
[11:56:53][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[11:56:53][D][pipsolar:926]: Sending polling command : QFLAG with length 5
[11:56:53][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[11:56:53][D][pipsolar:860]: checking crc on incoming message
[11:56:53][D][pipsolar:863]: CRC OK
[11:56:53][D][pipsolar:544]: Decode QFLAG
[11:56:54][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[11:56:54][D][pipsolar:926]: Sending polling command : QBATCD with length 6
[11:56:54][D][uart_debug:158]: <<< "QBATCD\xAA&\r"
[11:56:54][D][pipsolar:860]: checking crc on incoming message
[11:56:54][D][pipsolar:863]: CRC OK
[11:56:54][D][pipsolar:758]: Decode QBATCD
[11:56:54][D][pipsolar.select:031]: got value: QBATCD
[11:56:54][D][pipsolar.select:036]: could not find option QBATCD in mapping
[11:56:55][D][uart_debug:158]: >>> "QPIRI\xF8T\r"

syssi, 2024-06-11:
Take a careful look at the traffic:

```
[11:56:54][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[11:56:54][D][uart_debug:158]: <<< "QBATCD\xAA&\r"
```

This isn't a response of your inverter. You are receiving a local echo of your request. Do you have a shortcut between the TX and RX line?

sergeyshadow, 2024-06-11:
> Take a careful look at the traffic:
> 
> ```
> [11:56:54][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
> [11:56:54][D][uart_debug:158]: <<< "QBATCD\xAA&\r"
> ```
> 
> This isn't a response of your inverter. You are receiving a local echo of your request. Do you have a shortcut between the TX and RX line?

Yes, you are right, I mixed up rx tx in yaml when I tried to find the problem.
Now the log looks like this

> [12:46:57][D][pipsolar:843]: timeout command to poll: QMOD
[12:46:57][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[12:46:57][D][pipsolar:926]: Sending polling command : QFLAG with length 5
[12:47:02][D][pipsolar:843]: timeout command to poll: QFLAG
[12:47:02][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[12:47:02][D][pipsolar:926]: Sending polling command : QBATCD with length 6
[12:47:07][D][pipsolar:843]: timeout command to poll: QBATCD
[12:47:07][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[12:47:07][D][pipsolar:926]: Sending polling command : QPIRI with length 5
[12:47:12][D][pipsolar:843]: timeout command to poll: QPIRI
[12:47:12][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[12:47:12][D][pipsolar:926]: Sending polling command : QPIGS with length 5
[12:47:12][V][sensor:043]: 'Battery Power Discharging': Received new state nan
[12:47:12][D][sensor:094]: 'Battery Power Discharging': Sending state nan W with 2 decimals of accuracy
[12:47:12][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:47:12][V][json:058]: Size after shrink 108 bytes
[12:47:17][D][pipsolar:843]: timeout command to poll: QPIGS
[12:47:17][D][uart_debug:158]: >>> "QPIGS2h-\r"
[12:47:17][D][pipsolar:926]: Sending polling command : QPIGS2 with length 6
[12:47:22][D][pipsolar:843]: timeout command to poll: QPIGS2
[12:47:22][D][uart_debug:158]: >>> "QMODI\xC1\r"
[12:47:22][D][pipsolar:926]: Sending polling command : QMOD with length 4
[12:47:26][V][sensor:043]: 'Battery Power Charging': Received new state nan
[12:47:26][D][sensor:094]: 'Battery Power Charging': Sending state nan W with 2 decimals of accuracy
[12:47:26][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[12:47:26][V][json:058]: Size after shrink 108 bytes
[12:47:27][D][pipsolar:843]: timeout command to poll: QMOD
[12:47:27][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[12:47:27][D][pipsolar:926]: Sending polling command : QFLAG with length 5
[12:47:32][D][pipsolar:843]: timeout command to poll: QFLAG
[12:47:32][D][uart_debug:158]: >>> "QBATCD\xAA&\r"

syssi, 2024-06-11:
``` 
[12:47:17][D][uart_debug:158]: >>> "QPIGS2h-\r"
[12:47:22][D][uart_debug:158]: >>> "QMODI\xC1\r"
[12:47:27][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[12:47:32][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
```

There are no responses. Please check the wiring another time and try to check very component between your ESP and the inverter mainboard.

sergeyshadow, 2024-06-11:
Thank you again for your help.
The problem turned out to be with the rs232->ttl converter. Apparently it failed with the display module.

closed 2024-06-11 by syssi