syssi/esphome-pipsolar #138: WKS evo cirlce

lhanneus, 2024-07-29:
Any success with the WKS evo circle ?
[image]

I tried the test-protocols.yaml without answers back...

syssi, 2024-07-30:
Do you know this issue? Please double check your wiring.

https://github.com/syssi/esphome-pipsolar/issues/68

lhanneus, 2024-07-30:
Thanks for your answer this gives me hope. I have already soldered 2 counterfeits max3232 chip . Now, I have a functioning max3232 , I have checked with a logic analyser. But the https://github.com/syssi/esphome-pipsolar/issues/68 seems to warn off with wiring. I will order other cables and adapters to check more.

closed 2024-08-12 by syssi