syssi/esphome-pipsolar #142: Eps8266 board NodeMCU V3 can working?

tieptute2k, 2024-08-12:
Your application can working with Eps8266 board NodeMCU V3 ?

syssi, 2024-08-12:
Yes.

tieptute2k, 2024-08-12:
How should I configure .yaml? Thank you

syssi, 2024-08-12:
Just take this one: https://github.com/syssi/esphome-pipsolar/blob/main/esp8266-example.yaml

tieptute2k, 2024-08-12:
esp8266:
  board: d1_mini

How should I configure for NodeMCU V3?

syssi, 2024-08-12:
`nodemcuv2` is probably a good choice but it's not very important. See https://esphome.io/components/esp8266.html#configuration-variables

> The PlatformIO board ID that should be used. Choose the appropriate board from [this list](https://registry.platformio.org/platforms/platformio/espressif8266/boards) (the icon next to the name can be used to copy the board ID). This only affects pin aliases, flash size and some internal settings, if unsure choose a generic board from Espressif such as esp01_1m.

tieptute2k, 2024-08-12:
Great! Thank you very much.....

closed 2024-08-12 by tieptute2k