syssi/esphome-pipsolar #144: More select option and intergal sensor

duymelody9x, 2024-08-22:
[image]
Hi Syssi, I can't add other customizations of select and intergal sensor.If I code like this, I can flash it to ESP but it doesn't work perfectly in HASS. Is there any way to add other customizations? I also want to add a KWh sensor.
```  - platform: pipsolar
    pipsolar_id: inverter0
    current_max_ac_charging_current:
      name: ${name} Setting_battery_float_charging_voltage_select
      optionsmap:
        "54.5V": "PBFT54.5"
        "54.6V": "PBFT54.6"
        "54.7V": "PBFT54.7"
        "54.8V": "PBFT54.8"
        "54.9V": "PBFT54.9"
        "55.0V": "PBFT55.0"
      statusmap:
        "54.5": "54.5V"
        "54.6": "54.6V"
        "54.7": "54.7V"
        "54.8": "54.8V"
        "54.9": "54.9V"
        "55.0": "55.0V"```

duymelody9x, 2024-08-27:
[image]
I tried renaming it the same as pip8048 but it still can't work

syssi, 2024-08-27:
This isn't supported by ESPHome. Unsupported entities / new feature must be implemented at the external component first before they are available/usable at the YAML configuration. You cannot introduce new select entities by extending the YAML only.

duymelody9x, 2024-08-27:
So is there any way to do that or do I still have to declare it with an existing variable like "current_max_ac_charging_current"?

syssi, 2024-08-27:
The `current_max_ac_charging_current` is a key wired to some code of the external component. I cannot recommend to abuse foreign keys. Could you describe the new feature in words and we add the new entity to the component without any workarounds?

duymelody9x, 2024-08-27:
so can you help me add the key
```battery_float_voltage``` - this key helps me set the float voltage mode of the inverter. its protocol command is ```PBFTnn.n```
``battery_redischarge_voltage``` - this key sets the voltage number the inverter will use the battery. protocol is ```PBDVnn.n```
``battery_recharge_voltage``` - this key sets the voltage number the inverter will use from the grid. protocol is ```PBCVnn.n```
thank you very much for your attention to my problems.

syssi, 2024-08-28:
```
3.17 PBFT<nn.n><cr>: Setting battery float charging voltage
Computer: PBFT <nn.n><CRC><cr>
Inverter: (ACK<CRC><cr> if Inverter accepts this command, otherwise, responds (NAK<CRC><cr>
nn.n: 48.0V ~ 58.4V for 48V unit
012,013 model: 48.0V~64.0V

3.9 PBDV<nn.n><cr>: Set battery re-discharge voltage
Computer: PBDV<nn.n><CRC><cr>
Inverter: (ACK<CRC><cr> if Inverter accepts this command, otherwise, responds
(NAK<CRC><cr>
12V unit: 00.0V12V/12.3V/12.5V/12.8V/13V/13.3V/13.5V/13.8V/14V/14.3V/14.5
24V unit: 00.0V/24V/24.5V/25V/25.5V/26V/26.5V/27V/27.5V/28V/28.5V/29V
48V unit: 00.0V48V/49V/50V/51V/52V/53V/54V/55V/56V/57V/58V
00.0V means battery is full(charging in float mode).

3.8 PBCV<nn.n><cr>: Set battery re-charge voltage
Computer: PBCV<nn.n><CRC><cr>
Inverter: (ACK<CRC><cr> if Inverter accepts this command, otherwise, responds
(NAK<CRC><cr>
12V unit: 11V/11.3V/11.5V/11.8V/12V/12.3V/12.5V/12.8V
24V unit: 22V/22.5V/23V/23.5V/24V/24.5V/25V/25.5V
48V unit: 44V/45V/46V/47V/48V/49V/50V/51V
```

syssi, 2024-08-28:
```
**QPIRI**

...

K KK.K Battery re-charge voltage K is an Integer ranging from 0 to 9.
The units is V

...

N LL.L Battery float voltage L is an Integer ranging from 0 to 9.
The units is V.

...

V VV.V Battery re-discharge voltage
V is an Integer ranging from 0 to 9.
The unit is V
```

duymelody9x, 2024-08-28:
Can I run yaml with those 3 keys now?

syssi, 2024-08-28:
No. I'm just collecting details for the implementation.

closed 2026-06-06 by syssi