syssi/esphome-pipsolar #149: 2 PV monitoring

fastvd, 2024-09-11:
Greetings. I have an inverter, which I finally control (switch operating modes) and read data, but I don't read everything I need...
yesterday i connected an extra string to my inverter and now i have 2 inputs from my solar panels...

only in this firmware modification did I find any mention of PV1 and PV2:
https://github.com/syssi/esphome-pipsolar/blob/pi18/esp32-example.yaml

but if we UNcomment out everything related to PV2, then these data still do not reach us...

P.S.: if you connect a wifi dongle and monitor the data in smartESS, my 2nd string PV2 shows the correct data, the ones that I physically see on the inverter display....that means the data can EXACTLY be obtained, you just need to understand how!

syssi, 2024-09-11:
Do you know the PIP protocol version of your inverter? Have you tried the `pip8048` branch already?

https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

fastvd, 2024-09-11:
[image]
PI30

fastvd, 2024-09-11:
> Do you know the PIP protocol version of your inverter? Have you tried the `pip8048` branch already?
> 
> https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

yes, I tried this version...unfortunately we don't get information from PV2...but we can see separate PV1 and total PV

[image]

syssi, 2024-09-11:
Are you able to identify the commands to retrieve the PV2 measurements?

fastvd, 2024-09-11:
no, I did not find the command yet, but I checked all the firmwares that are here and realized that they all show ONLY PV1

[image]
from mqtt
[image]
my pv1 in live
[image]
my pv2 in live

syssi, 2024-09-11:
If you use this configuration: https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

What does the `pv2_input_current` and `pv2_input_voltage` sensor show?

fastvd, 2024-09-11:
> If you use this configuration: https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml
> 
> What does the `pv2_input_current` and `pv2_input_voltage` sensor show?

does not show anything... there is simply no such sensor, or it is not transmitted to mqtt.. as an option with this firmware, enable advanced logging and catch pieces of code during this request

syssi, 2024-09-11:
I would be happy about a ESPHome log. :-)

fastvd, 2024-09-11:
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/inverter-modbus.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
INFO Starting log output from pipsolar/debug
INFO Connected to MQTT broker!
[13:34:35][D][pipsolar:863]: CRC OK
[13:34:35][D][uart_debug:158]: >>> "QMODI\xC1\r"
[13:34:35][D][pipsolar:926]: Sending polling command : QMOD with length 4
[13:34:36][D][uart_debug:158]: <<< "(B\xE7\xC9\r"
[13:34:36][D][pipsolar:860]: checking crc on incoming message
[13:34:36][D][pipsolar:863]: CRC OK
[13:34:36][D][pipsolar:536]: Decode QMOD
[13:34:36][D][text_sensor:064]: 'pipsolar device_mode': Sending state 'B'
[13:34:36][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[13:34:36][D][pipsolar:926]: Sending polling command : QFLAG with length 5
[13:34:37][D][uart_debug:158]: <<< "(EaxzlDbjkuvygd\x11\xD2\r"
[13:34:37][D][pipsolar:860]: checking crc on incoming message
[13:34:37][D][pipsolar:863]: CRC OK
[13:34:37][D][pipsolar:544]: Decode QFLAG
[13:34:37][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[13:34:37][D][pipsolar:926]: Sending polling command : QPIRI with length 5
[13:34:38][D][uart_debug:158]: <<< "(220.0 46.3 220.0 50.0 46.3 0200 0200 48.0 51.0 48.0 55.4 55.4 2 120 140 1 2 1 1 01 0 0 54.0 0 1 51.0 10 44.0NB\r"
[13:34:38][D][pipsolar:860]: checking crc on incoming message
[13:34:38][D][pipsolar:863]: CRC OK
[13:34:38][D][pipsolar:479]: Decode QPIRI
[13:34:38][D][pipsolar.select:031]: got value: 120
[13:34:38][D][pipsolar.select:033]: found mapped option 120A for option 120
[13:34:38][D][select:015]: 'pipsolar current_max_ac_charging_current_select': Sending state 120A (index 12)
[13:34:38][D][pipsolar.select:031]: got value: 140
[13:34:38][D][pipsolar.select:036]: could not find option 140 in mapping
[13:34:38][D][pipsolar.select:031]: got value: 2
[13:34:38][D][pipsolar.select:033]: found mapped option Solar Battery Utility for option 2
[13:34:38][D][select:015]: 'inverter0_output_source_priority_select': Sending state Solar Battery Utility (index 2)
[13:34:38][D][pipsolar.select:031]: got value: 1
[13:34:38][D][pipsolar.select:033]: found mapped option Solar first for option 1
[13:34:38][D][select:015]: 'pipsolar charger_source_priority_select': Sending state Solar first (index 0)
[13:34:39][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[13:34:39][D][pipsolar:926]: Sending polling command : QBATCD with length 6
[13:34:39][D][uart_debug:158]: <<< "(NAKss\r"
[13:34:39][D][pipsolar:860]: checking crc on incoming message
[13:34:39][D][pipsolar:863]: CRC OK
[13:34:39][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[13:34:39][D][pipsolar:926]: Sending polling command : QPIGS with length 5
[13:34:40][D][uart_debug:158]: <<< "(246.3 49.9 219.3 49.9 2324 2268 025 436 53.90 051 057 0047 09.1 368.7 00.00 00000 00010110 00 00 03312 010\x85\\\r"
[13:34:40][D][pipsolar:860]: checking crc on incoming message
[13:34:40][D][pipsolar:863]: CRC OK
[13:34:40][D][pipsolar:502]: Decode QPIGS
[13:34:40][D][sensor:094]: 'pipsolar grid_voltage': Sending state 246.30000 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar grid_frequency': Sending state 49.90000 Hz with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar ac_output_voltage': Sending state 219.30000 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar ac_output_frequency': Sending state 49.90000 Hz with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar ac_output_apparent_power': Sending state 2324.00000 VA with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar ac_output_active_power': Sending state 2268.00000 W with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar output_load_percent': Sending state 25.00000 % with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar bus_voltage': Sending state 436.00000 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar battery_voltage': Sending state 53.90000 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar battery_charging_current': Sending state 51.00000 A with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar battery_capacity_percent': Sending state 57.00000 % with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar inverter_heat_sink_temperature': Sending state 47.00000 °C with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar pv1_input_current': Sending state 9.10000 A with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar pv1_input_voltage': Sending state 368.70001 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[13:34:40][D][sensor:094]: 'pipsolar pv1_charging_power': Sending state 3312.00000 W with 1 decimals of accuracy
[13:34:40][W][component:237]: Component esphome.coroutine took a long time for an operation (154 ms).
[13:34:40][W][component:238]: Components should block for at most 30 ms.
[13:34:40][D][uart_debug:158]: >>> "QPIGS2h-\r"
[13:34:40][D][pipsolar:926]: Sending polling command : QPIGS2 with length 6
[13:34:41][D][uart_debug:158]: <<< "(NAKss\r"
[... 129 lines cut ...]
[13:34:53][W][component:237]: Component esphome.coroutine took a long time for an operation (184 ms).
[... cut ...]
[13:34:55][D][pipsolar:860]: checking crc on incoming message
[13:34:55][D][pipsolar:863]: CRC OK
[13:34:55][D][pipsolar:544]: Decode QFLAG
[13:34:56][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[13:34:56][D][pipsolar:926]: Sending polling command : QPIRI with length 5
[13:34:56][D][uart_debug:158]: <<< "(220.0 46.3 220.0 50.0 46.3 0200 0200 48.0 51.0 48.0 55.4 55.4 2 120 140 1 2 1 1 01 0 0 54.0 0 1 51.0 10 44.0NB\r"
[13:34:57][D][pipsolar:860]: checking crc on incoming message
[13:34:57][D][pipsolar:863]: CRC OK

fastvd, 2024-09-11:
[image]
I am now physically near the inverter and I can clearly see the data from PV1 and PV2 on the display...and what is in the logs...my number from PV2 is not in the answer from the inverter...it is logical to assume that the request is not correct.. .

fastvd, 2024-09-11:
[image]
if I even comment on everything related to PV1 and leave only PV2..then everything physically corresponds to PV1

fastvd, 2024-09-11:
more precisely, not so... if I leave only PV2, then there are no data on PV channels at all

binarybase, 2024-09-14:
Hello, I have simillar issue with EASUN inverter, it has too 2 PVs, but the command QPIGS2 goes into the timeout

[12:45:01][D][uart_debug:158]: >>> "QPIGS2h-\r"
[12:45:01][D][pipsolar:924]: Sending polling command : QPIGS2 with length 6
[12:45:01][W][component:237]: Component esphome.coroutine took a long time for an operation (51 ms).
[12:45:01][W][component:238]: Components should block for at most 30 ms.
[12:45:01][D][uart_debug:158]: <<< "(NAKss\r"
[12:45:01][D][pipsolar:860]: checking crc on incoming message
[12:45:01][D][pipsolar:863]: CRC OK

I was facing this issue on Axpert 8kW inverter, I was unable to get this command working when the inverter was connected via USB interface, but via RS232 I was able to get QPIGS2 response. 

I don't know how to solve it, but this command is longer than the others (QPIRI, QMOD, QPIGS - all of them has 4 or 5 characters + 2 chars CRC and carriage return? It is 8 characters. But QPIGS2 including CRC is 9 characters. Maybe something in a protocol is stripping string to 8 characters? It's just an idea.

Thank you

fastvd, 2024-09-15:
[image]
[image]
yes, I also found the answer as it should be, but unfortunately it is not there...in response we get - NAK ...but it can definitely be monitored somehow, as I remind you that the wifi dongle in the solar2mqtt application receives this data correctly.. .who has any ideas?

fastvd, 2024-09-16:
[image]
well...I found my old RPi3...I installed solar assistant there...I connected an adapter via usb-rs232, and what do you think...solar assistant also reads ONLY PV1!!! although I remind you that the wifi dongle reads correctly 2 channels! I don't know where to dig anymore...

binarybase, 2024-09-16:
I think it might be a protocol issue, maybe they have changed some commands. I also asked the seller for communication protocol for the inverter. My inverter has 2 MPPTs, but acts like version with one MPPT.

Have you tried to identify your protocol using QPI command? My inverter looks simillar to yours and responds me with PI30 protocol, but it should be PI30MAX, shouldn't be? That is really weird.

syssi, 2024-09-16:
@fastvd Could you provide a link to the WiFi dongle or some photos? How is the dongle attached to the inverter?

fastvd, 2024-09-16:
> @fastvd Could you provide a link to the WiFi dongle or some photos? How is the dongle attached to the inverter?

[image]
[image]
[image]
connects to RJ45 port, standard according to all instructions...
1,2-RX,TX
 4-12v
 8-GND

syssi, 2024-09-16:
@fastvd: Do you see the pin header (RX, TX, GND) on the WiFi dongle board? Could you measure the voltages here (RX<->GND, TX<->GND). Is the logic level 3.3V here? If yes, we could attach a ESP to sniff the traffic here. I can provide a YAML which prints the data of both lines (RX, TX) to the log.

If you don't want to measure the voltage you could also measure the continuity / resistance of 0 ohm between the pin header pins and the RX/TX pins of the MAX3232 chip.

binarybase, 2024-09-16:
Yes, you are having PI30 protocol likely as me, this protocol does not support the QPIGS2 command, so you are unable to get any information about the second PV input. I have also tried QPGS0 command - it also responds with NAK.

Maybe this dongle is using different protocol for readings? (maybe MODBUS?)

syssi, 2024-09-16:
Just a guess: The checksum of the QPIGS2 request isn't fine from the inverters perspective. In this case the inverter responds also with a `NAK`.

binarybase, 2024-09-16:
> Just a guess: The checksum of the QPIGS2 request isn't fine from the inverters perspective. In this case the inverter responds also with a `NAK`.

I thought that so, so I have tried also QPGS0 which has the same length as QPIGS, the CRC should be fine, but it responded with NAK also. I have to notice that this inverter has no parallel connection support.

fastvd, 2024-09-16:
> @fastvd: Do you see the pin header (RX, TX, GND) on the WiFi dongle board? Could you measure the voltages here (RX<->GND, TX<->GND). Is the logic level 3.3V here? If yes, we could attach a ESP to [...]

[image]

fastvd, 2024-09-16:
[image]

syssi, 2024-09-16:
3.17V looks good. I've pushed a UART sniffer example to the project: https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-uart-sniffer.yaml

Please flash this YAML to an ESP8266 and attach GPIO4, GPIO5 and GND to the dongle. Plug the dongle back to the inverter and take a look at the log of the ESPHome node. You should see the RAW traffic here (in hex).

fastvd, 2024-09-16:
if I measure on the pins of the wifi dongle, then 3.17V...
if on the esp board where solar2mqtt is filled, then the RS232 input is 12V, and on the esp board itself it is the same 3.3V, as on the native wifi dongle

syssi, 2024-09-16:
My question was only about the OEM WiFi dongle. My goal was to make sure we don't destroy the GPIOs of an ESP if we attach an 3.3V tolerant ESP here to sniff the traffic of the OEM dongle.

fastvd, 2024-09-16:
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/sniffer.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing uart-sniffer (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/uart-sniffer/src/main.cpp.o
Linking .pioenvs/uart-sniffer/firmware.elf
RAM:   [====      ]  37.9% (used 31076 bytes from 81920 bytes)
Flash: [===       ]  34.7% (used 362111 bytes from 1044464 bytes)
Building .pioenvs/uart-sniffer/firmware.bin
esp8266_copy_factory_bin([".pioenvs/uart-sniffer/firmware.bin"], [".pioenvs/uart-sniffer/firmware.elf"])
esp8266_copy_ota_bin([".pioenvs/uart-sniffer/firmware.bin"], [".pioenvs/uart-sniffer/firmware.elf"])
========================= [SUCCESS] Took 13.97 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.12.17.184
INFO Uploading /data/build/uart-sniffer/.pioenvs/uart-sniffer/firmware.bin (366256 bytes)
INFO Compressed to 259943 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 7.89 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 10.12.17.184 using esphome API
INFO Successfully connected to uart-sniffer @ 10.12.17.184 in 11.205s
INFO Successful handshake with uart-sniffer @ 10.12.17.184 in 0.023s
[16:25:49][I][app:100]: ESPHome version 2024.8.3 compiled on Sep 16 2024, 16:25:17
[16:25:49][C][wifi:600]: WiFi:
[16:25:49][C][wifi:428]:   Local MAC: 80:7D:3A:4E:8D:CC
[16:25:49][C][wifi:433]:   SSID: [redacted]
[16:25:49][C][wifi:436]:   IP Address: 10.12.17.184
[16:25:49][C][wifi:439]:   BSSID: [redacted]
[16:25:49][C][wifi:441]:   Hostname: 'uart-sniffer'
[16:25:49][C][wifi:443]:   Signal strength: -63 dB ▂▄▆█
[16:25:49][C][wifi:447]:   Channel: 7
[16:25:49][C][wifi:448]:   Subnet: 255.255.255.0
[16:25:49][C][wifi:449]:   Gateway: 10.12.17.1
[16:25:49][C][wifi:450]:   DNS1: 8.8.8.8
[16:25:49][C][wifi:451]:   DNS2: 0.0.0.0
[16:25:49][C][logger:185]: Logger:
[16:25:49][C][logger:186]:   Level: DEBUG
[16:25:49][C][logger:188]:   Log Baud Rate: 115200
[16:25:49][C][logger:189]:   Hardware UART: UART0
[16:25:49][C][uart.arduino_esp8266:118]: UART Bus:
[16:25:50][C][uart.arduino_esp8266:120]:   RX Pin: GPIO13
[16:25:50][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[16:25:50][C][uart.arduino_esp8266:124]:   Baud Rate: 2400 baud
[16:25:50][C][uart.arduino_esp8266:125]:   Data Bits: 8
[16:25:50][C][uart.arduino_esp8266:126]:   Parity: NONE
[16:25:50][C][uart.arduino_esp8266:127]:   Stop bits: 1
[16:25:50][C][uart.arduino_esp8266:131]:   Using software serial
[16:25:50][C][uart.arduino_esp8266:118]: UART Bus:
[16:25:50][C][uart.arduino_esp8266:120]:   RX Pin: GPIO12
[16:25:50][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[16:25:50][C][uart.arduino_esp8266:124]:   Baud Rate: 2400 baud
[16:25:50][C][uart.arduino_esp8266:125]:   Data Bits: 8
[16:25:50][C][uart.arduino_esp8266:126]:   Parity: NONE
[16:25:50][C][uart.arduino_esp8266:127]:   Stop bits: 1
[16:25:50][C][uart.arduino_esp8266:131]:   Using software serial
[16:25:50][C][mdns:116]: mDNS:
[16:25:50][C][mdns:117]:   Hostname: uart-sniffer
[16:25:50][C][esphome.ota:073]: Over-The-Air updates:
[16:25:50][C][esphome.ota:074]:   Address: uart-sniffer.local:8266
[16:25:50][C][esphome.ota:075]:   Version: 2
[16:25:50][C][esphome.ota:078]:   Password configured
[16:25:50][C][safe_mode:018]: Safe Mode:
[... 56 lines cut ...]
[16:26:16][W][component:238]: Components should block for at most 30 ms.
[16:26:18][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[16:26:18][W][component:237]: Component uart took a long time for an operation (51 ms).
[16:26:18][W][component:238]: Components should block for at most 30 ms.
[16:26:19][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[16:26:19][W][component:237]: Component uart took a long time for an operation (51 ms).
[16:26:19][W][component:238]: Components should block for at most 30 ms.
[16:26:21][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:0 [... cut ...]
[16:26:29][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[16:26:29][W][component:237]: Component uart took a long time for an operation (52 ms).
[16:26:29][W][component:238]: Components should block for at most 30 ms.

fastvd, 2024-09-16:
substitutions:
  name: uart-sniffer
  rx_pin: GPIO13
  rx_pin2: GPIO12

esphome:
  name: ${name}
  min_version: 2024.6.0

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome
  password: "59e4413735b419182834a912a57fef57"

logger:
  level: DEBUG

api:
  reboot_timeout: 0s

uart:
  - id: uart_0
    baud_rate: 2400
    rx_pin: ${rx_pin}
    debug:
      direction: RX
      dummy_receiver: true
      sequence:
        - lambda: UARTDebug::log_hex(esphome::uart::UARTDirection::UART_DIRECTION_RX, bytes, ':');

  - id: uart_1
    baud_rate: 2400
    rx_pin: ${rx_pin2}
    debug:
      direction: RX
      dummy_receiver: true
      sequence:
        - lambda: UARTDebug::log_hex(esphome::uart::UARTDirection::UART_DIRECTION_TX, bytes, ':');

fastvd, 2024-09-16:
rx_pin: GPIO13
rx_pin2: GPIO12

correctly configured?
they are physically soldered to D6 and D7

fastvd, 2024-09-16:
I took the d1 mini board from the solar2mqtt project and put the firmware under the sniffer so as not to overwrite my working esp32...but I have never used it here before, because of esphome...that's why I asked about GPIO...but I see that something not so, so I will pour the sniffer firmware into a 100% working esp32

fastvd, 2024-09-16:
[image]

I put it on my 100% working board...but I don't see any log in the terminal...that's how it should be and will it be physically saved somewhere?

syssi, 2024-09-16:
You are too fast. A ESP8266 is fine, the assigned GPIOs looks good too. Just the captured data doesn't look good. A photo could help to identify issues like a missing GND line or additional attached components etc.

fastvd, 2024-09-16:
and why are there no logs on esp32

fastvd, 2024-09-16:
[image]
something is wrong...this config does not work either on w1 mini or esp32...both the boards and the cable themselves are 100% working here and there...esp32 I have your config running under pipsolar and solar2mqtt works on w1 mini ...I just change the firmware..but I don't have a log either there or there...

syssi, 2024-09-17:
Could you provide a photo of your setup/wiring and the used YAML configuration? I want to make sure everything is connected/wired/configured properly.

fastvd, 2024-09-17:
well...i put the old config back and now it's a 100% working esp32 board + github://syssi/esphome-pipsolar@pip8048

[image]
[image]

I don't take a screenshot of the end of the code, because everything is standard there...
now I take and leave the cap unchanged (except for replacing the API with mqtt) and only change the UART parameters
[image]
[image]

and this is already a screenshot of my new configs based on 100% verified Rx and TX parameters

fastvd, 2024-09-17:
[image]
code verification is successful... I'm uploading the config via Wi-Fi
[image]

fastvd, 2024-09-17:
[image]

fastvd, 2024-09-17:
[image]

fastvd, 2024-09-17:
[image]
here is step by step what I did! and I get 0 at the output

syssi, 2024-09-17:
Could you make a photo of the ESP connected to the WiFi dongle and plug the dongle into the Inverter? Are you able to receive data using the OEM app? Do you see the raw traffic at the ESPHome log too?

fastvd, 2024-09-17:
Not sure what you want? 1) take a photo of the esp32 board that is connected to the inverter with this code?
2) should I take a picture of the native wifi dongle???

I don't understand something about the idea...

fastvd, 2024-09-17:
I probably did not understand you correctly in the physical implementation of the sniffer itself... if I understood everything correctly, then I need to connect my esp32 RX to the RX of the native wifi dongle and TX to TX? and as if to parallelize 2 boards.. so that the native dongle does its work, and our sniffer is in parallel listening?

syssi, 2024-09-17:
Correct. First of all you have to make sure your OEM WiFi dongle is working general. If I remember correctly the dongle is able to retrieve measurements of PV2 and able to show the values somethere, correct?

1. Plug the WiFi dongle to the inverter
2. Start the App of the manufacturer and make sure you are able to see PV2 measurements here
3. Use a plain ESP32 (without attached anything else!!) and attach 3 wires to `GPIO16`, `GPIO17` and `GND`
4. Connect these three wires to the pin header of the OEM dongle (`TX` to `GPIO16`, `RX` to `GPIO17` and `GND` to `GND`)
5. Are you still able to receive data using the OEM App (as in step 2)?
6. Power the ESP and take a look at the ESPHome log. Do you see valid traffic this time?

Please don't post screenshots! The logs in plaintext are sufficient.

fastvd, 2024-09-17:
[image]
[image]

fastvd, 2024-09-17:
[image]
but now I have 23.09... so the data from PV1 and PV2 is 0

fastvd, 2024-09-17:
[image]

fastvd, 2024-09-17:
[image]

fastvd, 2024-09-17:
[image]

at this time, when the esp32 board with the sniffer is connected in parallel, the native wifi dongle still transmits the correct data to the cloud... for the test, so that the difference was visible, I turned on the additional load at home, and yes, after 4-5 minutes it was displayed in smartESS application
[image]
in these 4-5 minutes I saw such a picture in the logs

fastvd, 2024-09-17:
[23:11:05][C][safe_mode:021]:   Invoke after 10 boot attempts
[23:11:05][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[23:11:05][C][api:139]: API Server:
[23:11:05][C][api:140]:   Address: 10.12.17.172:6053
[23:11:05][C][api:144]:   Using noise encryption: NO
[23:11:17][D][uart_debug:114]: >>> 10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:FC:10:88:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:28:21:3D:AF:42:E5:10:8C:4A
[23:11:53][D][uart_debug:114]: >>> 10:8C:4A:10:8C:4A:10:8C:4A:42:8C:08:10:00:EF:F9:8C:4A:42:8C:08:10:8C:4A:10:C2:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:10:8C:4A:3A:09:A7:FE:10:88:4E:C6:CE:C2:A1:25:FF:10:8C:4A:FF:9C:8C:4A:10:8C:4A:3A:F8:9C:C6:CE:10:8C:4A:3A:09:A7:4A:20:EF:12:20:EF:FE:10:88:10:8C:4A:FF:A1:C6:08:C2:A1:25:FF:F9:24:8C:10:8C:CE:A1:C6:CE:9C:8C:4A:FF:10:8C:4A:4A:03:8C:9C:0B:F9:3B:42
[23:12:00][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[23:12:00][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:12:00][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[23:12:31][D][uart_debug:114]: >>> E5:C2:E2:31:1E:20:CB:38:09:3D:F8:8C:4A:4A:20:EC:C2:A1:25:4B:8C:08:A1:C6:CE:10:8C:4A:4A:03:25:4A:43:25:F9:24:A7:A5:9D:25:38:64:10:42:62:31:9C:8C:4A:A1:82:E7:10:8C:25:4A:20:EF:99:8C:4A:FF:38:09:A7:7B:46:A5:94:38:C4:38:24:10:10:8C:4A:FB:8C:4A:A5:88:C4:9C:8C:1B:FF:10:2A:4A:4A:20:EF:28:37:09:38:09:A7:4A:20:EF:10:8C:4A:4A:20:EF:FE:29:F9:3A:09:A5:10:8C:4A:12:03:EF
[23:13:06][D][uart_debug:114]: >>> 10:8C:4A:9C:8C:4A:10:2B:23:FF:10:2A:4A:FF:C2:E2:31:10:8C:4A:FF:10:8C:4A:FF:79:C6:CE:A5:A1:04:C2:A1:25:FF:FB:8C:23:FF:FE:88:04:C2:8C:4A:53:18:94:FF:FC:10:2B:4A:20:EF:38:09:3D:FF:A3:82:E7:10:8C:4A:0A:20:EF:FC:10:BE:42:8C:08:F9:8C:4A:10:8C:4A:9C:C2:4A:22:A1:25:FF:9D:8C:4A:F9:C6:08:EB:27:E1:10:8C:4A:FF:10:8C:08:E2:13:F2:10:C2:1B:FF:10:8C:4A:6F:98:FD:39:A5:A1:FB
[23:13:33][D][uart_debug:114]: >>> 8C:23:FF:A5:D7:EF:4E:82:E7:F9:8C:4A:F9:8C:13:FF:28:3E:09:F9:8C:4A:4A:03:25:4A:20:EF:FB:8C:23:EF:FE:10:04:4A:20:EF:C2:A1:25:FF:4E:82:E7:1B:A1:E1:8C:4A:FF:00:84:42:22:25:A5:29:A5:A1:C2:59:F8:DB:A1:80:09:99:00:42:49:08:F9:39:29:CA:09:E8:A5:42:25:73:29:F8:FF:B8:E3:10:2B:4A:FF:02:A1:25:FF:94:38:E3:9C:8C:4A:FF:FE:10:88:10:8C:25:4A:98:EF:4E:E1:08:42:8C:08:3C:A1:25
[23:14:09][D][uart_debug:114]: >>> 32:A1:88:FE:10:88:42:8C:84:FB:8C:4A:FF:A5:DD:EF:FD:18:BB:FE:88:89:4E:8C:A6:FC:10:88:FB:8C:4A:FF:10:2B:4A:10:8C:4A:FF:C2:A1:25:FF:10:8C:4A:FF:1D:20:EF:09:F2:A1:4C:E1:28:C6:08:FC:10:4E:A1:36:09:EB:A6:C4:28:19:A3:51:18:C2:FF:1D:03:EF:42:8C:08:FC:10:88:61:08:F9:28:29:F8:9C:8C:4A:C2:A1:25:FF:A1:C6:0C:1E:20:86:42:8C:08:C2:8C:4A:FF:FB:8C:4A:FF:29:88:C4:BD:88:04:38
[23:14:46][D][uart_debug:114]: >>> 09:A7:4A:A1:8E:4B:D6:88:4E:E1:0C:28:C6:25:42:8C:08:10:C2:4A:28:E3:A3:10:8C:4A:EB:38:E3:10:8C:23:FF:10:8C:CE:A1:82:CF:C7:4E:A7:A1:82:C7:94:38:C4:10:2B:4A:FF:FF:88:89:FE:10:88:28:4E:09:10:8C:4A:FF:F8:8C:25:12:20:EF:C6:62:31:C6:62:31:A5:D7:EF:B5:D7:EF:28:00:FA:38:09:32:C2:E2:31:F2:09:A7:4A:98:EF:10:8C:CE:FC:10:88:A5:42:EF:8C:46:F5:21:EF:C4:F2:B3:20:62:98:EF:4B
[23:15:24][D][uart_debug:114]: >>> B5:08:10:2B:4A:C2:62:31:A5:89:04:42:8C:08:FC:10:88:94:38:C4:42:8C:08:F2:B3:20:4E:C6:2A:10:22:4A:FF:10:00:C6:10:8C:03:FF:4A:00:EC:1E:00:EF:4A:00:EF:4A:00:EF:10:8C:4A:28:C6:A5:E2:43:25:4A:00:EF:42:8C:08:10:8C:4A:F9:24:A7:3A:09:A7:FE:10:04:9C:21:7D:01:29:CA:9C:82:E7:08:98:EF:10:8C:13:FF:94:66:C4:1F:00:EF:A1:09:A3:08:98:EF:42:B5:A5:1F:00:86:F2:48:20:4E:8C:86:4E
[23:16:00][D][uart_debug:114]: >>> C6:CE:42:8C:A5:10:8C:4A:FF:F9:8C:4A:FF:4B:4A:08:10:2B:4A:38:09:8C:D1:18:94:3A:09:A7:46:2F:C4:28:88:EF:9C:8C:4A:10:8C:4A:51:18:1D:38:09:A7:10:00:EC:10:C2:4A:10:8C:23:FF:F9:2A:4A:9C:E1:08:42:8C:5D:10:8C:4A:FF:C2:A6:25:FF:C2:A1:25:4A:00:EF:02:29:1E:29:A5:F8:10:8C:4A:FF:4A:A1:EF:10:C2:4A:10:8C:CE:C2:A1:25:7B:C6:84:10:00:EF:4A:00:CE:C2:A1:25:10:8C:4A:FF:18:C6:10
[23:16:16][D][uart_debug:114]: >>> 8C:CE:F9:64:A7:0B:8C:A5:53:18:94:FF:4B:8C:61:10:00:C9:08:98:EF:42:4A:08:4E:E3:A3:4E:E3:86:28:4E:09:4A:98:0E:10:00:EF:F2:A1:EF:FC:8C:4A:FF:08:98:AE:10:8C:4A
[23:16:17][D][uart_debug:114]: >>> 10:8C:4A
[23:16:55][D][uart_debug:114]: >>> 51:18:82:FF:08:98:EC:A0:00:EF:4A:00:EF:3A:09:8C:3A:09:A7:9C:8C:25:F2:A8:79:00:A5:F9:10:8C:4A:FF:FC:10:88:28:F5:A3:A1:82:E7:C2:A1:25:FF:FB:C2:CE:10:8C:4A:3A:09:A7:C6:62:4A:EB:38:C4:2A:C6:0C:18:00:EC:F2:98:EF:9C:8C:4A:10:C2:08:0A:B3:EC:42:B5:08:4A:98:EF:10:00:EF:4A:00:EF:38:09:3D:FC:10:04:1F:03:25:53:18:84:A1:D8:86:10:8C:1B:FF:F2:98:EF:4B:B5:08:08:4E:A7:10:C2
[23:17:31][D][uart_debug:114]: >>> 9D:2D:62:31:FF:6A:94:53:6A:94:FF:02:43:25:FF:88:2B:05:A5:7E:4E:AD:E9:10:8C:4A:FF:10:8C:4A:FF:3A:09:A7:F2:B3:20:38:09:8C:10:8C:4A:10:8C:4A:10:8C:4A:42:B5:61:4A:00:EF:FB:8C:4A:FF:28:C6:CE:1F:00:EF:4A:00:EC:4A:00:EF:42:B5:61:10:8C:4A:10:8C:4A:C2:E2:4A:28:09:86:10:8C:4A:F2:00:EF:4A:00:0E:2D:62:31:FF:FC:10:88:C2:A1:25:FF:38:09:A7:FC:10:88:4B:8C:A5:EB:38:C4:F9:2B
[23:18:09][D][uart_debug:114]: >>> 23:FF:28:82:E7:FC:10:2B:D3:18:94:4A:80:E2:63:AC:FC:58:08:FE:38:A5:E8:FC:10:88:4A:98:EF:23:EF:C2:28:8C:A3:3A:09:8C:FC:42:88:1E:28:04:C2:A1:25:FF:1F:00:EF:42:8C:08:4A:00:EC:FC:10:88:F2:95:CE:10:42:E7:46:38:E2:61:88:04:FB:2B:4A:38:09:A7:42:8C:08:F2:C8:10:51:18:C2:FF:A0:98:EF:94:66:C4:94:66:C4:4B:46:25:1F:80:E6:C2:E2:31:28:C6:25:C2:8C:4A:FE:10:C4:A5:F7:EF:4B:8C
[23:18:35][D][uart_debug:114]: >>> A5:B8:F9:9C:8C:4A:FF:B3:A1:C4:10:8C:23:FF:A1:C6:0C:61:08:F9:00:A5:F9:4A:98:EF:FE:88:04:38:09:9C:FB:8C:4A:1F:20:EF:F9:C6:CE:42:8C:08:A5:9D:25:05:40:1A:9C:A9:5C:08:0A:00:A5:AD: [... cut ...] :10:8C:4A:28:38:C4:10:C2:4A:4A:03:EF:10:8C:4A:FC:10:88:10:8C:4A:38:09:A7:FB:8C:4A:FF:46:10:F8:38:09:A7:C2
[23:22:17][D][uart_debug:114]: >>> E1:25:FF:10:8C:4A:FF:10:8C:4A:FE:10:88:10:8C:4A:A1:82:E7:38:09:A7:10:2B:4A:42:8C:08:02:00:CE:10:8C:23:FF:4B:8C:08:10:8C:4A:4A:98:EF:F9:8C:4A:A7:88:C4:4A:80:EF:38:09:A7:10:2B:CE:10:8C:4A:10:8C:4A:42:8C:08:9C:08:F9:C2:29:EB:42:8C:08:10:8C:4A:FC:10:88:10:8C:23:FF:4E:C6:CE:F2:48:20:C2:E2:31:4A:00:EF:FC:10:88:7B:39:A7:42:8C:08:EB:38:C4:10:AB:4A:FF:10:8C:4A:10:8C

fastvd, 2024-09-17:
for the test, to exclude RX and TX confusion, I replaced 
  tx_pin: GPIO17
  rx_pin: GPIO16

the picture has changed only in the direction of the arrow

all the time I received up-to-date data in smartESS

[image]

syssi, 2024-09-18:
The received data doesn't look like the PIP protocol and it's in one direction only. I guess this is the log output of the WiFi dongle and cannot be decoded because 2400 baud wasn't the correct baudrate.

Let's try to get an better idea of the schematics of the dongle. This is the pinout of the max3232 on the back of the board:

[image]

We are interested in `DIN1`, `ROUT1` (and if these aren't in use `DIN2` and `ROUT2`):

[image]

Are these pads accessible/routed to the pin header? Do you find any other easily accessible pads to connect to `DIN1` and `ROUT1`? I guess these pin of the MAX3232 aren't connected to the TX/RX pins at we are sniffing at the moment.

syssi, 2024-09-18:
Please attach GPIO16, GPIO17 and GND to this second 5 pin header (RX, TX, GND) and repeat the experiment of sniffing the SmartESS traffic.

fastvd, 2024-09-18:
I soldered 4 wires to 9-10-11-12 pins so that I can change them on the esp32 side... tell me the specific pinout where I should connect my TX(GPIO16) and RX(GPIO17) on the dongle side
[image]

syssi, 2024-09-18:
This doesn't look good. Are you sure the yellow and white cable aren't connected to the same pin now?

fastvd, 2024-09-18:
no, everything is ok... I checked with a multimeter for a short circuit between all the soldered taps

syssi, 2024-09-18:
I guess this should be the same:

[image]

Long story short: We want to capture the data of `DIN1` and `ROUT1` of the MAX3232. The order of the two GPIOs (GPIO16, GPIO17) isn't important. The difference is just the prefix (`<<<` vs. `>>>`) in front of the log messages at the moment.

fastvd, 2024-09-18:
I understand what you want...but the RX contact does not connect to the leg of the microcircuit (according to the idea of ​​12 pins), but the TX connected to pin 11...that's why I soldered it directly to 9-10-11-12.. .and I already managed to check...not a single pair gives data to our sniffer((((now I will try your option

syssi, 2024-09-18:
Does the SmartESS Cloud still shows valid data if the ESP is attached to DIN1 and ROUT1?

[19 more replies]

closed 2024-09-19 by syssi