syssi/esphome-pipsolar #151: Voltronic Axper IV Twin 6K new esp8266-example.yaml

Alarico77, 2024-09-30:
Hi, I'm trying to install the new esp8266-example.yaml, with the old syssi.esphome-pipsolar@main I read the inverter data (not all, but it's ok). When I install the .yaml file, it gives me some timeout errors, and in the dashboard all the sensors are unknown. I attach a log:

syssi, 2024-09-30:
Could you provide the log with debug output of the `uart` component enabled (see https://github.com/syssi/esphome-pipsolar?tab=readme-ov-file#debugging)? I would like to see the raw traffic of your inverter.

Alarico77, 2024-09-30:
INFO ESPHome 2024.9.1
INFO Reading configuration /config/esphome/voltronic-axpert.yaml...
WARNING Found '${tx_pin}' (see uart->tx_pin) which looks like a substitution, but 'tx_pin' was not declared
WARNING Found '${rx_pin}' (see uart->rx_pin) which looks like a substitution, but 'rx_pin' was not declared
WARNING Found '${tx_pin}' (see uart->tx_pin) which looks like a substitution, but 'tx_pin' was not declared
WARNING Found '${rx_pin}' (see uart->rx_pin) which looks like a substitution, but 'rx_pin' was not declared
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
Failed config

uart: [source /config/esphome/voltronic-axpert.yaml:48]
  - id: uart_0
    baud_rate: 2400
    
    Cannot resolve pin name '${tx_pin}' for board d1_mini.
    tx_pin: ${tx_pin}
    
    Cannot resolve pin name '${rx_pin}' for board d1_mini.
    rx_pin: ${rx_pin}
    debug: 
      direction: BOTH
      dummy_receiver: False
      after: 
        delimiter: 
      sequence:

Alarico77, 2024-09-30:
attach the log

Alarico77, 2024-09-30:
can't you see it????

[logs_voltronic-axpert_run (4).txt](https://github.com/user-attachments/files/17196484/logs_voltronic-axpert_run.4.txt)

syssi, 2024-10-01:
It looks like your inverter doesn't support the `QPIGS2` and `QBATCD` command. Do you miss measurements of the second PV string?

Alarico77, 2024-10-01:
I have charge regulators, so I have not connected PV strings to the inverter.
Instead, the commands to send to the inverter, you are right, they do not work. I tried to set "Current_max_ac_charging_current_select" from 20A to 10A, but it always remains at 20.

syssi, 2024-10-01:
For successful commands the inverter responds with `ACK`nowledge. If a command isn't accepted the response is `NAK` (not acknowledged).

Alarico77, 2024-10-01:
OK some do, some don't.
it's a bit unstable, in the sense that the connection drops often.

Alarico77, 2024-10-01:
I'd better check my wifi network first. ;-)

Alarico77, 2024-10-01:
so: wifi seems to be fine.
the component and its entities, stay online for a few minutes, then everything goes offline and the entities become: unavailable, then after a few minutes everything reactivates.