syssi/esphome-pipsolar #153: How to change Bulk and Float values

gptech2444, 2024-10-05:
Hi,
Is it possible to change the float and bulk battery charging values from HA?

carmelo42, 2024-12-12:
I'm looking for that too.

jszbp, 2025-03-02:
According to:
https://esphome.io/components/pipsolar.html
there are outputs and and an action: output.pipsolar.set_level
Action can be used in automation like this:
on_...:
  then:
  - output.pipsolar.set_level:
      id: my_pipsolar_output
      value: 48.0
I've got ISOLAR SML II 5.5KW (powland/Easun) inverter.
I flashed wifi dongle with esphome firmware and I can read in HA values from the inverter. I can also change from inside the HA "output source priority" of the inverter via three switch entities, each one for: "battery", "solar" and "utility".
I tried to link an input_number entity in my HA with my inverters Float Voltage by adding the following to the yaml configuration of esphome firmware of the dongle interface:
```
number:
  - platform: homeassistant
    id: ha_float_voltage
    entity_id: input_number.powland_float_voltage
    on_value:
      then:
        - output.pipsolar.set_level:
            id: battery_float_voltage
            value: !lambda "return id(ha_float_voltage).state;"
```
where: "input_number.powland_float_voltage" is name of the input_number entity in my HA and "battery_float_voltage" is name of the output of pipsolar/inverter. But nothing happens.

After then I added a sensor to debug above number entity:
`
sensor:
  - platform: template
    id: debug_float_voltage
    name: "debug float voltage"
number:
  - platform: homeassistant
    id: ha_float_voltage
    entity_id: input_number.powland_float_voltage
    on_value:
      then:
        - output.pipsolar.set_level:
            id: battery_float_voltage
            value: !lambda "return id(ha_float_voltage).state;"
        - sensor.template.publish:
            id: debug_float_voltage
            state: !lambda "return id(ha_float_voltage).state;"
`
and in HA I can see that "debug float voltage" correctly reflects "input_number.powland_float_voltage" values - but still Fload Voltage of the inverter does not change.

jszbp, 2025-03-02:
OK, I figured out how to setup Float Voltage. One can achieve that by using **output.pipsolar.set_level** action like in my example:
```
number:
  - platform: homeassistant
    id: ha_float_voltage
    entity_id: input_number.powland_float_voltage
    on_value:
      then:
        - output.pipsolar.set_level:
            id: battery_float_voltage
            value: !lambda "return id(ha_float_voltage).state;"
```

where: "input_number.powland_float_voltage" is name of the input_number entity in my HA and "battery_float_voltage" is name of the output of pipsolar/inverter

**BUT!**:
1. value set MUST BE in the allowed values, default for "battery_float_voltage" are: 48.0,49.0,50.0,51.0
2. value must be <= "battery_bulk_voltage" - you can't set it higher
At least that how it is for my inverter ISOLAR SML II 5.5KW. Float Voltage range for my inverter is 48.0 to 61.0 at increment 0.1

So, if one want's to be able to set Float Voltage to e.g 54.4V, this value must be added to possible_values list:
```
output:
  - platform: pipsolar
    pipsolar_id: inverter0  
    battery_float_voltage:
      id: battery_float_voltage
      possible_values: [50.0,50.5,51.0,52.0,53.0,54.2,54.3,54.4]
```
then Float Voltage can be set to the exact values in [] above, assuming these values are not greater than Battery Bulk Voltage set in inverter.
I believe Bulk voltage could be set via sending PCVVxx.x command directly by uart.