syssi/esphome-pipsolar #155: Data flow stops ESP32

gptech2444, 2024-10-16:
Hi,
Around 12.30am last night I noticed interruptions to the data reaching HA. Around every 5-10mins the ESP32 resets.
Everything has been working great for the last 2 weeks since setting this up. Nothing has changed, I'm not sure why it now has this problem.
Attached is the log.
[logs_sumry10kw_logs.txt](https://github.com/user-attachments/files/17387951/logs_sumry10kw_logs.txt)

syssi, 2024-10-16:
The simplest solution to get an idea what's going on is attaching a PC to the serial line (GPIO1/GPIO3 used by the `logger`) of the ESP and record a log including the log messages of the bootloader + exceptions/stack traces of the framework.

Please be aware of some `reboot_timeout`s. One `reboot_timeout` of `15min` is introduced by the `mqtt` component (See https://esphome.io/components/mqtt.html#configuration-variables -> `reboot_timeout`). If there is no active MQTT connection for 15 minutes the `mqtt` component triggers a reboot.

The same applies for the `wifi` component. If the ESP isn't associated to an accesspoint for 15 minutes the component triggers a reboot: https://esphome.io/components/wifi.html#configuration-variables -> `reboot_timeout`

This also applies for the `api` component if there is no native `api` connection. Your log looks like you aren't using the `api` component.

gptech2444, 2024-10-16:
Ok, I'm not using the api as it isn't included in the pip8048 yaml and I want to be able to change the inverters settings from HA.
The issue went away for a couple of hours and then it started again and was resetting itself 3 more times and then it's been ok for the last 20 hours. I've not touched or changed anything. 
This message is the one that stood out on the resets.
[10:01:31][C][mdns:116]: mDNS:
[10:01:31][C][mdns:117]:   Hostname: sumry10kw
[10:01:31][C][esphome.ota:073]: Over-The-Air updates:
[10:01:31][C][esphome.ota:074]:   Address: sumry10kw.local:3232
[10:01:31][C][esphome.ota:075]:   Version: 2
[10:01:31][C][safe_mode:018]: Safe Mode:
[10:01:31][C][safe_mode:020]:   Boot considered successful after 60 seconds
[10:01:31][C][safe_mode:021]:   Invoke after 10 boot attempts
[10:01:31][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[10:01:31][W][safe_mode:029]: Last reset occurred too quickly; safe mode will be invoked in 6 restarts

syssi, 2024-10-17:
You have to attach another device to the serial line of the ESP to get a better idea what's going on. The important information's cannot be delivered via MQTT. The message from the `safe_mode` handler sounds like hard crashes. The explicit exceptions can be captured on the serial line only!

gptech2444, 2024-11-10:
I was using the usb port on my HA pc for the powersupply to the esp32 and I think there may have been a ground loop interference happening, I've since gone to a wallplug usb power supply and the problem has gone away.
Strange how it worked fine for 2 weeks powered by the pc before any problems occured.

syssi, 2024-11-10:
Alright. Lets close this issue! Thanks for your feedback!

closed 2024-11-10 by syssi