syssi/esphome-pipsolar #158: PV Monitoring

vpritulin, 2024-11-04:
Hi, I have daxtromn with 2 PV. I tried to use this configuration https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml but I get an error. Pipsolar does not have information from PV2. I tried to find the address for PV2, but without success

[image]

syssi, 2024-11-04:
Did you try the `pip8048` branch? It supports two strings.

vpritulin, 2024-11-04:
@syssi yes, but it didn't help 
my yaml file [yaml.txt](https://github.com/user-attachments/files/17616158/yaml.txt)
 
[image]

syssi, 2024-11-04:
The mentioned `select` section is missing. You have to add a single `select` entity at least:

```
select:
  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_output_source_priority_select
      name: inverter0_output_source_priority_select
      optionsmap:
        "Utility first": "POP00"
        "Solar only": "POP01"
        "Solar Battery Utility": "POP02"
      statusmap:
        "0": "Utility first"
        "1": "Solar only"
        "2": "Solar Battery Utility"
```

See https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

vpritulin, 2024-11-04:
I don't get data from pv2 it's displayed as unknown

<<< "(222.5 50.0 230.1 49.9 0437 0386 004 448 56.50 000 097 0049 01.4 348.9 00.00 00000 00010110 00 00 00509 110\x9FH\r"

syssi, 2024-11-04:
Could you provide some more lines of the log? The single line without context doesn't help. Does your inverter respond to the `QPIGS2` request?

vpritulin, 2024-11-04:
I have attached the full log.
[logs_ibconnect_logs.txt](https://github.com/user-attachments/files/17617656/logs_ibconnect_logs.txt)

syssi, 2024-11-04:
Your inverter rejects the `QPIGS2` request:

```
[13:46:31][D][uart_debug:158]: >>> "QPIGS2h-\r"
[13:46:31][D][uart_debug:158]: <<< "(NAKss\r"
```

`QPIGS2` is the command to retrieve the measurements of the second PV string. Did you check the other issues of this github project? There are more people with the same problem.

vpritulin, 2024-11-04:
yes, I checked this issue https://github.com/syssi/esphome-pipsolar/issues/149. But as I understood, the person connected the module to the ESP and receives data from it

syssi, 2024-11-04:
At the end the inverter was interfaced using Modbus:

https://github.com/syssi/esphome-pipsolar/issues/149#issuecomment-2359224216
https://github.com/odya/esphome-powmr-hybrid-inverter/issues/31

vpritulin, 2024-11-05:
@syssi Thanks for your help. Another repository helped me. Closing this issue

closed 2024-11-05 by vpritulin

syssi, 2024-11-05:
Could you provide the solution here / point to the repo?