syssi/esphome-pipsolar #159: Charging suerce priority Axpert

Pokornypetr, 2024-11-07:
Hello, please help to create a switch or something to control the battery charging source for the axpert inverter. I need, for example, automation in HA to switch between recharging from the distributor's network in a low tariff and pure solar energy. at the same time, I would like to be able to change the "Back to Discharge Voltage" setting from full to 26v as it is possible in the Watch power application. thank you very much for your help. I tried to edit the code and try your changes from the last time I used it from you, but I can't add the SELECT section: it says Platform not found: 'select.pipsolar'.`esphome:
  name: "prevodnik-axpert"

esp8266:
  board: d1_mini

# Enable logging
logger:
  level: DEBUG
  # Don't write log messages to UART0 (GPIO1/GPIO3) if the inverter is connected to GPIO1/GPIO3
  baud_rate: 0

uart:
  id: Axpert
  baud_rate: 2400
  tx_pin: GPIO4
  rx_pin: GPIO5
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

# Enable Home Assistant API
api:

ota:
  - platform: esphome
    id: my_ota
    password: "petolinet"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pevodnik-axpert"
    password: ""
# Optional manual IP
  manual_ip:
    static_ip: 192.168.100.30
    gateway: 192.168.100.1
    subnet: 255.255.255.0

time:
  - platform: homeassistant
    id: homeassistant_time
  - platform: sntp
    id: my_time    

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

captive_portal:
  # Example configuration entry
#uart:
#  - id: Axpert
#    tx_pin: GPIO4
#    rx_pin: GPIO5
    # most devices use 2400 as baud_rate
 #   baud_rate: 2400
    
substitutions:
  friendly_name: "FVE Axpert"
  inv_id: "inverter0"
[... 266 lines cut ...]
        "100A": "MCHGC100"
        "110A": "MCHGC110"
        "120A": "MCHGC120"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
        "70": "70A"
        "80": "80A"
        "90": "90A"
        "100": "100A"
        "110": "110A"
        "120": "120A"
pipsolar:
  - uart_id: Axpert
    id: inverter0
    update_interval: 10s
 

# Example configuration entry
    
`

closed 2025-02-24 by Pokornypetr

VirtualFab, 2025-03-15:
Hi I need the same, how did you succeeded?

Pokornypetr, 2025-03-16:
> Hi I need the same, how did you succeeded?
 hi, it has to be built on esp32 then you just need to use the code from sisi
`  - platform: pipsolar
    pipsolar_id: inverter0
    output_source_priority:
      id: inverter0_output_source_priority_select
      name: Axpert, výběr výstupního zdroje
      optionsmap:
        "Síť": "POP00"
        "Solár": "POP01"
        "Solár a Baterie": "POP02"
      statusmap:
        "0": "Síť"
        "1": "Solár"
        "2": "Solár a Baterie"

#  - platform: pipsolar
#    pipsolar_id: inverter0
#    charging_discharging_control:
#      # See MAX_Communication_Protocol_20200526.pdf page 18
#      name: "${name} charging discharging control"
#      optionsmap:
#        "111": "PBATCD111"
#        "011": "PBATCD011"
#        "101": "PBATCD101"
#        "110": "PBATCD110"
#        "010": "PBATCD010"
#        "100": "PBATCD100"
#        "001": "PBATCD001"
#        "000": "PBATCD000"
#      statusmap:
#        "111": "111"
#        "011": "011"
#        "101": "101"
#        "110": "110"
#        "010": "010"
#        "100": "100"
#        "001": "001"
#        "000": "000"
#
  - platform: pipsolar
    pipsolar_id: inverter0
    charger_source_priority:
      name: Axpert, výběr prioritního nabíjení
      optionsmap:
        "Síť": "PCP00"
        "Solár": "PCP01"
        "Solár a síť": "PCP02"
        "Jedině solár": "PCP03"
      statusmap:
        "0": "Síť"
        "1": "Solár"
        "2": "Solár a síť"
        "3": "Jedině solár"
#
  - platform: pipsolar
    pipsolar_id: inverter0
    current_max_ac_charging_current:
      name: Axpert, maximální dobíjecí proud ze sítě
      optionsmap:
        "2A": "MUCHGC002"
        "10A": "MUCHGC010"
        "20A": "MUCHGC020"
        "30A": "MUCHGC030"
        "40A": "MUCHGC040"
        "50A": "MUCHGC050"
        "60A": "MUCHGC060"
        "70A": "MUCHGC070"
        "80A": "MUCHGC080"
        "90A": "MUCHGC090"
        "100A": "MUCHGC100"
#        "110A": "MUCHGC110"
#        "120A": "MUCHGC120"
      statusmap:
        "2": "2A"
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
        "70": "70A"
        "80": "80A"
        "90": "90A"
        "100": "100A"
 #       "110": "110A"
#        "120": "120A"`