syssi/esphome-pipsolar #160: Support for JsdSolar inverter

pivanet, 2024-12-04:
Hi,
connected to jsdSolar with test protocol config.
Got this as a result. Could you help me to understand it? ;)

This invertor is able to use USB wifi stick that sends data to SmartEss with protocol ID = 6489, or in hex it's 5919 (what ever it means))
------------------
[logs_jsdsolar_logs.txt](https://github.com/user-attachments/files/18012449/logs_jsdsolar_logs.txt)

Thx in advance ;)

Alentus, 2025-02-15:
I hope this is useful. I followed the dongle requests and got this. All that remains is to parse this data).

```
tx: GPDAT0 rx:(054002000229.949.99000.000.00229.450.0100.0023.6000.000000000000075013.9000.00000040.0
tx: GPV rx:(014.0023.600.0000.01000000000384.2395.0370.0372.0000.000000000000000000000000000100000000000000000000000000000000
tx: GBAT rx:(023.6000.000221.021.6
tx: TE? rx:(ACDEHKX
tx: Q1 rx:(000.0000.0229.900000.01.9640.010000000
tx: F rx:#230.0015024.050.0
tx: GMOD rx:(B
tx: SVFW rx:(4.002 rx:(20241123
tx: GTMP rx:(040.0015.0025.0000.0000.0
tx: GLINE rx:(000.000.00264.0154.0255.0163.070.0040.0000000000000000000000000
tx: GBAT rx:(023.6000.000221.021.6
tx: GBUS rx:(384.2380.0380.0
tx: GCHG rx:(384.2023.602000.0007.54000.0028.227.029.2060.0004800601207200020000.00000.00000.00000.02
tx: GOP rx:(229.950.01000.0000.0400000000430000000000000000001032700000000000000000000
tx: GINV rx:(229.950.02001.3
tx: FAN??? rx:(020001160011300
tx: GWS rx:(0000000000000000000000000000000000
tx: BL rx:BL075
tx: GPV rx:(013.9023.600.0000.02000000000384.1395.0370.0372.0000.000000000000000000000000000100000000000000000000000000000000

```

Alentus, 2025-03-01:
my reserch results
` F #230.0 015 024.0 50.0 - invertor type 
   ac voltage | ac current | battery voltage | fregvency
I #222222222222222222222222222R1.4.002
	version HW or ect
BL BL100
   battery level
GBAT (024.6 000.00 02 21.0 21.6 -battery status
      curent battery voltage | battery current?? | battery cels or type??? | battery shutdown voltage| battery low voltage
GBMS (00000 65535 65535 65535 65535 65535 65535 65535 00000 00000 65535 65535 65535 65535 00000 65535 65535
GBUS (386.4 380.0 380.0
GCHG (386.1 024.6 02 000.0 007.15 000.00 28.2 27.0 29.2 060.00 0480 060 120 720 0 0 0 0 000.00 000.00 000.00 000.0 2
     dc bus voltage| battery voltage| cels??? | curent???| ??? | ??? | constant voltage point| floatin charge point | equalization point| max charge curent | ???? | equalization time | equalization delay | equalization interval|
GINV (230.0 50.02 001.2
     curent output voltage | output fregvency | curent

GMOD (B curent worck mode
      B - battery

GPPV ( 1 013.7 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0 000.0
GTMP (037.0 015.0 024.0 000.0 000.0 temperature sensor
      curent temp | ??? | ??? | ??? | ??? |
GBOOT (04369 21845
GFAIL (00 05 00112 0000 0000 0000 0000 2814 4391 0302 0000 0137 0000 0410 471 00000 65535 00000 00000 2 00000 00000 00260
GFLAG (00 00 00 00 01 01 0 005.7
GLINE (000.0 00.00 264.0 185.0 255.0 194.0 60.00 40.00 00000 000 00000 00000 00000
GPARA (+00035 -00070 04441 10437 50000 +09.000 09000
GPIGS (000.0 00.0 024.6 000.0 038.1 038.6 00.0 000.1 041.5 000.0 386.3 00000 00000 00000 37.0 15.0 9.000 0 0 0000
GPIRI (40 007 000 03600 03600 027 0 0
GCF ( BT0000000000000000 RL0000000000001111 P10000000001110000 I10000000000000000 I20000001000000000
GOP (000.0 02.68 000.00 00.01 00000 00000 00000 00000 00000 000 10453 00000 00000 00000 00000
GPV (013.7 024.6 00.00 00.00 00000 00 0 0 384.0 402.5 377.5 379.5 000.0 00 00000 00000 0000 0000 0000 0010 00000 00000 00000 0000000000000000
GWS (00 0000000000000000 0000000000000000
NDI (22222222222222222
GPDAT0 (0 5 4002 0 00 230.1 49.99 000.0 00.00 229.7 50.01 00.00 23.5 000.0 000 0000 0000 075 013.7 000.0 0000 039.0 status 
       | ??? | ??? | ??? | ??? | ??? |output voltage| output fregvency| ??? | ??? |output voltage?| output fregvency? | ??? | battery voltage| ??? | ??? | ??? | ??? |battery level| ??? | ??? | ??? | ??? |
Q1 (000.0 000.0 230.1 000 00.0 1.95 39.0 10000000
	|???| ??? | output voltage| ??? | ??? | ??? | ??? | ??? |

SVFW (4.002 (20241123 - software version
FAN??? (020 00141 00135 0 0
		curent fan speed |hole sensor fan1??? |hole sensor fan2???| ??? | ??? |
OPR?? (00  - output priority
OPM?? (01  output mode
TBAT? (0  battery type
CPR?? (02 - charger priority
CHGC??? (060 - chareg current
EOD???? (21.0 - battery underlotage level
DATE?????? 00 01 01 -curent date
TIME?????? 00 20 39 -curent time
V??? (230 230  - output voltage
GCC??? (040 - mains chargin current

TE? запрос включенных бинарних параметров
TD? запрос выключених бинарних параметров

TEх установка бинароного параметра в высокий уровень х - буква A-Z
TDх установка бинароного параметра в низкий уровень уровень х - буква A-Z
	M - блокировка изменения параметров
	C - auto restart when overload p09
	D - auto restart when overtemperature p10
	E - main input cut warning p11
	I - energy savin mode p12
	G - overload transer to bypass p13
	O - silent mode p14
	Y - inverter soft start setting p25
	L - battery disconnection alarm `

Alentus, 2026-01-17:
> [protocol JSDSOLAR.docx](https://github.com/user-attachments/files/24690715/protocol.JSDSOLAR.docx) Dear Alentus, I need your help. I have a j5500hpc inverter and the kit did not include a WiFi logger. There is a USB-A connector for the logger on the inverter, what protocol is used there? RS232? What is the pinout of this connector? I want to connect it to Home Assistant.

НІ, in my j4000hc inverter, I have the following pinout  

[image]

red - +5v
white - RX inv
green - TX inv
black  - GND

i connet that way rx inv - tx esp, tx inv - rx esp, gnd - gnd. 
phis level is rs232 you need rs232 - uart converter
Thx for protocol docs.

Pank172, 2026-01-17:
But you're welcome. I was happy to help.
So I need a converter based on the MAX3232 chip?

Alentus, 2026-01-17:
> > > Ось мій yaml-файл, спробуйте.
> > > [J5500HPC.txt](https://github.com/user-attachments/files/24691234/J5500HPC.txt)
> > 
> > 
> > Я не використовую esphome я на базі kc868 a6 реальне керування переміканням авр. але всеодно буде корисно дякую.
> 
> Ви з України?

Так чернігвіська область хто ще буде так займатись тими китайськими інверторами як не ми.

Pank172, 2026-01-17:
Я теж з Чернігівщини. Бобровиця Ніжинського района.

Pank172, 2026-01-17:
КС868-А6 - це ESP32.

Alentus, 2026-01-17:
> КС868-С6 - це ESP32.

так це плата на базі есп32 

https://www.kincony.com/esp32-6-channel-relay-module-kc868-a6.html

Pank172, 2026-01-17:
Ми з Вами ще й земляки.

Pank172, 2026-01-18:
В мене походу дохла плата комунікації, або десь непропай. На роз'ємі USB-A тільки живлення, сигналів немає.

[image]

Pank172, 2026-01-18:
Є тільки з'єднання з БМС через RS485. Що то за перемички - невідомо. Даних на плату теж немає.

Pank172, 2026-01-19:
> > В мене походу дохла плата комунікації, або десь непропай. На роз'ємі USB-A тільки живлення, сигналів немає.
> > [image]
> 
> дивна плата на юсб не видно конвертора rs232 і тут окремо стоїть мк у мене тільки перетворювачі живлення і коветроти на rs232 і rs485 цікаво, що йде на розєм тайп б. Перемички скоріш за всі зміни логіки виходу сухого контакту реле.

Там є 10-нога мікруха, це драйвер USB CH340E. Це мабуть на USB-B.

ngoyat, 2026-06-08:
hi, anybody know which protocol, commands does gootu GT-H4865M27P9, 6.5kw (9000w PV) hybrid inverter uses?. It has a usb socket for communication. Pcb Board used in it is similar to many other hybrid solar inverters. i tried GBAT, QPIGS commands but no response from inverter. I tried 2400 and 9600 baud rates  on 8N1.  Voltages on usb port are black > red = 5.1v, Black > White 0.11 v, Black > Green -5.61 v, Red > White 5.09v, Red> Green 10.6 v, White > Green -5.6v. I don't have original dongle but i suspect DUWB-NK2 dongle works with it [https://www.maxhub-ess.com/upload/download/2025-08/688c6b87265f5.pdf]. I am using wemos d1 mini with rs232 to uart converter (connection picture attached). I am also uploading codes i am using in wemos. Thanks for you help in advance

[wemos_qpi_protocol_probe.txt](https://github.com/user-attachments/files/28705297/wemos_qpi_protocol_probe.txt)

[wemos_g_protocol_probe.txt](https://github.com/user-attachments/files/28705312/wemos_g_protocol_probe.txt)

[image]

[image]

lindehasseasy, 2026-09-16:
https://github.com/lindehasseasy/hass-wattseek

NewBotanik, 2026-10-13:
[Моя інтеграція через кабель USB-B](https://github.com/NewBotanik/J5500HPC-HA-Integration
)

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

pivanet, 2026-10-13:
> 31\. [Моя інтеграція через кабель USB-B](https://github.com/NewBotanik/J5500HPC-HA-Integration)

Друже, є можна деталніше з кабелем що куди паяти? я спробував з есп32 і платкою max3232 і в порт "юсб" і щось нічого не вийшло, дякую

NewBotanik, 2026-10-19:
> 39\. > 31. [Моя інтеграція через кабель USB-B](https://github.com/NewBotanik/J5500HPC-HA-Integration)
>          
>          
>          Друже, є можна деталніше з кабелем що куди паяти? я спробував з есп32 і платкою max3232 і в порт "юсб" і щось нічого не вийшло, дякую

Вітаю.
Я нічого не паяв, звичайний кабель з принтера, в мене з аліка, але це, думаю, не має значення, [можна купити на розетці](https://rozetka.com.ua/ua/tv-cables/c80073/tip-konnektora-2=usb-type-b/)
[image]
є гніздо  поруч з гніздом свистка, USB-B.

[image]

В мене саме в нього підключено.

pivanet, 2026-10-19:
> 31\. Я нічого не паяв, звичайний кабель з принтера, в мене з аліка, але це, думаю, не має значення, [можна купити на розетці](https://rozetka.com.ua/ua/tv-cables/c80073/tip-konnektora-2=usb-type-b/)
>      <img alt="Image" width="238" height="239" src="https://private-user-images.githubusercontent.com/29075524/677707140-915350e3-4fd2-4891-bb12-4cb6585d0ffe.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTI0MDY1MjAsIm5iZiI6MTc5MjQwNjIyMCwicGF0aCI6Ii8yOTA3NTUyNC82Nzc3MDcxNDAtOTE1MzUwZTMtNGZkMi00ODkxLWJiMTItNGNiNjU4NWQwZmZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjEwMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYxMDE5VDEwMzcwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlNjBjM2Y1ZGM4NmE0MzkyYmNjMDdhY2FmNDcwN2VhM2FmMTE5NDg5ZTZmZmJjMGQzOGQ0YjkyYTU4MjRhYzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.5SUpPh3ILWHMqfAreTOLbGI838oLFTKE-jMriHxTPbU">
>      є гніздо  поруч з гніздом свистка, USB-B.

зрозумів. дякую і його на піни в есп32, так? а живлення для есп32 з того самого юсб-б чи окремо? дякую

NewBotanik, 2026-10-20:
>          зрозумів. дякую і його на піни в есп32, так? а живлення для есп32 з того самого юсб-б чи окремо? дякую

Ні. в мене напряму кабелем з'єднано     MiniPC.USB   <- USB-A ---кабель---- USB-B ->    Invertet.USB-B.
esp32 я не використовую. Наразі в мене нема потреби, бо в мене мініПК буквально рядом з інвертором і бетереєю.
Для просто USB я колись паяв стандартну схему з esp32, точніше це не просто USB, це RS232 у формі USB, але я не доробив прошивку, бо небуло протокола обміну. 
Я думав додати можливість знімати дані з рідного свистка в свій проект, свисток передає звичайним MODBUS, але там треба перенаправити трафік роутером, бо в ньому зашитий IP DessMonitor. Наразі руки не дійшли.
Можна спробувати ще Elfin-EW10 або Elfin-EW10A (перший з внутрішньою антеною, другий з зовнішньою), у форку який лежить в основі вроді би є така можливість, його(Elfin-EW10) треба підключати в звичайний USB інвертора, і там швидкість трохи вища 9600 бод (проти 2400 в USB-B), і в звичайному USB є живлення 5V для живлення Elfin-EW10.

[image]