syssi/esphome-pipsolar #164: SMH-III 24V 4.2kW

fedottt, 2024-12-12:
this is not issue, this is part of my expirience and eating cactus with this f*cking inverter

for changing **current_max_charging_current** default didnt work
i founded this:

in yaml
```
  - platform: pipsolar
    pipsolar_id: inv1
    current_max_charging_current:
      name: "Charge - Current Total"
      optionsmap:
        "10A": "MNCHGC010"  #   <-- take a look for this, it different from default
        "20A": "MNCHGC020"
        "30A": "MNCHGC030"
        "40A": "MNCHGC040"
        "50A": "MNCHGC050"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
```

AND! 

did some changes to 
`pipsolar/components/pipsolar/output/__init__.py`

CONF_CURRENT_MAX_CHARGING_CURRENT: ([ 10, 20, 30, 40, 50 ], "M**N**CHGC**0**%03.0f"),

[image]

with this change i can set total current for this invrter

fedottt, 2024-12-13:
> I have a SMH-II 3.2kw inverter ... so I think your solution could help me. Would you mind providing your esphome yaml config file please ? I'm struggling for several days to get it working!

use [https://github.com/syssi/esphome-pipsolar/tree/pip8048](pip8048) 
i have this inverter at mom's house and this yaml works good without modify

nheinemans, 2024-12-21:
Looking at the changes you made in __init__.py, it looks like you're changing a different file than the one in esphome:
https://github.com/esphome/esphome/blob/2024.12.2/esphome/components/pipsolar/output/__init__.py

I tried changing lines 52 and 53 in this file, using your example, but I can't get these values from my inverter (I also have a SMH-III 24V 4.2kW). Can you perhaps point me to a working yaml?

fedottt, 2024-12-23:
https://github.com/fedottt/pipsolar/blob/main/components/pipsolar/output/__init__.py

esphome is 48v only
so you need custom. from syssi, for example, or you can clone my repo

full yaml of my inverter

`#-------------------------------------------
# SUBTITUTIONS - basic definitions and description of the device
#-------------------------------------------
substitutions:
  friendly_name: Inverter
  device_name: easun_home
  # Module: smd on board
  status_led_pin: D4
  # UART
  uart_tx: D5
  uart_rx: D6
  battery_capacite: "110"

uart:
  - id: uart_bus
    tx_pin: $uart_tx
    rx_pin: $uart_rx
    baud_rate: 2400

#-------------------------------------------
# ESP - MAIN SETTINGS
#-------------------------------------------
esphome:
  name: $device_name
  friendly_name: $friendly_name

esp8266:
  board: d1_mini

# Enable Home Assistant API
api:
  encryption:
    key: "KEY"

ota:
  - platform: esphome

#-------------------------------------------
# LOGGER
#-------------------------------------------
logger:
  # disable log over uart
  baud_rate: 0

#-------------------------------------------
# ETHERNET
#-------------------------------------------
wifi:

#-------------------------------------------
# EXTERNAL AND CUSTOM COMPONENT
#-------------------------------------------
external_components:
  - source: github://fedottt/pipsolar
    refresh: 0s

#-------------------------------------------
# STATUS LED
#-------------------------------------------
status_led:
  pin: 
    number: $status_led_pin
    
[... 356 lines cut ...]
        "40": "40A"
        "50": "50A"

  - platform: pipsolar
    pipsolar_id: effekta_ax
    current_max_charging_current:
      name: "Charge - Current Total"
      optionsmap:
        "10A": "MNCHGC010"
        "20A": "MNCHGC020"
        "30A": "MNCHGC030"
        "40A": "MNCHGC040"
        "50A": "MNCHGC050"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"

#-------------------------------------------
# TIME COMPONENT
#-------------------------------------------
time:
  - platform: homeassistant`

nheinemans, 2024-12-27:
Thanks a lot for sharing your config. I didn't realise I should use a different pipsolar component. However, when I'm trying to use your repo, I'm getting this error message:

```INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/esphome-web-e5a136.yaml...
INFO Updating https://github.com/fedottt/pipsolar.git@None
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing garage (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
Compiling .pioenvs/garage/src/esphome/components/pipsolar/pipsolar.cpp.o
Compiling .pioenvs/garage/src/esphome/components/select/select_traits.cpp.o
Compiling .pioenvs/garage/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/garage/src/esphome/components/sensor/filter.cpp.o
In file included from src/esphome/components/pipsolar/pipsolar.cpp:1:
src/esphome/components/pipsolar/pipsolar.h:7:10: fatal error: esphome/components/pipsolar/select/pipsolar_select.h: No such file or directory
    7 | #include "esphome/components/pipsolar/select/pipsolar_select.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/garage/src/esphome/components/pipsolar/pipsolar.cpp.o] Error 1
========================== [FAILED] Took 9.51 seconds ==========================```

I'm using this from esphome in Home Assistant, both running the latest versions. Any idea what I'm doing wrong?

syssi, 2024-12-27:
Please try a `CLEAN` build and make sure there is a `select:` section at your YAML configuration.

fedottt, 2024-12-30:
> I didn't realise I should use a different pipsolar component.
heh, well thats it)) 
if we looking in code, we see only 40+ voltage
[image]

> Any idea what I'm doing wrong?
we need your conf) to make review))

ImNewty, 2025-01-12:
So if I want to use this with the SMH III 4.2kw 24V I have to change that in the library or can I use fedottt's one and it will just work?

fedottt, 2025-01-15:
> can I use fedottt's one

yep. you can use it

ImNewty, 2025-02-02:
Used this it works perfectly but why can't I choose 2A for the utility charging current?

fedottt, 2025-02-02:
Do you have 2a in inverter menu?
Have 2 easun, nobody can 2a. Only 10x step.
If inverter can this step you can modify init.py from first post with your steps. Maybe it work

ImNewty, 2025-02-02:
I do Have 2A but only for grid charge.