syssi/esphome-pipsolar #166: Voltronic 24V/3kW not showing any data

gabriktm, 2024-12-28:
Hello! I configured the script on esphome but the result is that I can only see 0.00 values. Im new on that and I need to learn. Someone can help me? 

[image]

I saw the other post and I need to send debug logs. Im using esp01s and GPIO1 and GPIO3. Which script can debug correctly?

syssi, 2024-12-28:
Please enable the debug output by extending your YAML configuration: https://github.com/syssi/esphome-pipsolar?tab=readme-ov-file#debugging

Could you provide another log including the debug output in plaintext?

gabriktm, 2024-12-28:
[12:51:17][I][main:466]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[12:51:17][I][main:469]: This is the set of commands supported by the pipsolar component!
[12:51:17][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[12:51:17][D][uart_debug:158]: <<< "QPI\xBE\xAC\r"
[12:51:17][D][uart_debug:158]: <<< "\x90+\xEB\xE5\x00"
[12:51:18][D][uart_debug:158]: >>> "QDIq\e\r"
[12:51:18][D][uart_debug:158]: <<< "QDIq\e\r"
[12:51:19][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[12:51:19][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[12:51:20][D][uart_debug:158]: >>> "QMN\xBBd\r"
[12:51:20][D][uart_debug:158]: <<< "QMN\xBBd\r"
[12:51:21][D][uart_debug:158]: >>> "QMODI\xC1\r"
[12:51:21][D][uart_debug:158]: <<< "QMODI\xC1\r"
[12:51:22][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[12:51:22][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[12:51:23][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[12:51:23][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[12:51:24][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[12:51:24][D][uart_debug:158]: <<< "QPIWS\xB4\xDA\r"
[12:51:25][D][uart_debug:158]: >>> "QT\'\xFF\r"
[12:51:25][D][uart_debug:158]: <<< "QT\'\xFF\r"
[12:51:26][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[12:51:26][D][uart_debug:158]: <<< "QPGS0?\xDA\r"
[12:51:27][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[12:51:27][D][uart_debug:158]: <<< "QPGS1/\xFB\r"
[12:51:28][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[12:51:28][D][uart_debug:158]: <<< "QPGS2\x1F\x98\r"
[12:51:29][I][main:556]: Testing PI41 split phase / multiple strings commands...
[12:51:29][D][uart_debug:158]: >>> "QPIGS2h-\r"
[12:51:29][D][uart_debug:158]: <<< "QPIGS2h-\r"
[12:51:30][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[12:51:30][D][uart_debug:158]: <<< "QP2GS0\x14\x05\r"
[12:51:31][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[12:51:31][D][uart_debug:158]: <<< "QP2GS1\x04$\r"
[12:51:32][I][main:580]: Testing unsupported PI18 commands...
[12:51:32][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[12:51:32][D][uart_debug:158]: <<< "^P005PIq\x8B\r"
[12:51:33][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[12:51:33][D][uart_debug:158]: <<< "^P005GSX\x14\r"
[12:51:34][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[12:51:34][D][uart_debug:158]: <<< "^P006MOD\xDD\xBE\r"
[12:51:35][I][main:604]: Testing unsupported PI17 commands...
[12:51:35][D][uart_debug:158]: >>> "^P003PI\r"
[12:51:35][D][uart_debug:158]: <<< "^P003PI\r"
[12:51:36][D][uart_debug:158]: >>> "^P004MOD\r"
[12:51:36][D][uart_debug:158]: <<< "^P004MOD\r"
[12:51:37][D][uart_debug:158]: >>> "^P005FLAG\r"
[12:51:37][D][uart_debug:158]: <<< "^P005FLAG\r"
[12:51:38][I][main:628]: Testing unsupported PI16 commands...
[12:51:38][D][uart_debug:158]: >>> "QPI\r"
[12:51:38][D][uart_debug:158]: <<< "QPI\r"
[12:51:39][D][uart_debug:158]: >>> "QMOD\r"
[12:51:39][D][uart_debug:158]: <<< "QMOD\r"
[12:51:40][D][uart_debug:158]: >>> "QPIGS\r"
[12:51:40][D][uart_debug:158]: <<< "QPIGS\r"
[12:51:41][D][uart_debug:158]: >>> "QPIRI\r"
[12:51:41][D][uart_debug:158]: <<< "QPIRI\r"
[12:51:42][D][uart_debug:158]: >>> "QMOD\r"
[12:51:42][I][main:662]: Done. Repeating...

syssi, 2024-12-28:
```
[12:51:41][D][uart_debug:158]: >>> "QPIRI\r"
[12:51:41][D][uart_debug:158]: <<< "QPIRI\r"
```

This indicates a local echo / loopback because of a short circuit between the RX and TX pin or a broken converter.

gabriktm, 2024-12-28:
Thanks for your fast reply!
This is how I connected the esp01s and the converter.
Can you help me in some ways? 

[image]
[image]

I used this configuration of pin:

uart:
  id: uart_0
  baud_rate: 2400
  tx_pin: GPIO1
  rx_pin: GPIO3

Sorry if Im asking, but Im new and I still need to learn about this esp01 and rs communication!
If you can help me, thanks!!

gabriktm, 2024-12-28:
This is an RP series inverter. It's particular because it has toroidal transformer. Lot of people uses but there are no information about protocol and even the website where I bought has no information. This could be useful for other people
[image]

gabriktm, 2024-12-28:
I checked all the connections and the pinout. no short circuit and I tried another converter.

[14:22:20][I][main:466]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[14:22:20][I][main:469]: This is the set of commands supported by the pipsolar component!
[14:22:20][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[14:22:20][D][uart_debug:158]: <<< "QPI\xBE\xAC\r"
[14:22:20][D][uart_debug:158]: <<< "\x90+\xEB\xE5\x00"
[14:22:21][D][uart_debug:158]: >>> "QDIq\e\r"
[14:22:21][D][uart_debug:158]: <<< "QDIq\e\r"
[14:22:22][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[14:22:22][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[14:22:23][D][uart_debug:158]: >>> "QMN\xBBd\r"
[14:22:23][D][uart_debug:158]: <<< "QMN\xBBd\r"
[14:22:24][D][uart_debug:158]: >>> "QMODI\xC1\r"
[14:22:24][D][uart_debug:158]: <<< "QMODI\xC1\r"
[14:22:25][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[14:22:25][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[14:22:26][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[14:22:26][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[14:22:27][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[14:22:27][D][uart_debug:158]: <<< "QPIWS\xB4\xDA\r"
[14:22:28][D][uart_debug:158]: >>> "QT\'\xFF\r"
[14:22:28][D][uart_debug:158]: <<< "QT\'\xFF\r"
[14:22:29][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[14:22:29][D][uart_debug:158]: <<< "QPGS0?\xDA\r"
[14:22:30][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[14:22:30][D][uart_debug:158]: <<< "QPGS1/\xFB\r"
[14:22:31][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[14:22:31][D][uart_debug:158]: <<< "QPGS2\x1F\x98\r"
[14:22:32][I][main:556]: Testing PI41 split phase / multiple strings commands...
[14:22:32][D][uart_debug:158]: >>> "QPIGS2h-\r"
[14:22:32][D][uart_debug:158]: <<< "QPIGS2h-\r"
[14:22:33][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[14:22:33][D][uart_debug:158]: <<< "QP2GS0\x14\x05\r"
[14:22:34][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[14:22:34][D][uart_debug:158]: <<< "QP2GS1\x04$\r"
[14:22:35][I][main:580]: Testing unsupported PI18 commands...
[14:22:35][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[14:22:35][D][uart_debug:158]: <<< "^P005PIq\x8B\r"
[14:22:36][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[14:22:36][D][uart_debug:158]: <<< "^P005GSX\x14\r"
[14:22:37][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[14:22:37][D][uart_debug:158]: <<< "^P006MOD\xDD\xBE\r"
[14:22:38][I][main:604]: Testing unsupported PI17 commands...
[14:22:38][D][uart_debug:158]: >>> "^P003PI\r"
[14:22:38][D][uart_debug:158]: <<< "^P003PI\r"
[14:22:39][D][uart_debug:158]: >>> "^P004MOD\r"
[14:22:39][D][uart_debug:158]: <<< "^P004MOD\r"
[14:22:40][D][uart_debug:158]: >>> "^P005FLAG\r"
[14:22:40][D][uart_debug:158]: <<< "^P005FLAG\r"
[14:22:41][I][main:628]: Testing unsupported PI16 commands...
[14:22:41][D][uart_debug:158]: >>> "QPI\r"
[14:22:41][D][uart_debug:158]: <<< "QPI\r"
[14:22:42][D][uart_debug:158]: >>> "QMOD\r"
[14:22:42][D][uart_debug:158]: <<< "QMOD\r"
[14:22:43][D][uart_debug:158]: >>> "QPIGS\r"
[14:22:43][D][uart_debug:158]: <<< "QPIGS\r"
[14:22:44][D][uart_debug:158]: >>> "QPIRI\r"
[14:22:44][D][uart_debug:158]: <<< "QPIRI\r"
[14:22:45][D][uart_debug:158]: >>> "QMOD\r"
[14:22:45][I][main:662]: Done. Repeating...

[image]

Following the manual it says that pin 1 is rx, pin 2 is tx and pin3 gnd. Measuring I have GND-RX -5V and GND-TX 0.6V
I don't understand...

gabriktm, 2024-12-28:
[image]
[image]

syssi, 2024-12-28:
> I don't understand why if I don't put the sniffer.yaml it communicates with the inverter.

This is because an echo of an request frame has a valid checksum and the decoder isn't able to identify the incorrect reply. If you enable the debug mode of the `uart` you will see the invalid responses at the log too.

syssi, 2024-12-28:
If unplug the inverter is the local echo gone?

syssi, 2024-12-28:
Could you provide your YAML configuration? You have the make sure the `logger` doesn't use GPIO1/GPIO3 too.

gabriktm, 2024-12-28:
> If you unplug the inverter is the local echo gone?

INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/sniffer.yaml...
INFO Starting log output from 192.168.1.214 using esphome API
INFO Successfully connected to sniffer @ 192.168.1.214 in 0.005s
INFO Successful handshake with sniffer @ 192.168.1.214 in 2.396s
[15:55:22][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 28 2024, 12:46:04
[15:55:22][C][wifi:600]: WiFi:
[15:55:22][C][wifi:428]:   Local MAC: B4:E6:2D:70:2B:A0
[15:55:22][C][wifi:433]:   SSID: [redacted]
[15:55:22][C][wifi:436]:   IP Address: 192.168.1.214
[15:55:22][C][wifi:439]:   BSSID: [redacted]
[15:55:22][C][wifi:441]:   Hostname: 'sniffer'
[15:55:22][C][wifi:443]:   Signal strength: -66 dB ▂▄▆█
[15:55:22][C][wifi:447]:   Channel: 2
[15:55:22][C][wifi:448]:   Subnet: 255.255.255.0
[15:55:22][C][wifi:449]:   Gateway: 192.168.1.1
[15:55:22][C][wifi:450]:   DNS1: 0.0.0.0
[15:55:22][C][wifi:451]:   DNS2: 0.0.0.0
[15:55:22][C][logger:185]: Logger:
[15:55:22][C][logger:186]:   Level: DEBUG
[15:55:22][C][logger:188]:   Log Baud Rate: 0
[15:55:22][C][logger:189]:   Hardware UART: UART0
[15:55:22][C][uart.arduino_esp8266:118]: UART Bus:
[15:55:22][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[15:55:22][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[15:55:22][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[15:55:22][C][uart.arduino_esp8266:124]:   Baud Rate: 2400 baud
[15:55:22][C][uart.arduino_esp8266:125]:   Data Bits: 8
[15:55:22][C][uart.arduino_esp8266:126]:   Parity: NONE
[15:55:22][C][uart.arduino_esp8266:127]:   Stop bits: 1
[15:55:22][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[15:55:22][C][mdns:116]: mDNS:
[15:55:22][C][mdns:117]:   Hostname: sniffer
[15:55:22][C][esphome.ota:073]: Over-The-Air updates:
[15:55:22][C][esphome.ota:074]:   Address: 192.168.1.214:8266
[15:55:22][C][esphome.ota:075]:   Version: 2
[15:55:22][C][esphome.ota:078]:   Password configured
[15:55:22][C][safe_mode:018]: Safe Mode:
[15:55:22][C][safe_mode:019]:   Boot considered successful after 60 seconds
[15:55:22][C][safe_mode:021]:   Invoke after 10 boot attempts
[15:55:22][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[15:55:22][C][api:140]: API Server:
[15:55:22][C][api:141]:   Address: 192.168.1.214:6053
[15:55:22][C][api:143]:   Using noise encryption: YES
[15:55:40][I][main:466]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[15:55:40][I][main:469]: This is the set of commands supported by the pipsolar component!
[15:55:40][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[15:55:41][D][uart_debug:158]: >>> "QDIq\e\r"
[15:55:42][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[15:55:43][D][uart_debug:158]: >>> "QMN\xBBd\r"
[15:55:44][D][uart_debug:158]: >>> "QMODI\xC1\r"
[15:55:45][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[15:55:46][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[15:55:47][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[15:55:48][D][uart_debug:158]: >>> "QT\'\xFF\r"
[15:55:49][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[15:55:50][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[15:55:51][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[15:55:52][I][main:556]: Testing PI41 split phase / multiple strings commands...
[15:55:52][D][uart_debug:158]: >>> "QPIGS2h-\r"
[15:55:53][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[15:55:54][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[15:55:55][I][main:580]: Testing unsupported PI18 commands...
[15:55:55][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[15:55:56][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[15:55:57][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[15:55:58][I][main:604]: Testing unsupported PI17 commands...
[15:55:58][D][uart_debug:158]: >>> "^P003PI\r"
[15:55:59][D][uart_debug:158]: >>> "^P004MOD\r"
[15:56:00][D][uart_debug:158]: >>> "^P005FLAG\r"
[15:56:01][I][main:628]: Testing unsupported PI16 commands...
[15:56:01][D][uart_debug:158]: >>> "QPI\r"
[15:56:02][D][uart_debug:158]: >>> "QMOD\r"
[15:56:03][D][uart_debug:158]: >>> "QPIGS\r"
[15:56:04][D][uart_debug:158]: >>> "QPIRI\r"
[15:56:05][D][uart_debug:158]: >>> "QMOD\r"
[15:56:05][I][main:662]: Done. Repeating...

gabriktm, 2024-12-28:
> Could you provide your YAML configuration? You have the make sure the `logger` doesn't use GPIO1/GPIO3 too.

substitutions:
  name: sniffer
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  min_version: 2024.6.0

esp8266:
  board: esp01_1m

wifi:
  ssid: 
  password:
  manual_ip: 
    gateway: 192.168.1.1
    static_ip: 192.168.1.214
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Inverter Fallback Hotspot"
    password: "lDl0rIfR4jAy"

ota:
  - platform: esphome
    password: "e6a013b7d0629a3e78a5d07370dac4d2"

logger:
  level: DEBUG
  baud_rate: 0

api:
  encryption:
    key: "KOsBxlIyoVYpp6SfyuVU/e0Ffmd4jxinFo59ozRGi/M="
  reboot_timeout: 0s

uart:
  id: uart_0
  baud_rate: 2400
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

interval:
  - interval: 80s
    then:
      # PI30/PI30MAX/PI30REVO/PI41
      - logger.log:
          level: INFO
          format: "Testing PI30/PI30MAX/PI30REVO/PI41 commands..."
      - logger.log:
          level: INFO
          format: "This is the set of commands supported by the pipsolar component!"
      - uart.write: [0x51, 0x50, 0x49, 0xBE, 0xAC, 0x0D]  # QPI\xbe\xac\r
      - delay: 1s
      - uart.write: [0x51, 0x44, 0x49, 0x71, 0x1B, 0x0D]  # QDIq\x1b\r
      - delay: 1s
      - uart.write: [0x51, 0x46, 0x4C, 0x41, 0x47, 0x98, 0x74, 0x0D]  # QFLAG\x98t\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4E, 0xBB, 0x64, 0x0D]  # QMN\xbbd\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4F, 0x44, 0x49, 0xC1, 0x0D]  # QMODI\xc1\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x47, 0x53, 0xB7, 0xA9, 0x0D]  # QPIGS\xB7\xA9\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x52, 0x49, 0xF8, 0x54, 0x0D]  # QPIRI\xF8T\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x57, 0x53, 0xB4, 0xDA, 0x0D]  # QPIWS\xb4\xda\r
      - delay: 1s
      - uart.write: [0x51, 0x54, 0x27, 0xFF, 0x0D]  # QT'\xff\r
      - delay: 1s
      # Test some PI30 commands from Phocos Any-Grid PSW-H inverters
      - uart.write: [0x51, 0x50, 0x47, 0x53, 0x30, 0x3F, 0xDA, 0x0D]  # QPGS0<CRC>\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x47, 0x53, 0x31, 0x2F, 0xFB, 0x0D]  # QPGS1<CRC>\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x47, 0x53, 0x32, 0x1F, 0x98, 0x0D]  # QPGS2<CRC>\r
      - delay: 1s

      # PI41 split phase
      - logger.log:
          level: INFO
          format: "Testing PI41 split phase / multiple strings commands..."
      - uart.write: [0x51, 0x50, 0x49, 0x47, 0x53, 0x32, 0x68, 0x2D, 0x0D]  # QPIGS2h-\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x32, 0x47, 0x53, 0x30, 0x14, 0x05, 0x0D]  # QP2GS0\x14\x05\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x32, 0x47, 0x53, 0x31, 0x04, 0x24, 0x0D]  # QP2GS1\x04$\r
      - delay: 1s

      # PI18
      - logger.log:
          level: INFO
          format: "Testing unsupported PI18 commands..."
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x35, 0x50, 0x49, 0x71, 0x8B, 0x0D]  # ^P005PIq\x8b\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x35, 0x47, 0x53, 0x58, 0x14, 0x0D]  # ^P005GSX\x14\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x36, 0x4D, 0x4F, 0x44, 0xDD, 0xBE, 0x0D]  # ^P006MOD\xdd\xbe\r
      - delay: 1s

      # PI17
      - logger.log:
          level: INFO
          format: "Testing unsupported PI17 commands..."
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x33, 0x50, 0x49, 0x0D]  # ^P003PI\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x34, 0x4D, 0x4F, 0x44, 0x0D]  # ^P004MOD\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x35, 0x46, 0x4C, 0x41, 0x47, 0x0D]  # ^P005FLAG\r
      - delay: 1s

      # PI16
      - logger.log:
          level: INFO
          format: "Testing unsupported PI16 commands..."
      - uart.write: [0x51, 0x50, 0x49, 0x0D]  # QPI\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4F, 0x44, 0x0D]  # QMOD\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x47, 0x53, 0x0D]  # QPIGS\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x52, 0x49, 0x0D]  # QPIRI\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4F, 0x44, 0x0D]  # QMOD\r

      - logger.log:
          level: INFO
          format: "Done. Repeating..."

gabriktm, 2024-12-28:
> Could you provide your YAML configuration? You have the make sure the `logger` doesn't use GPIO1/GPIO3 too.

substitutions:
  name: sniffer
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  min_version: 2024.6.0

esp8266:
  board: esp01_1m

wifi:
  ssid: "MOWGLI AP 2.0"
  password: "N0:3QO[1j[LOi7-Sg('K6.m+iR"
  manual_ip: 
    gateway: 192.168.1.1
    static_ip: 192.168.1.214
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Inverter Fallback Hotspot"
    password: "lDl0rIfR4jAy"

ota:
  - platform: esphome
    password: "e6a013b7d0629a3e78a5d07370dac4d2"

logger:
  level: DEBUG
  baud_rate: 0

api:
  encryption:
    key: "KOsBxlIyoVYpp6SfyuVU/e0Ffmd4jxinFo59ozRGi/M="
  reboot_timeout: 0s

uart:
  id: uart_0
  baud_rate: 2400
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\r"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

interval:
  - interval: 80s
    then:
      # PI30/PI30MAX/PI30REVO/PI41
      - logger.log:
          level: INFO
          format: "Testing PI30/PI30MAX/PI30REVO/PI41 commands..."
      - logger.log:
          level: INFO
          format: "This is the set of commands supported by the pipsolar component!"
      - uart.write: [0x51, 0x50, 0x49, 0xBE, 0xAC, 0x0D]  # QPI\xbe\xac\r
      - delay: 1s
      - uart.write: [0x51, 0x44, 0x49, 0x71, 0x1B, 0x0D]  # QDIq\x1b\r
      - delay: 1s
      - uart.write: [0x51, 0x46, 0x4C, 0x41, 0x47, 0x98, 0x74, 0x0D]  # QFLAG\x98t\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4E, 0xBB, 0x64, 0x0D]  # QMN\xbbd\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4F, 0x44, 0x49, 0xC1, 0x0D]  # QMODI\xc1\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x47, 0x53, 0xB7, 0xA9, 0x0D]  # QPIGS\xB7\xA9\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x52, 0x49, 0xF8, 0x54, 0x0D]  # QPIRI\xF8T\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x57, 0x53, 0xB4, 0xDA, 0x0D]  # QPIWS\xb4\xda\r
      - delay: 1s
      - uart.write: [0x51, 0x54, 0x27, 0xFF, 0x0D]  # QT'\xff\r
      - delay: 1s
      # Test some PI30 commands from Phocos Any-Grid PSW-H inverters
      - uart.write: [0x51, 0x50, 0x47, 0x53, 0x30, 0x3F, 0xDA, 0x0D]  # QPGS0<CRC>\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x47, 0x53, 0x31, 0x2F, 0xFB, 0x0D]  # QPGS1<CRC>\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x47, 0x53, 0x32, 0x1F, 0x98, 0x0D]  # QPGS2<CRC>\r
      - delay: 1s

      # PI41 split phase
      - logger.log:
          level: INFO
          format: "Testing PI41 split phase / multiple strings commands..."
      - uart.write: [0x51, 0x50, 0x49, 0x47, 0x53, 0x32, 0x68, 0x2D, 0x0D]  # QPIGS2h-\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x32, 0x47, 0x53, 0x30, 0x14, 0x05, 0x0D]  # QP2GS0\x14\x05\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x32, 0x47, 0x53, 0x31, 0x04, 0x24, 0x0D]  # QP2GS1\x04$\r
      - delay: 1s

      # PI18
      - logger.log:
          level: INFO
          format: "Testing unsupported PI18 commands..."
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x35, 0x50, 0x49, 0x71, 0x8B, 0x0D]  # ^P005PIq\x8b\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x35, 0x47, 0x53, 0x58, 0x14, 0x0D]  # ^P005GSX\x14\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x36, 0x4D, 0x4F, 0x44, 0xDD, 0xBE, 0x0D]  # ^P006MOD\xdd\xbe\r
      - delay: 1s

      # PI17
      - logger.log:
          level: INFO
          format: "Testing unsupported PI17 commands..."
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x33, 0x50, 0x49, 0x0D]  # ^P003PI\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x34, 0x4D, 0x4F, 0x44, 0x0D]  # ^P004MOD\r
      - delay: 1s
      - uart.write: [0x5E, 0x50, 0x30, 0x30, 0x35, 0x46, 0x4C, 0x41, 0x47, 0x0D]  # ^P005FLAG\r
      - delay: 1s

      # PI16
      - logger.log:
          level: INFO
          format: "Testing unsupported PI16 commands..."
      - uart.write: [0x51, 0x50, 0x49, 0x0D]  # QPI\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4F, 0x44, 0x0D]  # QMOD\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x47, 0x53, 0x0D]  # QPIGS\r
      - delay: 1s
      - uart.write: [0x51, 0x50, 0x49, 0x52, 0x49, 0x0D]  # QPIRI\r
      - delay: 1s
      - uart.write: [0x51, 0x4D, 0x4F, 0x44, 0x0D]  # QMOD\r

      - logger.log:
          level: INFO
          format: "Done. Repeating..."

gabriktm, 2024-12-28:
> Could you provide your YAML configuration? You have the make sure the `logger` doesn't use GPIO1/GPIO3 too.

- can you help me?

syssi, 2024-12-29:
The YAML above looks good. Let's summarize our findings:

1. There is no short circuit between RX and TX!
2. As soon the inverter is connected the ESP receives a local echo

Next idea: 

Plug the data cable into the converter again but turn the inverter off now. Is the local echo gone again? I would like to understand the local echo is the feedback of the converter or a phyiscal connection between RX and TX at the inside of the inverter.

gabriktm, 2024-12-29:
> The YAML above looks good. Let's summarize our findings:
>
>
>
> 1. There is no short circuit between RX and TX!
>
> 2. As soon the inverter is connected the ESP receives a local echo
>
>
>
> Next idea:
>
>
>
> Plug [...]

Ok this is the try with cable connected but inverter off. Seems there is again.

17:06:32][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[17:06:33][I][main:466]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[17:06:33][I][main:469]: This is the set of commands supported by the pipsolar component!
[17:06:33][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[17:06:33][D][uart_debug:158]: <<< "QPI\xBE\xAC\r"
[17:06:34][D][uart_debug:158]: >>> "QDIq\e\r"
[17:06:34][D][uart_debug:158]: <<< "QDIq\e\r"
[17:06:35][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[17:06:35][D][uart_debug:158]: <<< "QFLAG\x98t\r"
[17:06:36][D][uart_debug:158]: >>> "QMN\xBBd\r"
[17:06:36][D][uart_debug:158]: <<< "QMN\xBBd\r"
[17:06:37][D][uart_debug:158]: >>> "QMODI\xC1\r"
[17:06:37][D][uart_debug:158]: <<< "QMODI\xC1\r"
[17:06:38][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[17:06:38][D][uart_debug:158]: <<< "QPIGS\xB7\xA9\r"
[17:06:39][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[17:06:39][D][uart_debug:158]: <<< "QPIRI\xF8T\r"
[17:06:40][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[17:06:40][D][uart_debug:158]: <<< "QPIWS\xB4\xDA\r"
[17:06:41][D][uart_debug:158]: >>> "QT\'\xFF\r"
[17:06:41][D][uart_debug:158]: <<< "QT\'\xFF\r"
[17:06:42][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[17:06:42][D][uart_debug:158]: <<< "QPGS0?\xDA\r"
[17:06:43][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[17:06:44][D][uart_debug:158]: <<< "QPGS1/\xFB\r"
[17:06:44][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[17:06:45][D][uart_debug:158]: <<< "QPGS2\x1F\x98\r"
[17:06:46][I][main:556]: Testing PI41 split phase / multiple strings commands...
[17:06:46][D][uart_debug:158]: >>> "QPIGS2h-\r"
[17:06:46][D][uart_debug:158]: <<< "QPIGS2h-\r"
[17:06:47][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[17:06:47][D][uart_debug:158]: <<< "QP2GS0\x14\x05\r"
[17:06:48][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[17:06:48][D][uart_debug:158]: <<< "QP2GS1\x04$\r"
[17:06:49][I][main:580]: Testing unsupported PI18 commands...
[17:06:49][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[17:06:49][D][uart_debug:158]: <<< "^P005PIq\x8B\r"
[17:06:50][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[17:06:50][D][uart_debug:158]: <<< "^P005GSX\x14\r"
[17:06:51][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[17:06:51][D][uart_debug:158]: <<< "^P006MOD\xDD\xBE\r"
[17:06:52][I][main:604]: Testing unsupported PI17 commands...
[17:06:52][D][uart_debug:158]: >>> "^P003PI\r"
[17:06:52][D][uart_debug:158]: <<< "^P003PI\r"
[17:06:53][D][uart_debug:158]: >>> "^P004MOD\r"
[17:06:53][D][uart_debug:158]: <<< "^P004MOD\r"
[17:06:54][D][uart_debug:158]: >>> "^P005FLAG\r"
[17:06:54][D][uart_debug:158]: <<< "^P005FLAG\r"
[17:06:55][I][main:628]: Testing unsupported PI16 commands...
[17:06:55][D][uart_debug:158]: >>> "QPI\r"
[17:06:55][D][uart_debug:158]: <<< "QPI\r"
[17:06:56][D][uart_debug:158]: >>> "QMOD\r"
[17:06:56][D][uart_debug:158]: <<< "QMOD\r"
[17:06:57][D][uart_debug:158]: >>> "QPIGS\r"
[17:06:57][D][uart_debug:158]: <<< "QPIGS\r"
[17:06:58][D][uart_debug:158]: >>> "QPIRI\r"
[17:06:58][D][uart_debug:158]: <<< "QPIRI\r"
[17:06:59][D][uart_debug:158]: >>> "QMOD\r"
[17:06:59][I][main:662]: Done. Repeating...
[17:06:59][D][uart_debug:158]: <<< "QMOD\r"

syssi, 2024-12-29:
Please remove the data cable and use a multimeter to measure the resistance between `pin 1` and `pin 2` of the RJ45 socket.

gabriktm, 2024-12-29:
> Please remove the data cable and use a multimeter to measure the resistance between `pin 1` and `pin 2` of the RJ45 socket.

Around 2.6kOhm
[image]

gabriktm, 2024-12-29:
> This is confusing. I'm out of ideas.

Can you explain me? The resistance value is strange or what? Maybe this 2 pins are not TX and RX. Maybe this rj45 is not rs232 but rs485, in fact on the manual I can see there is also the picture of RS485 
[image]

gabriktm, 2024-12-29:
@syssi Yes is checked the pinout on RJ45. This is a RS485 protocol!! Is there something can I do? I have ready the converter and the esp01s but I don't know how to do for the script on esphome

syssi, 2024-12-29:
I tried to replicate your setup:

If there is a direct connection between GPIO1 and GPIO3 I can see the local echo:

```
[17:56:28][D][uart_debug:158]: >>> "QPI\r"
[17:56:28][D][uart_debug:158]: <<< "QPI\r"
[17:56:29][D][uart_debug:158]: >>> "QMOD\r"
[17:56:29][D][uart_debug:158]: <<< "QMOD\r"
[17:56:30][D][uart_debug:158]: >>> "QPIGS\r"
[17:56:30][D][uart_debug:158]: <<< "QPIGS\r"
[17:56:31][D][uart_debug:158]: >>> "QPIRI\r"
[17:56:31][D][uart_debug:158]: <<< "QPIRI\r"
```

[image]

If I attach es 2,7 kOhm resistor between RX and TX there is no echo anymore:

```
[17:58:46][D][uart_debug:158]: >>> "QDIq\e\r"
[17:58:47][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[17:58:48][D][uart_debug:158]: >>> "QMN\xBBd\r"
[17:58:49][D][uart_debug:158]: >>> "QMODI\xC1\r"
[17:58:50][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[17:58:51][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[17:58:52][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
```

[image]

I'm confused because the resistance is the same on your side and you can see a local echo nevertheless. 

Let's the the RS485 route. This is probably the project you are looking for: https://github.com/syssi/esphome-smg-ii

gabriktm, 2024-12-29:
The project you linked me use the rs232 too

> I tried to replicate your setup:
> 
> If there is a direct connection between GPIO1 and GPIO3 I can see the local echo:
> 
> ```
> [17:56:28][D][uart_debug:158]: >>> "QPI\r"
> [17:56:28][D][uart_debug:158]: <<< "QPI\r"
> [17:56:29][D][uart_debug:158]: >>> "QMOD\r"
> [17:56:29][D][uart_debug:158]: <<< "QMOD\r"
> [17:56:30][D][uart_debug:158]: >>> "QPIGS\r"
> [17:56:30][D][uart_debug:158]: <<< "QPIGS\r"
> [17:56:31][D][uart_debug:158]: >>> "QPIRI\r"
> [17:56:31][D][uart_debug:158]: <<< "QPIRI\r"
> ```
> 
> ![uart-loopback](https://private-user-images.githubusercontent.com/2735933/399175751-4b5639e6-14c1-4999-86c1-4b2751258aaf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU0OTIwOTQsIm5iZiI6MTczNTQ5MTc5NCwicGF0aCI6Ii8yNzM1OTMzLzM5OTE3NTc1MS00YjU2MzllNi0xNGMxLTQ5OTktODZjMS00YjI3NTEyNThhYWYuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjlUMTcwMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTQxMGRmNWUwZTJiZTBmOGE2MzRhMGMxMjE5NDdjMDBlNzIyMmRjYjg4MTk5NDUwODFkMDFlOGNhMTIyNjBlZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.jsQFfkGVaNsOMK6di88IOtDkq3OPcZ_3vb8OmRimovw)
> 
> If I attach es 2,7 kOhm resistor between RX and TX there is no echo anymore:
> 
> ```
> [17:58:46][D][uart_debug:158]: >>> "QDIq\e\r"
> [17:58:47][D][uart_debug:158]: >>> "QFLAG\x98t\r"
> [17:58:48][D][uart_debug:158]: >>> "QMN\xBBd\r"
> [17:58:49][D][uart_debug:158]: >>> "QMODI\xC1\r"
> [17:58:50][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
> [17:58:51][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
> [17:58:52][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
> ```
> 
> ![uart-no-loopback](https://private-user-images.githubusercontent.com/2735933/399175752-b92e326b-7f32-4e27-87b0-66d55ad5af3f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU0OTIwOTQsIm5iZiI6MTczNTQ5MTc5NCwicGF0aCI6Ii8yNzM1OTMzLzM5OTE3NTc1Mi1iOTJlMzI2Yi03ZjMyLTRlMjctODdiMC02NmQ1NWFkNWFmM2YuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjlUMTcwMzE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWY4N2I4MDNjNmU1YTQ4ODU2MzVmN2M2OTVkZGU1NTNmZmQ2YmVmMDliYzY1YTUwMGEzODZmZWEyYTljNmU1YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ZF46Z1fXP0CtOMVMIZzhF5BTIO9D6zgE7QG8dGGtEpI)
> 
> I'm confused because the resistance is the same on your side and you can see a local echo nevertheless.
> 
> Let's the the RS485 route. This is probably the project you are looking for: https://github.com/syssi/esphome-smg-ii

Thanks now I understand what you mean! 

The project you linked me use the rs232 too. Can you help me?

syssi, 2024-12-29:
Could you share a picture of your RS485 converter? I guess your Inverter talks Modbus instead of the PIP protocol if it's equipped with a RS485 port.

gabriktm, 2024-12-29:
Do you mean the converter inside of the inverter?

> Could you share a picture of your RS485 converter? I guess your Inverter talks Modbus instead of the PIP protocol if it's equipped with a RS485 port.

syssi, 2024-12-29:
The converter you want to attach to the ESP. A photo of the inverters communication board would be nice too.

gabriktm, 2024-12-29:
> The converter you want to attach to the ESP. A photo of the inverters communication board would be nice too.

[image]

syssi, 2024-12-29:
Perfect! Just attach the converter to GPIO1 and GPIO3 + power.

gabriktm, 2024-12-29:
> Perfect! Just attach the converter to GPIO1 and GPIO3 + power.

Ok now I will connect everything. Which script do I need to flash?

syssi, 2024-12-29:
Let's start with the protocol test.

gabriktm, 2024-12-29:
> Let's start with the protocol test.

Ok so now I connected pin 2 (gnd) to the gnd  on converter. pin 3 (rs485-B) on B of the converter, and then pin 4 (rs485-a) on A on the converter. Now i start the test and I send logs

gabriktm, 2024-12-29:
@syssi 

Here the logs.

INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/sniffer.yaml...
INFO Starting log output from 192.168.1.214 using esphome API
INFO Successfully connected to sniffer @ 192.168.1.214 in 0.005s
INFO Successful handshake with sniffer @ 192.168.1.214 in 2.372s
[18:56:35][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 28 2024, 12:46:04
[18:56:35][C][wifi:600]: WiFi:
[18:56:35][C][wifi:428]:   Local MAC: B4:E6:2D:70:2B:A0
[18:56:35][C][wifi:433]:   SSID: [redacted]
[18:56:35][C][wifi:436]:   IP Address: 192.168.1.214
[18:56:35][C][wifi:439]:   BSSID: [redacted]
[18:56:35][C][wifi:441]:   Hostname: 'sniffer'
[18:56:35][C][wifi:443]:   Signal strength: -70 dB ▂▄▆█
[18:56:35][C][wifi:447]:   Channel: 2
[18:56:35][C][wifi:448]:   Subnet: 255.255.255.0
[18:56:35][C][wifi:449]:   Gateway: 192.168.1.1
[18:56:35][C][wifi:450]:   DNS1: 0.0.0.0
[18:56:35][C][wifi:451]:   DNS2: 0.0.0.0
[18:56:35][C][logger:185]: Logger:
[18:56:35][C][logger:186]:   Level: DEBUG
[18:56:35][C][logger:188]:   Log Baud Rate: 0
[18:56:35][C][logger:189]:   Hardware UART: UART0
[18:56:35][C][uart.arduino_esp8266:118]: UART Bus:
[18:56:35][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[18:56:35][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[18:56:35][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[18:56:35][C][uart.arduino_esp8266:124]:   Baud Rate: 2400 baud
[18:56:35][C][uart.arduino_esp8266:125]:   Data Bits: 8
[18:56:35][C][uart.arduino_esp8266:126]:   Parity: NONE
[18:56:35][C][uart.arduino_esp8266:127]:   Stop bits: 1
[18:56:35][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[18:56:35][C][mdns:116]: mDNS:
[18:56:35][C][mdns:117]:   Hostname: sniffer
[18:56:35][C][esphome.ota:073]: Over-The-Air updates:
[18:56:35][C][esphome.ota:074]:   Address: 192.168.1.214:8266
[18:56:35][C][esphome.ota:075]:   Version: 2
[18:56:35][C][esphome.ota:078]:   Password configured
[18:56:35][C][safe_mode:018]: Safe Mode:
[18:56:35][C][safe_mode:019]:   Boot considered successful after 60 seconds
[18:56:35][C][safe_mode:021]:   Invoke after 10 boot attempts
[18:56:35][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[18:56:35][C][api:140]: API Server:
[18:56:35][C][api:141]:   Address: 192.168.1.214:6053
[18:56:35][C][api:143]:   Using noise encryption: YES
[18:56:39][I][main:466]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[18:56:39][I][main:469]: This is the set of commands supported by the pipsolar component!
[18:56:39][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[18:56:40][D][uart_debug:158]: >>> "QDIq\e\r"
[18:56:41][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[18:56:42][D][uart_debug:158]: >>> "QMN\xBBd\r"
[18:56:43][D][uart_debug:158]: >>> "QMODI\xC1\r"
[18:56:44][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[18:56:45][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[18:56:46][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[18:56:47][D][uart_debug:158]: >>> "QT\'\xFF\r"
[18:56:48][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[18:56:49][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[18:56:50][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[18:56:51][I][main:556]: Testing PI41 split phase / multiple strings commands...
[18:56:51][D][uart_debug:158]: >>> "QPIGS2h-\r"
[18:56:52][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[18:56:53][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[18:56:53][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[18:56:54][I][main:580]: Testing unsupported PI18 commands...
[18:56:54][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[18:56:55][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[18:56:56][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[18:56:57][I][main:604]: Testing unsupported PI17 commands...
[18:56:57][D][uart_debug:158]: >>> "^P003PI\r"
[18:56:58][D][uart_debug:158]: >>> "^P004MOD\r"
[18:56:59][D][uart_debug:158]: >>> "^P005FLAG\r"
[18:57:00][I][main:628]: Testing unsupported PI16 commands...
[18:57:00][D][uart_debug:158]: >>> "QPI\r"
[18:57:01][D][uart_debug:158]: >>> "QMOD\r"
[18:57:02][D][uart_debug:158]: >>> "QPIGS\r"
[18:57:03][D][uart_debug:158]: >>> "QPIRI\r"
[18:57:04][D][uart_debug:158]: >>> "QMOD\r"
[18:57:04][I][main:662]: Done. Repeating..

syssi, 2024-12-29:
Does the TX LED of the RS485 converter blink?

gabriktm, 2024-12-29:
> Does the TX LED of the RS485 converter blink?

Yes it blink

gabriktm, 2024-12-29:
@syssi sorry I wronged only the RXD led blink

syssi, 2024-12-29:
If the RX LED blinks you have swapped TX & RX.

gabriktm, 2024-12-29:
> If the RX LED blinks you have swapped TX & RX.

Let me fix and I send again logs

gabriktm, 2024-12-29:
@syssi 
[19:11:43][I][main:466]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[19:11:44][I][main:469]: This is the set of commands supported by the pipsolar component!
[19:11:44][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[19:11:44][D][uart_debug:158]: >>> "QDIq\e\r"
[19:11:45][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[19:11:46][D][uart_debug:158]: >>> "QMN\xBBd\r"
[19:11:47][D][uart_debug:158]: >>> "QMODI\xC1\r"
[19:11:48][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[19:11:49][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[19:11:50][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[19:11:51][D][uart_debug:158]: >>> "QT\'\xFF\r"
[19:11:52][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[19:11:53][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[19:11:54][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[19:11:56][I][main:556]: Testing PI41 split phase / multiple strings commands...
[19:11:56][D][uart_debug:158]: >>> "QPIGS2h-\r"
[19:11:56][W][component:237]: Component interval took a long time for an operation (52 ms).
[19:11:56][W][component:238]: Components should block for at most 30 ms.
[19:11:57][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[19:11:57][W][component:237]: Component interval took a long time for an operation (51 ms).
[19:11:57][W][component:238]: Components should block for at most 30 ms.
[19:11:58][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[19:11:58][W][component:237]: Component interval took a long time for an operation (51 ms).
[19:11:58][W][component:238]: Components should block for at most 30 ms.
[19:11:59][I][main:580]: Testing unsupported PI18 commands...
[19:11:59][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[19:11:59][W][component:237]: Component interval took a long time for an operation (56 ms).
[19:11:59][W][component:238]: Components should block for at most 30 ms.
[19:12:00][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[19:12:00][W][component:237]: Component interval took a long time for an operation (55 ms).
[19:12:00][W][component:238]: Components should block for at most 30 ms.
[19:12:01][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[19:12:01][W][component:237]: Component interval took a long time for an operation (59 ms).
[19:12:01][W][component:238]: Components should block for at most 30 ms.
[19:12:02][I][main:604]: Testing unsupported PI17 commands...
[19:12:02][D][uart_debug:158]: >>> "^P003PI\r"
[19:12:03][D][uart_debug:158]: >>> "^P004MOD\r"
[19:12:03][W][component:237]: Component interval took a long time for an operation (51 ms).
[19:12:03][W][component:238]: Components should block for at most 30 ms.
[19:12:04][D][uart_debug:158]: >>> "^P005FLAG\r"
[19:12:04][W][component:237]: Component interval took a long time for an operation (55 ms).
[19:12:04][W][component:238]: Components should block for at most 30 ms.
[19:12:05][I][main:628]: Testing unsupported PI16 commands...
[19:12:05][D][uart_debug:158]: >>> "QPI\r"
[19:12:06][D][uart_debug:158]: >>> "QMOD\r"
[19:12:07][D][uart_debug:158]: >>> "QPIGS\r"
[19:12:08][D][uart_debug:158]: >>> "QPIRI\r"
[19:12:09][D][uart_debug:158]: >>> "QMOD\r"
[19:12:09][I][main:662]: Done. Repeating...

gabriktm, 2024-12-29:
@syssi now only TXD blink. It does sometimes and sometimes not, not frequently

syssi, 2024-12-29:
This is fine. As next step lets try the SMG YAML.

gabriktm, 2024-12-29:
> This is fine. As next step lets try the SMG YAML.

Really thanks for the help you are giving to me! Where can I find this yaml? :)

syssi, 2024-12-29:
I'm talking about this one: https://github.com/syssi/esphome-smg-ii/blob/main/esp8266-example.yaml

If you change the `{tx,rx}_pin` to `GPIO1/GPIO3` you have to disable the UART output of the logger again:

```
logger:
  baud_rate: 0
```

gabriktm, 2024-12-29:
> I'm talking about this one: https://github.com/syssi/esphome-smg-ii/blob/main/esp8266-example.yaml
> 
> If you change the `{tx,rx}_pin` to `GPIO1/GPIO3` you have to disable the UART output of the logger again:
> 
> ```
> logger:
>   baud_rate: 0
> ```

Ok i flashed this script putting this as logger: 

[19:38:00][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 29 2024, 19:33:08
[19:38:00][I][app:102]: Project syssi.esphome-smg-ii version 1.2.0
[19:38:00][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii output voltage'
[19:38:00][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii output frequency setting'
[19:38:00][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery overvoltage protection point'
[19:38:00][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii max charging voltage'
[19:38:00][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii floating charging voltage'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery discharge recovery point in mains mode'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery low voltage protection point in mains mode'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii battery low voltage protection point in off-grid mode'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii maximum charging current'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii maximum mains charging current'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii Eq Charging voltage'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii equalization Timeout exit'
[19:38:01][C][modbus.number:083]: modbus.numberModbus Number 'smg-ii two equalization charging intervals'
[19:38:01][C][modbus_controller:349]: ModbusController:
[19:38:01][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii fault code'
[19:38:01][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii warning code'
[19:38:01][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii operation mode id'
[19:38:01][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii ac voltage'
[19:38:01][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii ac frequency'
[19:38:01][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii average mains power'
[19:38:02][C][modbus_controller.sensor:010]: modbus_controller.sensorModbus Controller Sensor 'smg-ii effective inverter voltage'
[19:38:03][D][modbus_controller:039]: Modbus command to device=1 register=0x131 no response received - removed from send queue
[19:38:04][D][modbus_controller:039]: Modbus command to device=1 register=0x283 no response received - removed from send queue
[19:38:05][D][modbus_controller:039]: Modbus command to device=1 register=0x64 no response received - removed from send queue
[19:38:06][D][modbus_controller:039]: Modbus command to device=1 register=0x6C no response received - removed from send queue
[19:38:06][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=313 count=1
[19:38:06][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=320 count=2
[19:38:07][D][modbus_controller:039]: Modbus command to device=1 register=0x139 no response received - removed from send queue
[19:38:08][D][modbus_controller:039]: Modbus command to device=1 register=0x140 no response received - removed from send queue
[19:38:09][D][modbus_controller:039]: Modbus command to device=1 register=0x143 no response received - removed from send queue
[19:38:11][D][modbus_controller:039]: Modbus command to device=1 register=0x149 no response received - removed from send queue
[19:38:12][D][modbus_controller:039]: Modbus command to device=1 register=0x14B no response received - removed from send queue
[19:38:13][D][modbus_controller:039]: Modbus command to device=1 register=0x196 no response received - removed from send queue
[19:38:14][D][modbus_controller:039]: Modbus command to device=1 register=0x1A4 no response received - removed from send queue
[19:38:15][D][modbus_controller:039]: Modbus command to device=1 register=0x64 no response received - removed from send queue
[19:38:16][D][modbus_controller:039]: Modbus command to device=1 register=0x6C no response received - removed from send queue
[19:38:16][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=201 count=17
[19:38:16][W][modbus_controller:185]: Duplicate modbus command found: type=0x3 address=225 count=3
[19:38:17][D][modbus_controller:039]: Modbus command to device=1 register=0xC9 no response received - removed from send queue
[19:38:18][D][modbus_controller:039]: Modbus command to device=1 register=0xE1 no response received - removed from send queue
[19:38:19][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[19:38:19][D][modbus_controller:039]: Modbus command to device=1 register=0xE5 no response received - removed from send queue
[19:38:20][D][modbus_controller:039]: Modbus command to device=1 register=0xE8 no response received - removed from send queue
[19:38:22][D][modbus_controller:039]: Modbus command to device=1 register=0x12C no response received - removed from send queue
[... cut ...]
[19:39:01][D][modbus_controller:039]: Modbus command to device=1 register=0x12C no response received - removed from send queue
[19:39:02][D][modbus_controller:039]: Modbus command to device=1 register=0x131 no response received - removed from send queue
[19:39:03][D][modbus_controller:039]: Modbus command to device=1 register=0x283 no response received - removed from send queue
[19:39:04][D][modbus_controller:039]: Modbus command to device=1 register=0x64 no response received - removed from send queue

syssi, 2024-12-29:
Does the TX LED of the RS485 converter blink again using this YAML?

syssi, 2024-12-29:
Could you try to provide some more details about your inverter? A photo of the name plate f.e.?

gabriktm, 2024-12-29:
> Does the TX LED of the RS485 converter blink again using this YAML?

Yes only the TX. It flashes continuously. Now I will send pictures of the inverter

gabriktm, 2024-12-29:
@syssi 

This is the inverter, as you can see the screen is the same as voltronic with the 4 buttons:

[image]
[image]

[image]

[image]

[image]

This is the board where i plug RJ45:

[image]

[image]

[image]

And then where the RJ45 board is plugged:

[image]

[image]

[image]

[image]

gabriktm, 2024-12-29:
@syssi Is there any way to sniff the protocol?

syssi, 2024-12-29:
Do you have any other software which is able to talk to the Inverter?

gabriktm, 2024-12-29:
> Do you have any other software which is able to talk to the Inverter?

I have the RS485 to usb cable for the pc

syssi, 2024-12-29:
Do you have some windows application to talk to the inverter?

gabriktm, 2024-12-29:
> Do you have some windows application to talk to the inverter?

Sent from the productor not. Is there something that I can try?

syssi, 2024-12-29:
[image]

This is a RS485 driver: https://www.ti.com/lit/ds/symlink/sn75179b.pdf

Long story short: The left RJ45 port at the picture is RS485 for sure.

syssi, 2024-12-29:
Could you double check these pins are connected to the RJ45 pins you are using at the moment?

[image]

gabriktm, 2024-12-29:
> [image]
> 
> 
> 
> This is a RS485 driver: https://www.ti.com/lit/ds/symlink/sn75179b.pdf
> 
> 
> 
> Long story short: The left RJ45 port at the picture is RS485 for sure.

Yes it's RS485 100% but the problem is try to understand the protocol

gabriktm, 2024-12-29:
> Could you double check these pins are connected to the RJ45 pins you are using at the moment?
> 
> 
> 
> [image]
> 
> 

No im not using that pins
[image]

On documentation of the inverter I need to use pin 2/3/4

So im using these pins:

[image]

syssi, 2024-12-29:
I cannot identify pin1 of the RS485 driver. Could you try to identify the GND pin of the chip?

gabriktm, 2024-12-29:
> I cannot identify pin1 of the RS485 driver. Could you try to identify the GND pin of the chip? 

The chip is SN75176B

https://www.alldatasheet.com/datasheet-pdf/download/192858/TI/75176B.html

syssi, 2024-12-29:
Aaaah. This is the correct datasheet: https://www.ti.com/lit/ds/symlink/sn75176b.pdf

[image]

Could you measure pin6 und pin7 is connected to pin3&4 of the RJ45?

syssi, 2024-12-29:
Alright! We have verified the pin-out now and can be sure the RS485+ESP is connected to the correct pins.

I've trouble to find any details about your device in the internet. Could you provide any links? Can we be sure your inverter can  be interfaced via software and the RS485 port isn't for a external display only?

gabriktm, 2024-12-29:
> Alright! We have verified the pin-out now and can be sure the RS485+ESP is connected to the correct pins.
>
>
>
> I've trouble to find any details about your device in the internet. Could you provide any [...]

The inverter is called "RP series" bought from alibaba. The strange thing is that the producer doesn't know the protocol or any info about this rs485 port

https://m.alibaba.com/x/xCQQg8W?ck=pdp

I found the rj45 board of the inverter on aliexpress 

https://it.aliexpress.com/item/1005006436729240.html

Even if it's for an external display we should be able to read data like voltag, current ecc. no?

[125 more replies]

closed 2025-01-01 by syssi