syssi/esphome-pipsolar #169: Sorotec 8kW - not complete data

HA1Andrzej, 2025-01-09:
Hi, 
Successfully started reading on Sorotec 8kW inverter, but not all data is correct and some is missing.
Sample data ( as you can see PV input and pv charge is the same ) 

[image]

For communication I use ( I also tested others but without success) 
https://github.com/syssi/esphome-pipsolar/blob/main/esp32-example.yaml

Below I am uploading the logs I was able to read from the inverter: 

> INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/test2.yaml...
INFO Starting log output from 192.168.5.241 using esphome API
INFO Successfully connected to pipsolar @ 192.168.5.241 in 0.078s
INFO Successful handshake with pipsolar @ 192.168.5.241 in 0.082s
[14:10:00][I][app:100]: ESPHome version 2024.12.2 compiled on Jan  9 2025, 10:35:44
[14:10:00][C][wifi:600]: WiFi:
[14:10:00][C][wifi:428]:   Local MAC: 20:43:A8:64:4B:80
[14:10:00][C][wifi:433]:   SSID: 'Daikin'[redacted]
[14:10:00][C][wifi:436]:   IP Address: 192.168.5.241
[14:10:00][C][wifi:440]:   BSSID: FA:A3:4F:6E:2A:3A[redacted]
[14:10:00][C][wifi:441]:   Hostname: 'pipsolar'
[14:10:00][C][wifi:443]:   Signal strength: -66 dB ▂▄▆█
[14:10:00][C][wifi:447]:   Channel: 1
[14:10:00][C][wifi:448]:   Subnet: 255.255.255.0
[14:10:00][C][wifi:449]:   Gateway: 192.168.5.1
[14:10:00][C][wifi:450]:   DNS1: 192.168.5.1
[14:10:00][C][wifi:451]:   DNS2: 0.0.0.0
[14:10:00][C][logger:185]: Logger:
[14:10:00][C][logger:186]:   Level: DEBUG
[14:10:00][C][logger:188]:   Log Baud Rate: 115200
[14:10:00][C][logger:189]:   Hardware UART: UART0
[14:10:00][C][uart.arduino_esp32:151]: UART Bus 1:
[14:10:00][C][uart.arduino_esp32:152]:   TX Pin: GPIO16
[14:10:00][C][uart.arduino_esp32:153]:   RX Pin: GPIO17
[14:10:00][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[14:10:00][C][uart.arduino_esp32:157]:   Baud Rate: 2400 baud
[14:10:00][C][uart.arduino_esp32:158]:   Data Bits: 8
[14:10:00][C][uart.arduino_esp32:159]:   Parity: NONE
[14:10:00][C][uart.arduino_esp32:160]:   Stop bits: 1
[14:10:00][C][mdns:116]: mDNS:
[14:10:00][C][mdns:117]:   Hostname: pipsolar
[14:10:00][C][esphome.ota:073]: Over-The-Air updates:
[14:10:00][C][esphome.ota:074]:   Address: pipsolar.local:3232
[14:10:00][C][esphome.ota:075]:   Version: 2
[14:10:00][C][esphome.ota:078]:   Password configured
[14:10:00][C][safe_mode:018]: Safe Mode:
[14:10:00][C][safe_mode:020]:   Boot considered successful after 60 seconds
[14:10:00][C][safe_mode:021]:   Invoke after 10 boot attempts
[14:10:00][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[14:10:00][C][api:140]: API Server:
[14:10:00][C][api:141]:   Address: pipsolar.local:6053
[14:10:00][C][api:143]:   Using noise encryption: YES
[14:10:08][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[14:10:08][D][esp32.preferences:114]: Saving 1 preferences to flash...
[14:10:08][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[14:10:13][I][main:497]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[14:10:13][I][main:500]: This is the set of commands supported by the pipsolar component!
[14:10:13][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[14:10:13][D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[14:10:14][D][uart_debug:158]: >>> "QDIq\e\r"
[14:10:15][D][uart_debug:158]: <<< "(230.0 50.0 0000 42.0 54.0 56.4 46.0 60 0 0 2 0 0 0 0 0 0 1 0 0 1 0 54 0 0 1l\x88\r"
[14:10:15][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[14:10:15][D][uart_debug:158]: <<< "(EakxyDbjuvz/)\r"
[14:10:16][D][uart_debug:158]: >>> "QMN\xBBd\r"
[14:10:16][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:17][D][uart_debug:158]: >>> "QMODI\xC1\r"
[14:10:17][D][uart_debug:158]: <<< "(L\x06\a\r"
[14:10:18][D][uart_debug:158]: <<< "\xFF"
[14:10:18][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[14:10:19][D][uart_debug:158]: <<< "(225.0 49.9 225.0 49.9 0000 2094 026 401 58.10 000 100 0026 0221 226.0 58.10 00000 00010110 00 00 00221 010$\x16\r"
[14:10:19][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[14:10:20][D][uart_debug:158]: <<< "(230.0 20.0 230.0 50.0 20.0 3600 3600 48.0 51.0 42.3 58.0 58.0 2 60 06P 1 0 1 6 01 0 0 52.0 0 1\xFA\xE3\r"
[14:10:20][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[14:10:21][D][uart_debug:158]: <<< "(00000100000000000000000000000000\xFE\x82\r"
[14:10:21][D][uart_debug:158]: >>> "QT\'\xFF\r"
[14:10:22][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:22][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[14:10:23][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:23][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[14:10:24][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:24][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[14:10:25][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:25][I][main:587]: Testing PI41 split phase / multiple strings commands...
[14:10:25][D][uart_debug:158]: >>> "QPIGS2h-\r"
[14:10:26][D][uart_debug:158]: <<< "(224.0 49.9 224.0 49.9 0000 2108 026 401 58.00 000 100 0026 0221 227.0 58.00 00000 00010110 00 00 00221 010\x84d\r"
[14:10:27][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[14:10:27][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:28][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[14:10:28][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:29][I][main:611]: Testing unsupported PI18 commands...
[14:10:29][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[14:10:29][D][uart_debug:158]: <<< "^D00518;\x03\r"
[14:10:30][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[14:10:30][D][uart_debug:158]: <<< "^D1062250,499,2240,499,2116,2117,026,581,581,581,000,000,100,033,000,000,0091,0131,2280,2630,0,0,2,1,1,1,2,0\x1F\xCA\r"
[14:10:31][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[14:10:31][D][uart_debug:158]: <<< "^D00505\xD9\x9F\r"
[14:10:32][I][main:635]: Testing unsupported PI17 commands...
[14:10:32][D][uart_debug:158]: >>> "^P003PI\r"
[14:10:32][D][uart_debug:158]: <<< "0\e\xE3\r"
[14:10:33][D][uart_debug:158]: >>> "^P004MOD\r"
[14:10:33][D][uart_debug:158]: <<< "0\e\xE3\r"
[14:10:34][D][uart_debug:158]: >>> "^P005FLAG\r"
[14:10:34][D][uart_debug:158]: <<< "0\e\xE3\r"
[14:10:35][I][main:659]: Testing unsupported PI16 commands...
[14:10:35][D][uart_debug:158]: >>> "QPI\r"
[14:10:35][D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[14:10:36][D][uart_debug:158]: >>> "QMOD\r"
[14:10:36][D][uart_debug:158]: <<< "(L\x06\a\r"
[14:10:37][D][uart_debug:158]: >>> "QPIGS\r"
[14:10:37][D][uart_debug:158]: <<< "(225.0 49.9 225.0 49.9 0000 2136 026 401 58.10 000 100 0026 0222 228.0 58.10 00000 00010110 00 00 00222 010\xF4/\r"
[14:10:38][D][uart_debug:158]: >>> "QPIRI\r"
[14:10:38][D][uart_debug:158]: <<< "(230.0 20.0 230.0 50.0 20.0 3600 3600 48.0 51.0 42.3 58.0 58.0 2 60 06P 1 0 1 6 01 0 0 52.0 0 1\xFA\xE3\r"
[14:10:39][D][uart_debug:158]: >>> "QMOD\r"
[14:10:39][I][main:693]: Testing unsupported Qx commands...
[14:10:39][D][uart_debug:158]: >>> "Q1\r"
[14:10:39][D][uart_debug:158]: <<< "(L\x06\a\r"
[14:10:40][D][uart_debug:158]: >>> "Q2\r"
[14:10:40][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:41][D][uart_debug:158]: >>> "Q4\r"
[14:10:41][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:42][D][uart_debug:158]: >>> "Q6\r"
[14:10:42][D][uart_debug:158]: <<< "(NAKss\r"
[14:10:43][I][main:724]: Done. Repeating...

How do you think it will be possible to extract the correct data ? It would also be helpful to be able to take readings from PV1 and PV2. 
Thanks for your help.

wojtbar, 2025-01-14:
Hi, 
Sorotec 8 kW has two mppt.
Change in yaml file:

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

wojtbar, 2025-01-14:
Customize the .yaml file to :
https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml