syssi/esphome-pipsolar #177: HMS-3K-24V Wrong Status Map in Charging Source Priority "Solar charging only"

ferilagi, 2025-03-06:
my inverter HMS-3K-24V, my yaml code
`substitutions:
  name: hms
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  friendly_name: hms-inverter
  comment: "Monitor and control a HMS inverter"

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
# mqtt:
  # broker: !secret mqtt_host
  # username: !secret mqtt_username
  # password: !secret mqtt_password
  # id: mqtt_client

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.55.50.225
    gateway: 10.55.50.1
    subnet: 255.255.255.0
  domain: .invismesagi.hotspot

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hms-Esp32 Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:

uart:
  - id: uart_0
    baud_rate: 2400
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
#    debug:
#      direction: BOTH
#      dummy_receiver: false
#      after:
#        delimiter: "\r"
#      sequence:
#        - lambda: UARTDebug::log_string(direction, bytes);
[... 249 lines cut ...]
    current_max_ac_charging_current:
      name: "${name} current_max_ac_charging_current_select"
      optionsmap:
#        "2A": "MUCHGC0002"
#        "10A": "MUCHGC0010"
        "20A": "MUCHGC0020"
        "30A": "MUCHGC0030"
      statusmap:
#        "2": "2A"
#        "10": "10A"
        "20": "20A"
        "30": "30A"

#-------------------------------------------
# TIME COMPONENT
#-------------------------------------------
time:
  - platform: homeassistant

`

when changing via home assistant select input, all working without problem when i check in actual inverter,
but the problem when changing charger_source_priority to Solar charging only, always return "solar first" not "solar charging only"

[image]

rgnyldz, 2025-03-21:
I'm also trying to use an HMS-3K-24V with this branch but I get the below error. @ferilagi  Can you share your yaml code as a file for reference?

```
INFO ESPHome 2025.3.0
INFO Reading configuration /config/esphome/solar-logger-c6.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@hms-3k-24v
ERROR Unable to load component pipsolar.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/fbb7e816/components/pipsolar/sensor/__init__.py", line 76, in <module>
    CONF_GRID_RATING_VOLTAGE: sensor.sensor_schema(
                              ^^^^^^^^^^^^^^^^^^^^^
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 4 were given
Failed config

sensor.pipsolar: [source /config/esphome/solar-logger-c6.yaml:61]
  
  Platform not found: 'sensor.pipsolar'.
  platform: pipsolar
  pipsolar_id: inverter0
  grid_voltage: 
    name: solar-logger-c6 grid_voltage
  grid_frequency: 
    name: solar-logger-c6 grid_frequency
  ac_output_voltage: 
    name: solar-logger-c6 ac_output_voltage
  ac_output_frequency: 
    name: solar-logger-c6 ac_output_frequency
  ac_output_apparent_power: 
    name: solar-logger-c6 ac_output_apparent_power
```

ferilagi, 2025-03-24:
@rgnyldz sure, i use this yaml
```
substitutions:
  name: hms
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  friendly_name: hms-inverter
  comment: "Monitor and control a HMS inverter"

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
# mqtt:
  # broker: !secret mqtt_host
  # username: !secret mqtt_username
  # password: !secret mqtt_password
  # id: mqtt_client

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.55.50.225
    gateway: 10.55.50.1
    subnet: 255.255.255.0
  domain: .invismesagi.hotspot

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hms-Esp32 Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

captive_portal:

uart:
  - id: uart_0
    baud_rate: 2400
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
#    debug:
#      direction: BOTH
#      dummy_receiver: false
#      after:
#        delimiter: "\r"
#      sequence:
[... 239 lines cut ...]
      statusmap:
#        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
#        "70": "70A"

  - platform: pipsolar
    pipsolar_id: inverter0
    current_max_ac_charging_current:
      name: "${name} current_max_ac_charging_current_select"
      optionsmap:
#        "2A": "MUCHGC0002"
#        "10A": "MUCHGC0010"
        "20A": "MUCHGC0020"
        "30A": "MUCHGC0030"
      statusmap:
#        "2": "2A"
#        "10": "10A"
        "20": "20A"
        "30": "30A"

```

closed 2025-07-13 by syssi