syssi/esphome-pipsolar #179: M5Stack ATOM Lite ESP32

blucabar, 2025-03-27:
Hi, congratulations on your skill and thanks for the help you give us. I have a Voltronic Axpert Max Twin 11Kw and i wanted to ask you if the "yaml" file used to program the normal "ESP32" is the same to program a M5Stack ATOM Lite ESP32? I use a normal ESP32 and it works well (the 2A current charging doesn't work and it doesn't put the inverter in standby), then I tried to program the Atom, but I couldn't. In the end it gives me an error on HA and says "api" doesn't exist.......What can you tell me about it?

syssi, 2025-03-27:
Please change the `board` setting to `board: m5stack-core-esp32`:

```
esp32:
  board: m5stack-core-esp32
  framework:
    type: esp-idf
```

And press the "clean build files" button before build the setup another time. Does it work now?

blucabar, 2025-03-27:
sorry if I take advantage, but I'm new to HA....Where is "clean build files button"?? Thanks

PS: which yaml do you recommend for the Atom? It is important that it has 2A charging enabled and above all the inverter switching off and on.........

syssi, 2025-03-27:
Could you explain how did you start? There are so many options.

syssi, 2025-03-27:
This is the button of the ESPHome dashboard I'm talking about:

[image]

blucabar, 2025-03-27:
1)
I would like to start from the yaml file that works. the one you posted of 350 lines is not read. The one I post later works well for me, but it does not have the 2A AC charging and above all it does not have the standby, useful for eliminating self-consumption......
Which yaml file do you recommend?
Coffee guaranteed!!!!!

syssi, 2025-03-27:
Could you post the working YAML configuration?

blucabar, 2025-03-27:
substitutions:
  name: voltronic
  friendly_name: Voltronic Axpert Max 8000 RS-232

esphome:
  name: ${name}
  comment: "Voltronic Axpert Max 8000 PIP30 protocol"
  platformio_options:
    upload_speed: 115200
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768

esp32:
  board: esp32dev

# Enable Home Assistant API
api:
  encryption:
    key: "L3eUipnck1YD+bl3JxzkEFabJSI8UOQlQGBsELgJBus="

ota:
  - platform: esphome
    password: "e4800408aac15d787290af812f509762"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Voltronic-Test Fallback Hotspot"
    password: "JZnJoHRyedIl"

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s    

# Enable logging
logger:
    baud_rate: 0

#mqtt:
   #broker: !secret mqtt_host
   #username: !secret mqtt_username
   #password: !secret mqtt_password
   #id: mqtt_client

uart:
  id: uart_bus
  baud_rate: 2400
  tx_pin: GPIO16
  rx_pin: GPIO17

pipsolar:
  uart_id: uart_bus
  id: inverter0

sensor:
  # QPIRI
  - platform: pipsolar
    pipsolar_id: inverter0
    grid_rating_voltage:
      name: "${name} grid_rating_voltage"
    grid_rating_current:
      name: "${name} grid_rating_current"
    ac_output_rating_voltage:
      name: "${name} ac_output_rating_voltage"
    ac_output_rating_frequency:
      name: "${name} ac_output_rating_frequency"
    ac_output_rating_current:
[... 266 lines cut ...]
        "140": "140A"
######Aggiunte per Inverter KING #######
 # - platform: pipsolar
 #   pipsolar_id: inverter0
 #  output_source_priority:
 #     id: inverter0_operation_logic_select
 #     name: inverter0_operation_logic_select
 #     optionsmap:
 #       "Auto": "POPLG00"
 #       "On Line": "POPLG01"
 #       "ECO": "POPLG02"
 #     statusmap:
 #       "0": "Auto"
 #       "1": "On Line"
 #       "2": "ECO"
 
#text_sensor:
#    operation_logic:
#      id: ${device_id}_operation_logic
#      name: Operation Logic
#      filters:
#      - map:
#        - 0 -> Auto
#        - 1 -> On-Line
#        - 2 -> Eco

blucabar, 2025-03-27:
Sorry for the different size of font.....

syssi, 2025-03-27:
To improve your setup you can change this section:

```
esp32:
  board: esp32dev
```

to

```
esp32:
  board: m5stack-core-esp32
  framework:
    type: esp-idf
```

It wouldn't change much. The different framework (`ESP-IDF`) can improve the stability.

syssi, 2025-03-27:
Please update this section:

```
  - platform: pipsolar
    pipsolar_id: inverter0
    current_max_ac_charging_current:
      name: "${name} current_max_ac_charging_current_select"
      optionsmap:
        "2A": "MUCHGC002"
        "10A": "MUCHGC0010"
        "20A": "MUCHGC0020"
        "30A": "MUCHGC0030"
        "40A": "MUCHGC0040"
        "50A": "MUCHGC0050"
        "60A": "MUCHGC0060"
        
      statusmap:
        "2" : "2"
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
```

to

```
  - platform: pipsolar
    pipsolar_id: inverter0
    current_max_ac_charging_current:
      name: ${name} current_max_ac_charging_current_select
      optionsmap:
        "2A": "MUCHGC0002"
        "10A": "MUCHGC0010"
        "20A": "MUCHGC0020"
        "30A": "MUCHGC0030"
        "40A": "MUCHGC0040"
        "50A": "MUCHGC0050"
        "60A": "MUCHGC0060"
      statusmap:
        "2": "2A"
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
```

Does this do the job now?

blucabar, 2025-03-27:
# Warning:
#
# If you configure a lot of the possible sensors etc. it could be that you run
# out of memory (on esp8266). If you configure nearly all sensors etc. you run
# in a stack-size issue. In this case you have to increase stack size!
#
#  https://github.com/esphome/issues/issues/855

substitutions:
  name: pipsolar

esphome:
  name: ${name}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pipsolar@pip8084"
    version: 1.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  baud_rate: 0

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

uart:
  - id: uart_0
    baud_rate: 2400
    tx_pin: GPIO1
    rx_pin: GPIO3

pipsolar:
  uart_id: uart_0
  id: inverter0

sensor:
  - platform: pipsolar
    pipsolar_id: inverter0
    # QPIRI
#    grid_rating_voltage:
#      name: "${name} grid_rating_voltage"
#    grid_rating_current:
#      name: "${name} grid_rating_current"
#    ac_output_rating_voltage:
#      name: "${name} ac_output_rating_voltage"
#    ac_output_rating_frequency:
#      name: "${name} ac_output_rating_frequency"
#    ac_output_rating_current:
#      name: "${name} ac_output_rating_current"
#    ac_output_rating_apparent_power:
#      name: "${name} ac_output_rating_apparent_power"
#    ac_output_rating_active_power:
[... 255 lines cut ...]
        "10A": "MCHGC010"
        "20A": "MCHGC020"
        "30A": "MCHGC030"
        "40A": "MCHGC040"
        "50A": "MCHGC050"
        "60A": "MCHGC060"
        "70A": "MCHGC070"
        "80A": "MCHGC080"
        "90A": "MCHGC090"
        "100A": "MCHGC100"
        "110A": "MCHGC110"
        "120A": "MCHGC120"
      statusmap:
        "10": "10A"
        "20": "20A"
        "30": "30A"
        "40": "40A"
        "50": "50A"
        "60": "60A"
        "70": "70A"
        "80": "80A"
        "90": "90A"
        "100": "100A"
        "110": "110A"
        "120": "120A"

syssi, 2025-03-27:
Why do you post a different YAML configuration this time?

blucabar, 2025-03-27:
Sorry. I think that its better...

blucabar, 2025-03-27:
To recap, do I take the file that works and make the changes you told me?

syssi, 2025-03-27:
Should should pick the file which fits to your inverter.

blucabar, 2025-03-27:
Unfortunately I don't know the code for the command....My question is to have the best yaml file for my Axpert Max twin 11Kw, which I don't know....

syssi, 2025-03-27:
I cannot help here too. I don't know your inverter.

blucabar, 2025-03-27:
Ok my friend, now i prove what you say and i tell you......

See you later....

blucabar, 2025-03-28:
Hi, dont operate!!

When i found the Esphome devices, say "impossible to connect to a ESP. Be sure that your file yaml have "api"..

[image]

[image]

syssi, 2025-03-28:
Are you sure about the IP address?

blucabar, 2025-03-28:
The ip address that you see is the IP of the Atom. Inside the yaml i dont have insert the ip address because in the file there is no space to it.....

To program the Atom i use Crome with the link "https://web.esphome.io/"

syssi, 2025-03-28:
You can retrieve the log of the ESP using a USB connection. Could you provide the log?

blucabar, 2025-03-28:
i dont know where i can found the log of the ESP using the usb connection.....Now i prove to see......

blucabar, 2025-03-29:
could you briefly explain to me how it's done?

syssi, 2025-03-29:
If you use `web.esphome.io` to flash the device there should be a `log & console` button right after the flash process. Is my assumption correct: You are flashing the ESP using a USB connection?

blucabar, 2025-03-30:
Yes, i use a USB connection to flash the Atom.....

syssi, 2025-03-30:
Did you retrieve the log?

blucabar, 2025-03-31:
[logs.txt](https://github.com/user-attachments/files/19531387/logs.txt)

blucabar, 2025-04-03:
did you have to see my log?

syssi, 2025-04-04:
It's a boot loop but the root cause isn't reported unfortunately

```
[09:37:46]ets Jun  8 2016 00:22:57
[09:37:46]
[09:37:46]rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[09:37:46]configsip: 188777542, SPIWP:0xee
[09:37:46]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[09:37:46]mode:DIO, clock div:1
[09:37:46]load:0x3fff0030,len:7112
[09:37:46]load:0x40078000,len:15624
[09:37:46]load:0x40080400,len:4
[09:37:46]load:0x40080404,len:3876
[09:37:46]entry 0x4008064c
[09:37:46]I (59) boot: ESP-IDF 5.1.5 2nd stage bootloader
[09:37:46]I (59) boot: compile time Mar 28 2025 12:42:50
[09:37:46]I (59) boot: Multicore bootloader
[09:37:46]I (64) boot: chip revision: v1.1
[09:37:46]I (68) boot.esp32: SPI Speed      : 80MHz
[09:37:46]I (73) boot.esp32: SPI Mode       : DIO
[09:37:46]I (77) boot.esp32: SPI Flash Size : 4MB
[09:37:46]W (82) boot.esp32: PRO CPU has been reset by WDT.
[09:37:46]W (87) boot.esp32: WDT reset info: PRO CPU PC=0x4008210d
[09:37:46]W (93) boot.esp32: WDT reset info: APP CPU PC=0x4008210d
[09:37:46]I (99) boot: Enabling RNG early entropy source...
[09:37:46]I (104) boot: Partition Table:
[09:37:46]I (108) boot: ## Label            Usage          Type ST Offset   Length
[09:37:46]I (115) boot:  0 otadata          OTA data         01 00 00009000 00002000
[09:37:46]I (123) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[09:37:46]I (131) boot:  2 app0             OTA app          00 10 00010000 001c0000
[09:37:46]I (138) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[09:37:46]I (146) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[09:37:46]I (153) boot: End of partition table
[09:37:46]I (158) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2f060h (192608) map
[09:37:46]I (224) esp_image: segment 1: paddr=0003f088 vaddr=3ffb0000 size=00f90h (  3984) load
[09:37:46]I (226) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=a7e78h (687736) map
[09:37:46]I (437) esp_image: segment 3: paddr=000e7ea0 vaddr=3ffb0f90 size=02b7ch ( 11132) load
[09:37:46]I (441) esp_image: segment 4: paddr=000eaa24 vaddr=40080000 size=15af0h ( 88816) load
[09:37:46]I (485) boot: Loaded app from partition at offset 0x10000
[09:37:46]I (485) boot: Disabling RNG early entropy source...
[09:37:46]I (497) cpu_start: Multicore app
[09:37:46]I (498) cpu_start: Pro cpu up.
[09:37:46]I (498) cpu_start: Starting app cpu, entry point is 0x40081cd0
[09:37:46]I (0) cpu_start: App cpu up.
[09:37:46]I (515) cpu_start: Pro cpu start user code
[09:37:46]I (515) cpu_start: cpu freq: 160000000 Hz
[09:37:46]I (515) cpu_start: Application information:
[09:37:46]I (520) cpu_start: Project name:     voltronic
[09:37:46]I (525) cpu_start: App version:      2025.3.2
[09:37:46]I (530) cpu_start: Compile time:     Mar 28 2025 12:39:21
[09:37:46]I (536) cpu_start: ELF file SHA256:  68e1c5ac59e57637...
[09:37:46]I (542) cpu_start: ESP-IDF:          5.1.5
[09:37:46]I (547) cpu_start: Min chip rev:     v0.0
[09:37:46]I (552) cpu_start: Max chip rev:     v3.99 
[09:37:46]I (557) cpu_start: Chip rev:         v1.1
[09:37:46]I (561) heap_init: Initializing. RAM available for dynamic allocation:
[09:37:46]I (568) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[09:37:46]I (575) heap_init: At 3FFB8150 len 00027EB0 (159 KiB): DRAM
[09:37:46]I (581) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[09:37:46]I (587) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[09:37:46]I (594) heap_init: At 40095AF0 len 0000A510 (41 KiB): IRAM
[09:37:46]I (601) spi_flash: detected chip: gd
[09:37:46]I (604) spi_flash: flash io: dio
[09:37:46]I (609) app_start: Starting scheduler on CPU0
[09:37:46]I (613) app_start: Starting scheduler on CPU1
[09:37:46]I (613) main_task: Started on CPU0
[09:37:46]I (623) main_task: Calling app_main()
##### Crash! ####
[09:37:46]ets Jun  8 2016 00:22:57
[09:37:46]
[... 57 lines cut ...]
[09:37:46]I (604) spi_flash: flash io: dio
[09:37:46]I (609) app_start: Starting scheduler on CPU0
[09:37:46]I (613) app_start: Starting scheduler on CPU1
[09:37:46]I (613) main_task: Started on CPU0
[09:37:46]I (623) main_task: Calling app_main()
##### Crash! ####
```

Does it work if you stick to:

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

and if you avoid using:

```
esp32:
  board: m5stack-core-esp32
  framework:
    type: esp-idf
```