syssi/esphome-pipsolar #180: Inverter with 2 MPPT

ribalki, 2025-03-27:
Hi, congratulations on your skill and thanks for the help you give us.
I have a Inverter ARNEN 11KW, I'm using your code ESP32 example. It seems that the PV2 is not read and I'm not a expert to understand where is the problem.
Please see below my log and screen shot.
Thanks in advance for your help.

My log ESP32

INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/pipsolar.yaml...
INFO Updating https://github.com/syssi/esphome-pipsolar.git@pip8048
INFO Starting log output from pipsolar/debug
INFO Connected to MQTT broker!
[17:11:40][D][pipsolar:924]: Sending polling command : QPIRI with length 5
[17:11:40][D][pipsolar:860]: checking crc on incoming message
[17:11:40][D][pipsolar:863]: CRC OK
[17:11:40][D][pipsolar:479]: Decode QPIRI
[17:11:40][D][pipsolar.select:031]: got value: 50
[17:11:40][D][pipsolar.select:033]: found mapped option 50A for option 50
[17:11:40][D][select:015]: 'pipsolar current_max_ac_charging_current_select': Sending state 50A (index 5)
[17:11:40][D][pipsolar.select:031]: got value: 60
[17:11:40][D][pipsolar.select:033]: found mapped option 60A for option 60
[17:11:40][D][select:015]: 'pipsolar current_max_charging_current_select': Sending state 60A (index 5)
[17:11:40][D][pipsolar.select:031]: got value: 1
[17:11:40][D][pipsolar.select:033]: found mapped option Solar only for option 1
[17:11:40][D][select:015]: 'inverter0_output_source_priority_select': Sending state Solar only (index 1)
[17:11:40][D][pipsolar.select:031]: got value: 0
[17:11:40][D][pipsolar.select:036]: could not find option 0 in mapping
[17:11:41][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[17:11:41][D][pipsolar:860]: checking crc on incoming message
[17:11:41][D][pipsolar:863]: CRC OK
[17:11:42][D][pipsolar:924]: Sending polling command : QPIGS with length 5
[17:11:42][D][pipsolar:860]: checking crc on incoming message
[17:11:42][D][pipsolar:863]: CRC OK
[17:11:42][D][pipsolar:502]: Decode QPIGS
[17:11:42][D][sensor:093]: 'pipsolar grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar ac_output_voltage': Sending state 230.20000 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar ac_output_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar ac_output_apparent_power': Sending state 276.00000 VA with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar ac_output_active_power': Sending state 217.00000 W with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar output_load_percent': Sending state 3.00000 % with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar bus_voltage': Sending state 464.00000 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar battery_voltage': Sending state 57.60000 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar battery_capacity_percent': Sending state 100.00000 % with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar inverter_heat_sink_temperature': Sending state 49.00000 °C with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar pv1_input_current': Sending state 1.50000 A with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar pv1_input_voltage': Sending state 401.70001 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[17:11:42][D][sensor:093]: 'pipsolar pv1_charging_power': Sending state 621.00000 W with 1 decimals of accuracy
[17:11:42][W][component:237]: Component esphome.coroutine took a long time for an operation (122 ms).
[17:11:42][W][component:238]: Components should block for at most 30 ms.
[17:11:43][D][pipsolar:924]: Sending polling command : QPIGS2 with length 6
[17:11:43][D][pipsolar:860]: checking crc on incoming message
[17:11:43][D][pipsolar:863]: CRC OK
[17:11:44][D][pipsolar:924]: Sending polling command : QMOD with length 4
[17:11:44][D][pipsolar:860]: checking crc on incoming message
[17:11:44][D][pipsolar:863]: CRC OK
[17:11:44][D][pipsolar:536]: Decode QMOD
[17:11:44][D][text_sensor:064]: 'pipsolar device_mode': Sending state 'B'
[17:11:45][D][pipsolar:924]: Sending polling command : QFLAG with length 5
[17:11:45][D][pipsolar:860]: checking crc on incoming message
[17:11:45][D][pipsolar:863]: CRC OK
[17:11:45][D][pipsolar:544]: Decode QFLAG
[17:11:46][D][pipsolar:924]: Sending polling command : QPIRI with length 5
[17:11:46][D][pipsolar:860]: checking crc on incoming message
[17:11:46][D][pipsolar:863]: CRC OK
[17:11:46][D][pipsolar:479]: Decode QPIRI
[17:11:46][D][pipsolar.select:031]: got value: 50
[17:11:46][D][pipsolar.select:033]: found mapped option 50A for option 50
[17:11:46][D][select:015]: 'pipsolar current_max_ac_charging_current_select': Sending state 50A (index 5)
[17:11:46][D][pipsolar.select:031]: got value: 60
[17:11:46][D][pipsolar.select:033]: found mapped option 60A for option 60
[17:11:46][D][select:015]: 'pipsolar current_max_charging_current_select': Sending state 60A (index 5)
[17:11:46][D][pipsolar.select:031]: got value: 1
[17:11:46][D][pipsolar.select:033]: found mapped option Solar only for option 1
[17:11:46][D][select:015]: 'inverter0_output_source_priority_select': Sending state Solar only (index 1)
[17:11:46][D][pipsolar.select:031]: got value: 0
[17:11:46][D][pipsolar.select:036]: could not find option 0 in mapping
[17:11:47][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[17:11:47][D][pipsolar:860]: checking crc on incoming message
[17:11:47][D][pipsolar:863]: CRC OK
[17:11:48][D][pipsolar:924]: Sending polling command : QPIGS with length 5
[17:11:48][D][pipsolar:860]: checking crc on incoming message
[17:11:48][D][pipsolar:863]: CRC OK
[17:11:48][D][pipsolar:502]: Decode QPIGS
[17:11:48][D][sensor:093]: 'pipsolar grid_voltage': Sending state 0.00000 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar grid_frequency': Sending state 0.00000 Hz with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar ac_output_voltage': Sending state 230.39999 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar ac_output_frequency': Sending state 50.00000 Hz with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar ac_output_apparent_power': Sending state 276.00000 VA with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar ac_output_active_power': Sending state 247.00000 W with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar output_load_percent': Sending state 3.00000 % with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar bus_voltage': Sending state 464.00000 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar battery_voltage': Sending state 57.60000 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar battery_charging_current': Sending state 0.00000 A with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar battery_capacity_percent': Sending state 100.00000 % with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar inverter_heat_sink_temperature': Sending state 49.00000 °C with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar pv1_input_current': Sending state 1.60000 A with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar pv1_input_voltage': Sending state 401.50000 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar battery_voltage_scc': Sending state 0.00000 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar battery_discharge_current': Sending state 0.00000 A with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar battery_voltage_offset_for_fans_on': Sending state 0.00000 V with 1 decimals of accuracy
[17:11:48][D][sensor:093]: 'pipsolar pv1_charging_power': Sending state 634.00000 W with 1 decimals of accuracy
[17:11:48][W][component:237]: Component esphome.coroutine took a long time for an operation (126 ms).
[17:11:48][W][component:238]: Components should block for at most 30 ms.
[17:11:49][D][pipsolar:924]: Sending polling command : QPIGS2 with length 6
[17:11:49][D][pipsolar:860]: checking crc on incoming message
[17:11:49][D][pipsolar:863]: CRC OK

[image]

syssi, 2025-03-27:
Could you enable the debug output and provide another log?

https://github.com/syssi/esphome-pipsolar?tab=readme-ov-file#debugging

I guess your inverter doesn't respond properly to the `QPIGS2` request.

ribalki, 2025-03-27:
Thanks. Please find the new log... Let me know ☺️

INFO Upload took 4.40 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from pipsolar/debug
INFO Connected to MQTT broker!
[18:48:25][D][esp-idf:000]: I (5881789) mqtt_client: Client asked to disconnect

[18:48:29][W][component:170]: Component mqtt cleared Warning flag
[18:48:29][I][mqtt:291]: MQTT Connected!
[18:48:29][I][app:062]: setup() finished successfully!
[18:48:29][D][uart_debug:158]: <<< "(230.0 44.3 230.0 50.0 44.3 0200 0200 48.0 46.0 41.0 58.4 58.0 3 050 060 1 1 0 1 01 0 0 54.0 0 1 46.0 10 44.0\xE5\xF5\r"
[18:48:29][I][app:100]: ESPHome version 2025.3.2 compiled on Mar 27 2025, 18:48:04
[18:48:29][I][app:102]: Project syssi.esphome-pipsolar@pip8084 version 1.0.0
[18:48:29][C][wifi:600]: WiFi:
[18:48:29][C][wifi:428]:   Local MAC: E4:65:B8:78:F3:7C
[18:48:29][C][wifi:433]:   SSID: 'maison'[redacted]
[18:48:29][C][wifi:436]:   IP Address: 192.168.50.249
[18:48:29][C][wifi:439]:   BSSID: 10:7C:61:A7:27:E0[redacted]
[18:48:29][C][wifi:441]:   Hostname: 'pipsolar'
[18:48:29][C][wifi:443]:   Signal strength: -58 dB ▂▄▆█
[18:48:29][C][wifi:447]:   Channel: 9
[18:48:29][C][wifi:448]:   Subnet: 255.255.255.0
[18:48:29][C][wifi:449]:   Gateway: 192.168.50.1
[18:48:29][C][wifi:450]:   DNS1: 103.86.96.103
[18:48:29][C][wifi:451]:   DNS2: 8.8.8.8
[18:48:29][D][pipsolar:860]: checking crc on incoming message
[18:48:29][D][pipsolar:863]: CRC OK
[18:48:29][C][logger:177]: Logger:
[18:48:29][C][logger:178]:   Max Level: DEBUG
[18:48:29][C][logger:179]:   Initial Level: DEBUG
[18:48:29][C][logger:181]:   Log Baud Rate: 0
[18:48:29][C][logger:182]:   Hardware UART: UART0
[18:48:29][D][pipsolar:479]: Decode QPIRI
[18:48:29][D][pipsolar.select:031]: got value: 50
[18:48:29][D][pipsolar.select:033]: found mapped option 50A for option 50
[18:48:29][D][select:015]: 'pipsolar current_max_ac_charging_current_select': Sending state 50A (index 5)
[18:48:29][D][pipsolar.select:031]: got value: 60
[18:48:29][D][pipsolar.select:033]: found mapped option 60A for option 60
[18:48:29][D][select:015]: 'pipsolar current_max_charging_current_select': Sending state 60A (index 5)
[18:48:29][D][switch:055]: 'pipsolar input_voltage_range': Sending state ON
[18:48:29][D][pipsolar.select:031]: got value: 1
[18:48:29][D][pipsolar.select:033]: found mapped option Solar only for option 1
[18:48:29][D][select:015]: 'inverter0_output_source_priority_select': Sending state Solar only (index 1)
[18:48:29][D][switch:055]: 'pipsolar output_source_priority_utility': Sending state OFF
[18:48:29][D][switch:055]: 'pipsolar output_source_priority_solar': Sending state ON
[18:48:29][D][switch:055]: 'pipsolar output_source_priority_battery': Sending state OFF
[18:48:29][D][pipsolar.select:031]: got value: 0
[18:48:29][D][pipsolar.select:036]: could not find option 0 in mapping
[18:48:29][D][switch:055]: 'pipsolar pv_ok_condition_for_parallel': Sending state OFF
[18:48:29][D][switch:055]: 'pipsolar pv_power_balance': Sending state ON
[18:48:29][C][uart.idf:159]: UART Bus 0:
[18:48:29][C][uart.idf:160]:   TX Pin: GPIO16
[18:48:29][C][uart.idf:161]:   RX Pin: GPIO17
[18:48:29][C][uart.idf:163]:   RX Buffer Size: 256
[18:48:29][C][uart.idf:165]:   Baud Rate: 2400 baud
[18:48:29][C][uart.idf:166]:   Data Bits: 8
[18:48:29][C][uart.idf:167]:   Parity: NONE
[18:48:29][C][uart.idf:168]:   Stop bits: 1
[18:48:29][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[18:48:30][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[18:48:30][C][pipsolar:948]: Pipsolar:
[18:48:30][C][pipsolar:949]: used commands:
[18:48:30][C][pipsolar:952]: QPIGS
[18:48:30][C][pipsolar:952]: QPIGS2
[18:48:30][C][pipsolar:952]: QMOD
[18:48:30][C][pipsolar:952]: QFLAG
[18:48:30][C][pipsolar:952]: QPIRI
[18:48:30][C][pipsolar:952]: QBATCD
[... 328 lines cut ...]
[18:48:46][D][pipsolar:860]: checking crc on incoming message
[18:48:46][D][pipsolar:863]: CRC OK
[18:48:46][D][pipsolar:544]: Decode QFLAG
[18:48:47][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[18:48:47][D][pipsolar:924]: Sending polling command : QPIRI with length 5
[18:48:48][D][uart_debug:158]: <<< "(230.0 44.3 230.0 50.0 44.3 0200 0200 48.0 46.0 41.0 58.4 58.0 3 050 060 1 1 0 1 01 0 0 54.0 0 1 46.0 10 44.0\xE5\xF5\r"
[18:48:48][D][pipsolar:860]: checking crc on incoming message
[18:48:48][D][pipsolar:863]: CRC OK
[18:48:48][D][pipsolar:479]: Decode QPIRI
[18:48:48][D][pipsolar.select:031]: got value: 50
[18:48:48][D][pipsolar.select:033]: found mapped option 50A for option 50
[18:48:48][D][select:015]: 'pipsolar current_max_ac_charging_current_select': Sending state 50A (index 5)
[18:48:48][D][pipsolar.select:031]: got value: 60
[18:48:48][D][pipsolar.select:033]: found mapped option 60A for option 60
[18:48:48][D][select:015]: 'pipsolar current_max_charging_current_select': Sending state 60A (index 5)
[18:48:48][D][pipsolar.select:031]: got value: 1
[18:48:48][D][pipsolar.select:033]: found mapped option Solar only for option 1
[18:48:48][D][select:015]: 'inverter0_output_source_priority_select': Sending state Solar only (index 1)
[18:48:48][D][pipsolar.select:031]: got value: 0
[18:48:48][D][pipsolar.select:036]: could not find option 0 in mapping
[18:48:48][D][uart_debug:158]: >>> "QBATCD\xAA&\r"
[18:48:48][D][pipsolar:924]: Sending polling command : QBATCD with length 6
[18:48:48][D][uart_debug:158]: <<< "(NAKss\r"
[18:48:48][D][pipsolar:860]: checking crc on incoming message
[18:48:48][D][pipsolar:863]: CRC OK

syssi, 2025-03-27:
This is the interesting part from the log:

```
[18:48:31][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[18:48:31][D][uart_debug:158]: <<< "(000.0 00.0 230.0 50.0 0391 0368 004 423 53.00 000 100 0032 00.1 090.3 00.00 00008 00010110 00 00 00010 0101\xBE\r"

[18:48:32][D][uart_debug:158]: >>> "QPIGS2h-\r"
[18:48:32][D][uart_debug:158]: <<< "(NAKss\r"
```

`QPIGS` is the command to request all details about the first MPPT. You can see a proper response here:

```
(000.0 00.0 230.0 50.0 0391 0368 004 423 53.00 000 100 0032 00.1 090.3 00.00 00008 00010110 00 00 00010 0101\xBE\r
```

The response to `QPIGS2` (measurements of the second MPPT) is rejected:

```
(NAKss\r
```

In other words: Your inverter doesn't provide any measurements using the PIP protocol. Please search for other issues of this repository/project. There are a lot of users having the same issue. If I remember correctly some inverter does support more than one protocol and may be you can query some details via Modbus too.

ribalki, 2025-03-27:
Ok thanks for your help and your time. Nice evening !

closed 2025-03-27 by syssi