syssi/esphome-pipsolar #181: 3.2KW SMH - Cannot switch mode to SBU

DrJaymz, 2025-04-03:
I had a mare getting this working, because I used a gender changer and ended up having to swap both TX/RX and the MAX232 Lines because of that - which was very confusing... Anyway its working.

In Home Assistant I have 6 switches available:

friendly_name: pipsolar input_voltage_range
friendly_name: pipsolar output_source_priority_battery
friendly_name: pipsolar output_source_priority_hybrid
friendly_name: pipsolar output_source_priority_solar
friendly_name: pipsolar output_source_priority_utility
friendly_name: pipsolar pv_power_balance

I cannot switch the mode.  Its in utility mode, but I cannot set to SBU.  When I selected utility mode that worked, but I cannot turn it out of that mode now.  If I'm being dumb please let me know.  I noted with the ess app setting solar was ignored, and I could only select SBU and U mode.

I set it up in HA showing me the difference between generated and output and all of that seems to work nicely.

[image]

DrJaymz, 2025-04-04:
It did switch from Line Mode later on.  So my question really becomes; with this protocol which command / switch should I be issuing?  The log seems to say its repeatedly telling the inverter to be in line mode - or is it the inverter that is just telling me its in line mode?

syssi, 2025-07-13:
Are you still looking for a solution here?

remiserriere, 2025-12-04:
I am also facing the same issue (I think) with a SMH 2.2KW with the ESP8266 example. I can switch the inverter to battery mode once, switch it back to grid, but then I can't change the mode anymore until I reboot the ESP8266. @madurapa is this code supposed to go in the device yaml config file or in Home Assistant?