syssi/esphome-pipsolar #185: Help interpreting the log

pmfernandes, 2025-05-11:
Hi,

I've applied this [configuration](https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml). I need your help interpreting this log:

This is the set of commands supported by the pipsolar component!
15:42:04 [D] [uart_debug:158] >>> "QPI\xBE\xAC\r"
15:42:05 [D] [uart_debug:158] >>> "QDIq\e\r"
15:42:06 [D] [uart_debug:158] >>> "QFLAG\x98t\r"
15:42:07 [D] [uart_debug:158] >>> "QMN\xBBd\r"
15:42:08 [D] [uart_debug:158] >>> "QMODI\xC1\r"
15:42:09 [D] [uart_debug:158] >>> "QPIGS\xB7\xA9\r"
15:42:10 [D] [uart_debug:158] >>> "QPIRI\xF8T\r"
15:42:12 [D] [uart_debug:158] >>> "QPIWS\xB4\xDA\r"
15:42:13 [D] [uart_debug:158] >>> "QT\'\xFF\r"
15:42:13 [D] [sensor:093] 'PIP Solar WiFi RSSI': Sending state -72.00000 dBm with 0 decimals of accuracy
15:42:14 [D] [uart_debug:158] >>> "QPGS0?\xDA\r"
15:42:15 [D] [uart_debug:158] >>> "QPGS1/\xFB\r"
15:42:16 [D] [uart_debug:158] >>> "QPGS2\x1F\x98\r"
15:42:17 [I] [main:574] Testing PI41 split phase / multiple strings commands...
15:42:17 [D] [uart_debug:158] >>> "QPIGS2h-\r"
15:42:17 [W] [component:239] Component interval took a long time for an operation (52 ms).
15:42:17 [W] [component:240] Components should block for at most 30 ms.
15:42:18 [D] [uart_debug:158] >>> "QP2GS0\x14\x05\r"
15:42:18 [W] [component:239] Component interval took a long time for an operation (51 ms).
15:42:18 [W] [component:240] Components should block for at most 30 ms.
15:42:19 [D] [uart_debug:158] >>> "QP2GS1\x04$\r"
15:42:19 [W] [component:239] Component interval took a long time for an operation (51 ms).
15:42:19 [W] [component:240] Components should block for at most 30 ms.
15:42:20 [I] [main:598] Testing unsupported PI18 commands...
15:42:20 [D] [uart_debug:158] >>> "^P005PIq\x8B\r"
15:42:20 [W] [component:239] Component interval took a long time for an operation (55 ms).
15:42:20 [W] [component:240] Components should block for at most 30 ms.
15:42:21 [D] [uart_debug:158] >>> "^P005GSX\x14\r"
15:42:21 [W] [component:239] Component interval took a long time for an operation (56 ms).
15:42:21 [W] [component:240] Components should block for at most 30 ms.
15:42:22 [D] [uart_debug:158] >>> "^P006MOD\xDD\xBE\r"
15:42:22 [W] [component:239] Component interval took a long time for an operation (59 ms).
15:42:22 [W] [component:240] Components should block for at most 30 ms.
15:42:23 [I] [main:622] Testing unsupported PI17 commands...
15:42:23 [D] [uart_debug:158] >>> "^P003PI\r"
15:42:24 [D] [uart_debug:158] >>> "^P004MOD\r"
15:42:25 [D] [uart_debug:158] >>> "^P005FLAG\r"
15:42:25 [W] [component:239] Component interval took a long time for an operation (56 ms).
15:42:25 [W] [component:240] Components should block for at most 30 ms.
15:42:26 [I] [main:646] Testing unsupported PI16 commands...
15:42:26 [D] [uart_debug:158] >>> "QPI\r"
15:42:27 [D] [uart_debug:158] >>> "QMOD\r"
15:42:28 [D] [uart_debug:158] >>> "QPIGS\r"
15:42:29 [D] [uart_debug:158] >>> "QPIRI\r"
15:42:30 [D] [uart_debug:158] >>> "QMOD\r"
15:42:30 [I] [main:680] Testing unsupported Qx commands...
15:42:30 [D] [uart_debug:158] >>> "Q1\r"
15:42:30 [W] [component:239] Component interval took a long time for an operation (57 ms).
15:42:30 [W] [component:240] Components should block for at most 30 ms.
15:42:31 [D] [uart_debug:158] >>> "Q2\r"
15:42:32 [D] [uart_debug:158] >>> "Q4\r"
15:42:33 [D] [uart_debug:158] >>> "Q6\r"
15:42:35 [I] [main:711] Done. Repeating..

syssi, 2025-05-11:
This (`>>>`) is outgoing traffic only. A response would be prefixed by `<<<`. Please check your wiring if you are sure the inverter should support one of the PIP protocols.

pmfernandes, 2025-05-11:
Got it. I'll double check the wiring.