syssi/esphome-pipsolar #188: How to read pv voltage, current and power from MPPT 1 & 2 for EASUN 8kW ?

ornolio, 2025-07-05:
How can we have access to pv voltage, current and power sensors separately by each MPPT input no. 1 and no. 2, to monitor pv strings parameters ?
Thank You.

syssi, 2025-07-08:
Does your inverter support the `QPIGS2` command properly?

ornolio, 2025-07-09:
Currently I use Qpigs commands into HA Esphome. 
I don't know if my inverter/release supports qpigs2.
How can check it?

syssi, 2025-07-09:
Could you flash this protocol tester and provide an ESPHome log: https://github.com/syssi/esphome-pipsolar/blob/main/tests/esp8266-test-protocols.yaml

ornolio, 2025-07-14:
Herunder the log of the test-protocols you suggest:
INFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/test-qpigs2.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test-qpigs2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/esp8266/preferences.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/logger/task_log_buffer.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/ota/ota_backend.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/ota/ota_backend_arduino_libretiny.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/safe_mode/safe_mode.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/socket/lwip_sockets_impl.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component_esp8266.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component_host.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component_libretiny.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_component_rp2040.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/uart/uart_debugger.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/wifi/wifi_component_libretiny.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/core/application.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/core/color.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/core/component.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/test-qpigs2/src/esphome/core/controller.cpp.o
[... 512 lines cut ...]
[15:07:41][D][uart_debug:158]: <<< "(NAKss\r"
[15:07:42][D][uart_debug:158]: >>> "QMOD\r"
[15:07:42][D][uart_debug:158]: <<< "(NAKss\r"
[15:07:43][D][uart_debug:158]: >>> "QPIGS\r"
[15:07:43][D][uart_debug:158]: <<< "(NAKss\r"
[15:07:44][D][uart_debug:158]: >>> "QPIRI\r"
[... cut ...]
[15:07:46][D][uart_debug:158]: <<< "(NAKss\r"
[15:07:47][D][uart_debug:158]: >>> "Q4\r"
[15:07:47][D][uart_debug:158]: <<< "(NAKss\r"
[15:07:48][D][uart_debug:158]: >>> "Q6\r"
[15:07:48][D][uart_debug:158]: <<< "(NAKss\r"
[15:07:49][I][main:722]: Done. Repeating...

Please note on the attached screenshot 

[image]

I got using the WhatchPower APP on an android smatphone by wifi connection I'm able to read the typical QPIGS & QPIGS2 parameter, so I argued the firmware release is at a right level to show the requestes PV values.

ornolio, 2025-07-14:
Thank You a lot for your support.

ornolio, 2025-08-17:
I did clean my YAML file. Hereunder is the log output:
[11:55:59][I][main:749]: Done. Repeating...
[11:56:47][I][main:522]: Testing PI30/PI30MAX/PI30REVO/PI41 commands...
[11:56:47][I][main:525]: This is the set of commands supported by the pipsolar component!
[11:56:47][D][uart_debug:158]: >>> "QPI\xBE\xAC\r"
[11:56:47][D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[11:56:48][D][uart_debug:158]: >>> "QDIq\e\r"
[11:56:48][D][uart_debug:158]: <<< "(230.0 50.0 0030 44.0 54.0 56.4 46.0 60 0 0 2 0 0 0 0 0 1 1 1 0 1 0 54.0 0 1 224\xEB\xBC\r"
[11:56:49][D][uart_debug:158]: >>> "QFLAG\x98t\r"
[11:56:49][D][uart_debug:158]: <<< "(EuxyzDabjkvG\x81\r"
[11:56:50][D][uart_debug:158]: >>> "QMN\xBBd\r"
[11:56:50][D][uart_debug:158]: <<< "(MKS2-8000\xB2\x8D\r"
[11:56:51][D][uart_debug:158]: >>> "QMODI\xC1\r"
[11:56:51][D][uart_debug:158]: <<< "(C\xF7\xE8\r"
[11:56:53][D][uart_debug:158]: >>> "QPIGS\xB7\xA9\r"
[11:56:53][D][uart_debug:158]: <<< "(000.0 00.0 000.0 00.0 0000 0000 000 443 54.00 000 058 0042 00.0 392.4 00.00 00000 00000110 00 00 00010 000B\xBA\r"
[11:56:54][D][uart_debug:158]: >>> "QPIRI\xF8T\r"
[11:56:54][D][uart_debug:158]: <<< "(230.0 34.7 230.0 50.0 34.7 8000 8000 48.0 51.0 48.0 54.4 54.0 2 002 060 0 1 3 9 01 0 0 52.0 0 1 480 0 060|\xBE\r"
[11:56:55][D][uart_debug:158]: >>> "QPIWS\xB4\xDA\r"
[11:56:55][D][uart_debug:158]: <<< "(000001000000000001000000000000000000l\xB8\r"
[11:56:56][D][uart_debug:158]: >>> "QT\'\xFF\r"
[11:56:56][D][uart_debug:158]: <<< "(20250817111347\xB1Z\r"
[11:56:57][D][uart_debug:158]: >>> "QPGS0?\xDA\r"
[11:56:58][D][uart_debug:158]: <<< "(0 92932212102232 P 00 000.0 00.00 000.0 00.00 0000 0000 000 54.0 000 058 392.0 000 00000 00000 000 10100100 0 3 060 120 002 00 000 392.4 00\xD8\xCA\r"
[11:56:58][D][uart_debug:158]: >>> "QPGS1/\xFB\r"
[11:56:59][D][uart_debug:158]: <<< "(0 92932212102232 P 00 000.0 00.00 000.0 00.00 0000 0000 000 54.0 000 058 392.4 000 00000 00000 000 10100100 0 3 060 120 002 00 000 393.0 00\xA8\xCA\r"
[11:56:59][D][uart_debug:158]: >>> "QPGS2\x1F\x98\r"
[11:57:00][D][uart_debug:158]: <<< "(0 00000000000000 P 00 000.0 00.00 000.0 00.00 0000 0000 000 00.0 000 000 000.0 000 00000 00000 000 00000000 0 0 000 000 000 00 000 000.0 00\xB5J\r"
[11:57:00][I][main:612]: Testing PI41 split phase / multiple strings commands...

> [11:57:00][D][uart_debug:158]: >>> "QPIGS2h-\r"
> [11:57:00][D][uart_debug:158]: <<< "(00.0 392.3 00036 \xDF\xC2\r"

[11:57:01][D][uart_debug:158]: >>> "QP2GS0\x14\x05\r"
[11:57:01][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:03][D][uart_debug:158]: >>> "QP2GS1\x04$\r"
[11:57:03][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:04][I][main:636]: Testing unsupported PI18 commands...
[11:57:04][D][uart_debug:158]: >>> "^P005PIq\x8B\r"
[11:57:04][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:05][D][uart_debug:158]: >>> "^P005GSX\x14\r"
[11:57:05][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:06][D][uart_debug:158]: >>> "^P006MOD\xDD\xBE\r"
[11:57:06][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:07][I][main:660]: Testing unsupported PI17 commands...
[11:57:07][D][uart_debug:158]: >>> "^P003PI\r"
[11:57:07][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:08][D][uart_debug:158]: >>> "^P004MOD\r"
[11:57:08][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:09][D][uart_debug:158]: >>> "^P005FLAG\r"
[11:57:09][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:10][I][main:684]: Testing unsupported PI16 commands...
[11:57:10][D][uart_debug:158]: >>> "QPI\r"
[11:57:10][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:11][D][uart_debug:158]: >>> "QMOD\r"
[11:57:11][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:12][D][uart_debug:158]: >>> "QPIGS\r"
[11:57:12][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:13][D][uart_debug:158]: >>> "QPIRI\r"
[11:57:13][D][uart_debug:158]: <<< "(PI30\x9A\v\r"
[11:57:14][D][uart_debug:158]: >>> "QMOD\r"
[11:57:14][I][main:718]: Testing unsupported Qx commands...
[11:57:14][D][uart_debug:158]: >>> "Q1\r"
[11:57:14][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:15][D][uart_debug:158]: <<< "(00001 20031 01 00 00 035 032 042 032 02 00 000 0030 0026 0000 00.00 13 0 060 030 060 030 54.00 000 120 0 0000\a\xFE\r"
[11:57:15][D][uart_debug:158]: >>> "Q2\r"
[11:57:15][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:17][D][uart_debug:158]: >>> "Q4\r"
[11:57:17][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:18][D][uart_debug:158]: >>> "Q6\r"
[11:57:18][D][uart_debug:158]: <<< "(NAKss\r"
[11:57:19][I][main:749]: Done. Repeating...