syssi/esphome-pipsolar #189: Split phase LV 5048

cheomfp, 2025-07-18:
[logs_test6_logs.txt](https://github.com/user-attachments/files/21307417/logs_test6_logs.txt)

syssi, 2025-07-19:
Could you enable the debug output and provide another log?

https://github.com/syssi/esphome-pipsolar?tab=readme-ov-file#debugging

I would like to see the raw traffic of your inverter.

cheomfp, 2025-07-19:
[logs_invsolar_logs (1).txt](https://github.com/user-attachments/files/21330883/logs_invsolar_logs.1.txt)

cheomfp, 2025-07-19:
**Let me know if I'm doing anything wrong. Here's my current YAML file after the changes.**

[YALM file.txt](https://github.com/user-attachments/files/21331673/YALM.file.txt)

cheomfp, 2025-07-19:
[PI41_LV5048.5KW.protocol-20190222.for.customer (1).pdf](https://github.com/user-attachments/files/21331679/PI41_LV5048.5KW.protocol-20190222.for.customer.1.pdf)

It is a 5kVA MPPSOLAR split-phase inverter. 110 volts phase-to-neutral and 240 volts phase-to-phase. This is a two-in-one inverter