syssi/esphome-pipsolar #192: yaml file inverter Smh III 48v.

pnrntn, 2025-08-04:
Hi, I need help configuring my SMH III 48V inverter with Home Assistant. I'm using an ESP32 mod az-delivery-devkit-v2 with an RS232 to TTL module. I need a valid yaml configuration to get started. My Home Assistant recognizes the connection to the ESP32 but the data is all null.

madurapa, 2025-08-10:
Example YML is already in the repo: https://github.com/syssi/esphome-pipsolar/blob/pip8048/esp32-example.yaml

pnrntn, 2025-08-11:
Thanks, now I've managed to get it working. I uploaded the file that allows you to change the charging amps from the grid, but I can't change the value from Home Assistant. I can work with the buttons, but not with entities like this one.
select.pipsolar_pipsolar_current_max_charging_current_select

madurapa, 2025-08-11:
Take a look at the bottom of this YML: https://github.com/syssi/esphome-pipsolar/issues/164#issuecomment-2560092312

Or you can paste your YML so we can check what issues you are facing.